espressif/esp-idf #6889: Failed to use Rotary_encoder_example

zsjlovestudy, 2021-04-16:
Because I used pins 10 and 11 on the PCB , when I changed the GPIO pins in the example, there was an error.

I need to use four rotary encoders simultaneously。However, after I made four copies according to the template of the sample code, he made a mistake of interrupting the repeated registration. Could you please tell me how to solve this?

When I comment out the code that breaks registration, I can run all three at once, except for pins 10 and 11.

[image]

[image]

[image]

suda-morris, 2021-04-16:
GPIO10 and GPIO11 on ESP32 are used by SPI flash by default, please refer to datasheet and  `soc/esp32/include/soc/io_mux_reg.h` for more details.
As for the example itself, if you want to drive multiple encoders with overflow handling (in interrupt handler), you need to alter the example, make sure calling `pcnt_isr_service_install` only once globally. And handle the overflow event for each pcnt unit respectively in the ISR callback.

zsjlovestudy, 2021-04-16:
Hello, I have read the io_mux_reg.h file, but I don't quite understand it.First I want to know if 10 and 11 can be modified to be used as encoders.Secondly, if possible, how to modify it？

suda-morris, 2021-04-19:
Hi @zsjlovestudy GPIO11 is `SPI_CS0_GPIO_NUM` you can't use it for other purposes.

zsjlovestudy, 2021-04-21:
> Hi @zsjlovestudy GPIO11 is `SPI_CS0_GPIO_NUM` you can't use it for other purposes.

Hello, could you please tell me if there are any other pins like 11 pins? Because I need to specify the pin  to PWM and encoder at present.
[image]

suda-morris, 2021-04-21:
The following GPIOs are used by SPI Flash by default:
```c
#define SPI_CLK_GPIO_NUM                            6
#define SPI_CS0_GPIO_NUM                            11
#define SPI_Q_GPIO_NUM                              7
#define SPI_D_GPIO_NUM                              8
#define SPI_WP_GPIO_NUM                             10
#define SPI_HD_GPIO_NUM                             9
```

camrbuss, 2021-05-03:
In order to create multiple encoder instances the `ROTARY_CHECK` needs to be avoided from the `pcnt_isr_service_install` call. I added this switch to the "register interrupt handler" section of rotary_encoder_pcnt_ec11.c so that I could have multiple encoder instances.

``` c
	// register interrupt handler
	switch(pcnt_isr_service_install(0))
	{
		case ESP_OK:
			break;
		case ESP_ERR_INVALID_STATE:
			ESP_LOGI(TAG, "pcnt isr already installed");
			break;
		default:
			ROTARY_CHECK(1==0, "install isr service failed", err, ESP_FAIL);
	}
```

Does anyone want a pull request for something like this? Is this component going to be an "official" component one day?

suda-morris, 2021-05-06:
Hi @camrbuss yes, the issue you found is correct. The interrupt is shared by PCNT channels, it should be installed only once.

>  Is this component going to be an "official" component one day?

We don't have a plan to raise the level of this example component, but we're working on new PCNT driver APIs to make that peripheral more flexible, especially for encoders use cases.

yangwc, 2021-07-13:
I ran into the same problem. I configured two encoder. 
[image]

[image]

closed 2021-07-16 by espressif-bot