espressif/esp-idf #6890: Can get an examples to run on ESP32-C3 - pulled current master

lisper, 2021-04-16:
## Environment
Ubuntu 20.04.2 LTS

## Problem Description
I got an ESP32-C3 in the mail.  I decided to try it out.  
I cloned the repository and set the branch to master.  I followed all the instructions to install as if for the previous chip revision and installed the idf.  That went well and worked.

I then followed the instructions on the paper letter which came with the C3.  Those instruction are below:

```
cd esp-idf
git checkout master
git pull
git submodule update --init --rescursive     <-- the git commands did nothing, since I had just cloned and installed the repo

./install.sh
. ./export.sh
cd examples/get-started-blink
idf.py --preview set-target esp32c3
idf.py flash monitor
```

The install and build went fine.  The flash worked.  But when it went to start the monitor the command failed because there was no argument for the --revision argument.   Which ever script which created the idf_monitor.py command line had "" (null string) as the revision.  You could see it because it printed out the command.

So, I ran the command by hand, and put "3" in as the revision.  This produced an endless stream of output, as follows:

```
SP-ROM:esp32c3-20200918
Build:Sep 18 2020
rst:0x3 (RTC_SW_SYS_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x403d11b2
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1630
load:0x403ce000,len:0x880
load:0x403d0000,len:0x2ba4
entry 0x403ce000
I (34) boot: ESP-IDF v4.4-dev-960-gcf457d412-dirty 2nd stage bootloader
I (35) boot: compile time 16:24:36
I (35) boot: chip revision: 2
E (38) boot_comm: can't run on lower chip revision, expected 2, found 3
```

I then tried following the instruction on the web page.  Basically making a copy of the hello_world example directory in ~/esp

I did these commands:

`cd ~/esp/hello_world`
`idf.py set-target esp32c3`
`idf.py menuconfig`                 <-- didn't change anything, but did save the config in the local directory
`idf.py build`
`idf.py flash`
`idf.py monitor`

Again, the build worked fine. The flash worked, but the monitor spewed out the same thing above, over and over.

Seems like the bootloader is being build with a wrong version #?  I found where the line is printed and it's checking some version # out of a flash header (I think) against a constant.

I'm new to this, and I will figure it out.  I just thought I'd report that I could not get the base instruction to work.

Thanks!
  

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

negativekelvin, 2021-04-17:
https://github.com/espressif/esp-idf/blob/cf457d412a7748139b77c94e7debe72ee86199af/components/esp32c3/Kconfig#L26-L42

https://github.com/espressif/esp-idf/commit/c425ac7bed1a63bb9332bd83a6e075499a244078

https://github.com/espressif/esp-idf/commit/10d1e4ab1be9ae71cd8e149dbcc158b11056c6e8

chegewara, 2021-04-17:
> I (35) boot: chip revision: 2
E (38) boot_comm: can't run on lower chip revision, expected 2, found 3

Its a bug in printed message. Shouls be:
```
expected 3, found 2
```

thinhx2, 2021-04-17:
> https://github.com/espressif/esp-idf/blob/cf457d412a7748139b77c94e7debe72ee86199af/components/esp32c3/Kconfig#L26-L42
> 
> [c425ac7](https://github.com/espressif/esp-idf/commit/c425ac7bed1a63bb9332bd83a6e075499a244078)
> 
> [10d1e4a](https://github.com/espressif/esp-idf/commit/10d1e4ab1be9ae71cd8e149dbcc158b11056c6e8)

thank you, seem like ESPRESSIF having new revision of esp32-c3 mcu. mostly dev kit sended to tester is revision 2

projectgus, 2021-04-19:
Hi @lisper,

Sorry for the inconvenience. As noted above, the default minimum revision for ESP32-C3 was recently changed to revision 3 but some ESP32-C3 samples are revision 2. You can change [this configuration setting to rev 2](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/kconfig.html#config-esp32c3-rev-min), rebuild the project and flash, and it will start booting.

The fix for the idf_monitor.py argument has been merged internally and will be deployed to GitHub soon.

Regarding the bootloader log error message being confusing, we will fix this ASAP as well and update the issue.

lisper, 2021-04-22:
Thanks!  that's perfect.  and thanks for being so responsive!