espressif/esp-idf #6891: modbus mb_m.c， 发送指令前如何现清除上一个指令遗留的接收数据？

hnhkj, 2021-04-17:
这个事件`EV_ERROR_RECEIVE_DATA`没有对已经接收到的数据进行清除处理。这会造成下一个指令接收到消息后，将没有清除的数据作为返回数据处理。造成后续指令不能正确执行。

```c
eMBErrorCode
eMBMasterPoll( void )
{
    static UCHAR    *ucMBFrame = NULL;
    static UCHAR    ucRcvAddress;
    static UCHAR    ucFunctionCode;
    static USHORT   usLength;
    static eMBException eException;
    int             i;
    int             j;
    eMBErrorCode    eStatus = MB_ENOERR;
    eMBMasterEventType    eEvent;
    eMBMasterErrorEventType errorType;

    /* Check if the protocol stack is ready. */
    if( eMBState != STATE_ENABLED )
    {
        return MB_EILLSTATE;
    }

    /* Check if there is a event available. If not return control to caller.
     * Otherwise we will handle the event. */
    if ( xMBMasterPortEventGet( &eEvent ) == TRUE )
    {
        // In some cases it is possible that more than one event set
        // together (even from one subset mask) than process them consistently
        if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_READY ) ) {
            ESP_LOGD(MB_PORT_TAG, "%s:EV_MASTER_READY", __func__);
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_READY );
        } else if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_FRAME_RECEIVED ) ) {
            eStatus = peMBMasterFrameReceiveCur( &ucRcvAddress, &ucMBFrame, &usLength);
            ESP_LOG_BUFFER_HEX_LEVEL("POLL RCV buffer:", (void*)ucMBFrame, (uint16_t)usLength, ESP_LOG_DEBUG);
            ESP_LOG_BUFFER_HEX("*1*", (void*)ucMBFrame,(uint16_t)usLength);
            // Check if the frame is for us. If not ,send an error process event.
            if ( ( eStatus == MB_ENOERR ) && ( ucRcvAddress == ucMBMasterGetDestAddress() ) )
            {
                ESP_LOGD(MB_PORT_TAG, "%s: Packet data received successfully (%u).", __func__, eStatus);
                ( void ) xMBMasterPortEventPost( EV_MASTER_EXECUTE );
            }
            else
            {
                printf("*1*error\n");
                vMBMasterSetErrorType(EV_ERROR_RECEIVE_DATA);
                ESP_LOGI( MB_PORT_TAG, "%s: Packet data receive failed (addr=%u)(%u).",
                                       __func__, ucRcvAddress, eStatus);
                ( void ) xMBMasterPortEventPost( EV_MASTER_ERROR_PROCESS );
            }
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_FRAME_RECEIVED );
        } else if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_EXECUTE ) ) {
            if ( !ucMBFrame )
            {
                return MB_EILLSTATE;
            }
            ESP_LOGD(MB_PORT_TAG, "%s:EV_MASTER_EXECUTE", __func__);
            ucFunctionCode = ucMBFrame[MB_PDU_FUNC_OFF];
            eException = MB_EX_ILLEGAL_FUNCTION;
            /* If receive frame has exception. The receive function code highest bit is 1.*/
            if (ucFunctionCode & MB_FUNC_ERROR)
            {
                eException = (eMBException)ucMBFrame[MB_PDU_DATA_OFF];
            } else {
                for ( i = 0; i < MB_FUNC_HANDLERS_MAX; i++ )
                {
                    /* No more function handlers registered. Abort. */
                    if (xMasterFuncHandlers[i].ucFunctionCode == 0)
                    {
                        break;
                    }
                    if (xMasterFuncHandlers[i].ucFunctionCode == ucFunctionCode)
                    {
                        vMBMasterSetCBRunInMasterMode(TRUE);
                        /* If master request is broadcast,
                         * the master need execute function for all slave.
                         */
                        if ( xMBMasterRequestIsBroadcast() )
                        {
                            usLength = usMBMasterGetPDUSndLength();
                            for(j = 1; j <= MB_MASTER_TOTAL_SLAVE_NUM; j++)
                            {
                                vMBMasterSetDestAddress(j);
                                eException = xMasterFuncHandlers[i].pxHandler(ucMBFrame, &usLength);
                            }
                        }
                        else
                        {
                            eException = xMasterFuncHandlers[i].pxHandler( ucMBFrame, &usLength );
                        }
                        vMBMasterSetCBRunInMasterMode( FALSE );
                        break;
                    }
                }
            }
            /* If master has exception ,Master will send error process.Otherwise the Master is idle.*/
            if (eException != MB_EX_NONE)
            {
                vMBMasterSetErrorType(EV_ERROR_EXECUTE_FUNCTION);
                ( void ) xMBMasterPortEventPost( EV_MASTER_ERROR_PROCESS );
            }
            else
            {
                vMBMasterSetErrorType(EV_ERROR_OK);
                ( void ) xMBMasterPortEventPost( EV_MASTER_ERROR_PROCESS );
            }
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_EXECUTE );
        } else if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_FRAME_TRANSMIT ) ) {
            ESP_LOGD(MB_PORT_TAG, "%s:EV_MASTER_FRAME_TRANSMIT", __func__);
            /* Master is busy now. */
            vMBMasterGetPDUSndBuf( &ucMBFrame );
            eStatus = peMBMasterFrameSendCur( ucMBMasterGetDestAddress(), ucMBFrame, usMBMasterGetPDUSndLength() );
            if (eStatus != MB_ENOERR)
            {
                ESP_LOGE( MB_PORT_TAG, "%s:Frame send error. %d", __func__, eStatus );
            }
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_FRAME_TRANSMIT );
        } else if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_FRAME_SENT ) ) {
            ESP_LOGD( MB_PORT_TAG, "%s:EV_MASTER_FRAME_SENT", __func__ );
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_FRAME_SENT );
        } else if ( MB_PORT_CHECK_EVENT( eEvent, EV_MASTER_ERROR_PROCESS ) ) {
            ESP_LOGD( MB_PORT_TAG, "%s:EV_MASTER_ERROR_PROCESS", __func__ );
            /* Execute specified error process callback function. */
            errorType = eMBMasterGetErrorType( );
            vMBMasterGetPDUSndBuf( &ucMBFrame );
            switch ( errorType )
            {
                case EV_ERROR_RESPOND_TIMEOUT:
                    vMBMasterErrorCBRespondTimeout( ucMBMasterGetDestAddress( ),
                            ucMBFrame, usMBMasterGetPDUSndLength( ) );
                    break;
                case EV_ERROR_RECEIVE_DATA:
                    vMBMasterErrorCBReceiveData( ucMBMasterGetDestAddress( ),
                            ucMBFrame, usMBMasterGetPDUSndLength( ) );
                    break;
                case EV_ERROR_EXECUTE_FUNCTION:
                    vMBMasterErrorCBExecuteFunction( ucMBMasterGetDestAddress( ),
                            ucMBFrame, usMBMasterGetPDUSndLength( ) );
                    break;
                case EV_ERROR_OK:
                    vMBMasterCBRequestSuccess( );
                    break;
                default:
                    ESP_LOGE( MB_PORT_TAG, "%s: incorrect error type = %d.", __func__, errorType);
                    break;
            }
            vMBMasterRunResRelease( );
            MB_PORT_CLEAR_EVENT( eEvent, EV_MASTER_ERROR_PROCESS );
        }
        if ( eEvent ) {
            // Event processing is done, but some poll events still set then
            // postpone its processing for next poll cycle (rare case).
            ESP_LOGW( MB_PORT_TAG, "%s: Unprocessed event %d.", __func__, eEvent );
            ( void ) xMBMasterPortEventPost( eEvent );
        }
    } else {
        // Something went wrong and task unblocked but there are no any correct events set
        ESP_LOGE( MB_PORT_TAG, "%s: Unexpected event triggered %d.", __func__, eEvent );
        eStatus = MB_EILLSTATE;
    }
    return eStatus;
}
```

hnhkj, 2021-04-18:
在`eErrStatus = MB_MRE_EXE_FUN;`后面增加了`vMBMasterRunResRelease();`调用，问题消除。但是这应该不是一个理想的方案。

我看到在`eMBMasterPoll()`函数中对错误消息的处理有对`vMBMasterRunResRelease()`进行调用，为什么没有效果？在`EV_MASTER_ERROR_EXECUTE_FUNCTION`消息产生后，多一次调用`vMBMasterRunResRelease()`问题消除。

我想应该还是故障恢复的问题。

```
eMBMasterReqErrCode eMBMasterWaitRequestFinish( void ) {
    eMBMasterReqErrCode eErrStatus = MB_MRE_NO_ERR;
    eMBMasterEventType xRecvedEvent;

    EventBits_t uxBits = xEventGroupWaitBits( xEventGroupMasterHdl, // The event group being tested.
                                                MB_EVENT_REQ_MASK,  // The bits within the event group to wait for.
                                                pdTRUE,             // Masked bits should be cleared before returning.
                                                pdFALSE,            // Don't wait for both bits, either bit will do.
                                                1000 );             // Wait forever for either bit to be set. //portMAX_DELAY
    xRecvedEvent = (eMBMasterEventType)(uxBits);
    if (xRecvedEvent) {
        ESP_LOGD(MB_PORT_TAG,"%s: returned event = 0x%x", __func__, xRecvedEvent);
        if (!(xRecvedEvent & MB_EVENT_REQ_MASK)) {
            // if we wait for certain event bits but get from poll subset
            ESP_LOGE(MB_PORT_TAG,"%s: incorrect event set = 0x%x", __func__, xRecvedEvent);
        }
        if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_PROCESS_SUCCESS)) {
            eErrStatus = MB_MRE_NO_ERR;
        } else if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_ERROR_RESPOND_TIMEOUT)) {
            eErrStatus = MB_MRE_TIMEDOUT;
        } else if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_ERROR_RECEIVE_DATA)) {
            eErrStatus = MB_MRE_REV_DATA;
        } else if (MB_PORT_CHECK_EVENT(xRecvedEvent, EV_MASTER_ERROR_EXECUTE_FUNCTION)) {
            eErrStatus = MB_MRE_EXE_FUN;
            vMBMasterRunResRelease();
        }
    } else {
        ESP_LOGE(MB_PORT_TAG,"%s: Incorrect event or timeout xRecvedEvent = 0x%x", __func__, uxBits);
        assert(0);
    }
    return eErrStatus;
}
```

shiweichn, 2021-04-21:
我也遇到了这个问题。感觉idf的modbus库不是很稳啊，遇到了各种奇怪问题。

alisitsyn, 2021-04-23:
@hnhkj,

Sorry for the delay with response. The issue is known and still not fixed because of the delays with merging. I see that your version of ESP-IDF is not latest could you let me know the actual version you are using?

> EV_MASTER_ERROR_EXECUTE_FUNCTION 这个消息产生后如何是modbus复位到初始状态？

In normal FSM sequence (if the slave response time set correctly and no any tasks with higher priority on the same CPU) the events are cleared automatically after triggering:

``` code
eMBMasterReqErrCode eMBMasterWaitRequestFinish( void ) {
    eMBMasterReqErrCode eErrStatus = MB_MRE_NO_ERR;
    eMBMasterEventType xRecvedEvent;

    EventBits_t uxBits = xEventGroupWaitBits( xEventGroupMasterHdl, // The event group being tested.
                                                MB_EVENT_REQ_MASK,  // The bits within the event group to wait for.
                                                pdTRUE,             // <<<<<< clear triggered event here // Masked bits should be cleared before returning.
                                                pdFALSE,            // Don't wait for both bits, either bit will do.
                                                portMAX_DELAY );    // Wait forever for either bit to be set.
```

``` code
BOOL
xMBMasterPortEventGet( eMBMasterEventType* eEvent )
{
    EventBits_t uxBits;
    BOOL    xEventHappened = FALSE;
    uxBits = xEventGroupWaitBits( xEventGroupMasterHdl, // The event group being tested.
                                    MB_EVENT_POLL_MASK, // The bits within the event group to wait for.
                                    pdTRUE,             // <<<<<< clear triggered event here  Masked bits should be cleared before returning.
                                    pdFALSE,            // Don't wait for both bits, either bit will do.
                                    portMAX_DELAY);     // Wait forever for either bit to be set.
```
> 1. ESP32 作为modbus主机模式 
2. modbus发送一个指令，可能出现一个错误EV_ERROR_RECEIVE_DATA，这会造成这个指令结束，返回一个错误消息。
3. 这以后系统就不能正常工作了。

The issue is known and related to the handling of slave response to previous transaction which was received after respond timeout.

> 这个事件EV_ERROR_RECEIVE_DATA没有对已经接收到的数据进行清除处理。这会造成下一个指令接收到消息后，将没有清除的数据作为返回数据处理。造成后续指令不能正确执行。
eMBMasterPoll()函数状态切换，正常和错误的状态切换

The receive task trigger the event `EV_MASTER_FRAME_RECEIVED` when new modbus frame is received. 
When it happens the receive buffer contains actual received frame. Do not need to initialize the buffer after each transaction because it reflects the data received from slave.
The parameter data can be incorrect or can contain the data of previous transaction and actual reason for the error `EV_ERROR_RECEIVE_DATA` with incorrect data set for parameter will be explained below.

> 在eErrStatus = MB_MRE_EXE_FUN;后面增加了vMBMasterRunResRelease();调用，问题消除。但是这应该不是一个理想的方案。
 我看到在eMBMasterPoll()函数中对错误消息的处理有对vMBMasterRunResRelease()进行调用，为什么没有效果？在EV_MASTER_ERROR_EXECUTE_FUNCTION消息产生后，多一次调用vMBMasterRunResRelease()问题消除
我想应该还是故障恢复的问题。

All errors including `EV_ERROR_RECEIVE_DATA` are handled in the FSM loop: `eMBMasterPoll()` on `EV_MASTER_ERROR_PROCESS` stage. 
The release of Runsema in `eMBMasterWaitRequestFinish()` will not solve the issue properly because it will start the new transaction before completion of error handling in FSM.

The description of root reason for this issue:

All errors including `EV_ERROR_RECEIVE_DATA` are handled in the FSM loop: `eMBMasterPoll()` on `EV_MASTER_ERROR_PROCESS` stage. 
The release of Runsema in `eMBMasterWaitRequestFinish()` will not solve the issue properly because it will start the new transaction before completion of error handling in FSM.

The description of root reason for this issue:

1. The response timeout is set to the value < actual slave response time. Other thing can apply if the Modbus tasks are often suspended by other high priority tasks that cause processing latency and similar issues.
2. Master request is sent.
3. Response timeout occurred (measured by timer group) and Runsema is released after handling of EV_MASTER_ERROR_PROCESS event (vMBMasterRunResRelease()).
4. Master starts next transaction (request to read/write next set of registers is sent). 
5. The response from previous transaction is received within scope of new transaction we can not reliably recognize if the frame is correct.
6. The packet does not contain any ID field and only CRC field, address + function can be checked. So, the stack interprets the frame as response to new transaction. 
7. FSM starts to handle the received buffer (EV_MASTER_EXECUTE event) which is response to previous command. 
8. FSM executes command handler function to transfer received data buffer into parameter and sends additional FSM Handling events (EV_MASTER_ERROR_RECEIVE_DATA, EV_MASTER_ERROR_EXECUTE_FUNCTION ...) based on result of handling.
As a result the parameter buffer has incorrect data (the data of previous parameter if the handling completed successfully).
9. FSM releases the RunSema to start new transaction on the bus but response to previous request is received.
Repeat the cycle from number 5. If the CRC check, address is the same we repeat the cycle with data shifted between transactions or receive corresponded errors if check not pass.
[... 17 lines cut ...]
E (1999) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x107) (ESP_ERR_TIMEOUT).
02 20 I (2007) MB_PORT_COMMON: Transaction end(err reason): 2.   
[... cut ...]
The below recommendations can help to avoid communication issues:
1. Set the affinity option for Modbus tasks to avoid suspending of them by higher priority tasks often and cause processing latency. 
2. Set the CONFIG_FMB_MASTER_TIMEOUT_MS_RESPOND to maximum slave response time.
3. Retry the request if the slave responded with error or timeout.

The above recommendations allow avoiding errors. 

While the Modbus fixes are not merged I can create the accumulated patch to fix the known issues for the IDF version you are using.

alisitsyn, 2021-04-23:
Update: in your particular application the error is happend in this sequence: cmd 4 @10@8@2@20@2@4@20        // 正常
``` log
Sequence 1: @10@8@2@20 - @transmit_request@sent_request@receive_frame@handle_error(slave responded with incorrect crc or incorrect address EV_MASTER_ERROR_PROCESS -> EV_ERROR_RECEIVE_DATA).
```
```
Sequence 2: @2@4@20 - @receive_frame(good crc and address)@execute_function_handler@handle_error(EV_MASTER_ERROR_PROCESS -> EV_ERROR_OK)
```
The situaltion here is little bit different but casuses the same issue I was describing above. The stack receive `Sequence 2` without previous request that means outside of transaction time frame because the error was handled for command with incorrect crc in `Sequence 1`. 
I have the fix for this issue but the current sequence will be handled as error because of error in `Sequence_2` but next sequences 
```
Sequence_3: cmd 1 @10@8                    // 错误
cmd 2 @2@4@20@10@8             // 错误
cmd 3 @2@4@20@10               // 错误
cmd 4 @8@2@4@20@10@8@2@4@20    // 错误
```
will be handled without error.
Please send me your extended log to get more information about this error and let me know your ESP_IDF version.

alisitsyn, 2021-04-30:
@hnhkj , 

Could you give me the IDF version you are using? I have a fix for the issue described above.
Thanks.

Alvin1Zhang, 2021-05-06:
@hnhkj Thanks for reporting, would you please help share the IDF version you are using? Thanks.

shiweichn, 2021-05-13:
> @hnhkj ,
> 
> Could you give me the IDF version you are using? I have a fix for the issue described above.
> Thanks.

Hello, the 4.1 version I am using has this problem, can you create a patch for release/v4.2?

alisitsyn, 2021-05-17:
Hello @shiweichn,

The patch for esp-idf v4.2 is below. It contains set of most important fixes (not all for now) and should be able to solve most of the issues including the once described above. Please checkout the release/v4.2 and appy the patch using the `git apply 0001_accumulated_fix_set_v42.diff` command. Please set the slave response time `FMB_MASTER_TIMEOUT_MS_RESPOND` accordingly and use the `FMB_PORT_TASK_AFFINITY` to start Modbus tasks on other CPU if you have other higher priority tasks that may prevent Modbus to work properly. 

[0001_accumulated_fix_set_v42.diff.log](https://github.com/espressif/esp-idf/files/6495047/0001_accumulated_fix_set_v42.diff.log)
Let me know if you have any issues.

Thanks.

shiweichn, 2021-05-19:
> Hello @shiweichn,
>
> The patch for esp-idf v4.2 is below. It contains set of most important fixes (not all for now) and should be able to solve most of the issues including the once described above. [...]

Thanks, When will these changes be merged into v4.2 ?

alisitsyn, 2021-05-19:
Unfortunately, I can not provide expectations on when it will be merged. The priority is shifted to support of functionality for new targets first.

shiweichn, 2021-12-29:
> Hello @shiweichn,
>
> The patch for esp-idf v4.2 is below. It contains set of most important fixes (not all for now) and should be able to solve most of the issues including the once described above. [...]

Hi, Is there any plan to merge this fix into the 4.3.x  version?

shiweichn, 2021-12-29:
> Hello @shiweichn,
>
> The patch for esp-idf v4.2 is below. It contains set of most important fixes (not all for now) and should be able to solve most of the issues including the once described above. [...]

Is this problem related to the [#6134](https://github.com/espressif/esp-idf/issues/6134) problem? The version I use is esp-idf-v4.3.1. I encountered these two problems during the test. Which patch file should I use to solve these two problems?

alisitsyn, 2022-01-10:
Hi @shiweichn,

Sorry for the late response.
The patch above should fix the issue in #6134. I am going to port the patch to v4.3.1 and verify it. I will keep you posted about the results.

zhangxujun16, 2022-09-07:
Environment
Development Kit: [ESP32-Wrover-Kit]
Kit version (for WroverKit/PicoKit/DevKitC): [v4.4]
Module or chip used: [ESP32-WROVER-B]
IDF version: [v4.4]

Modbus master works well for several hours at first, but suddenly an error information happan, E (1767200) MB_PORT_COMMON: vMBMasterRunResRelease(194): Resource release failure. E (1767200) MB_CONTROLLER_MASTER: mbc_master_get_parameter(83): Master get parameter failure, error=(0x108) (ESP_ERR_INVALID_RESPONS , it lasts telling this error info, and never recover, unless reset esp32 CPU . If reset esp32 CPU , it will also work well for several hours but then such error happens again.

I think the bug is improved and patch merge into V4.4. But in fact the bug maybe still exist.

Alvin1Zhang, 2022-09-07:
> Environment Development Kit: [ESP32-Wrover-Kit] Kit version (for WroverKit/PicoKit/DevKitC): [v4.4] Module or chip used: [ESP32-WROVER-B] IDF version: [v4.4]
>
> Modbus master works well for several [...]

Thanks for sharing, seems you already created a new issue at https://github.com/espressif/esp-idf/issues/9737, thanks.

alisitsyn, 2023-12-11:
The issue is obsolete and fixed in the latest version of esp-modbus component.

closed 2023-12-11 by alisitsyn