espressif/esp-idf #6894: [BLE5] esp_ble_gap_start_ext_scan params

chegewara, 2021-04-17:
- esp32 C3 rev 2
- arduino/master

I am having weird experience with `esp_ble_gap_start_ext_scan` and i would eventually like to update info in docs if it is not a bug:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/api-reference/bluetooth/esp_gap_ble.html#_CPPv426esp_ble_gap_start_ext_scan8uint32_t8uint16_t

The weirdness that first params, which is `duration` is in `n * 10ms` and `period` is in `n * 1s`. Is that correct observation?

Thanks

chegewara, 2021-04-17:
In addition with some parameters i am getting errors trying to start ext scan:
```
E (2771) BT_BTM: LE ES En=1: cmd err=0x12
E (2771) BT_HCI: CC evt: op=0x2042, status=0x12
[  2618][E]scan start failed, error status = 7
```
- duration - 500 (5 seconds?)
- period - 3 (3 seconds?)

This may lead to conclusion that `period >= duration`. Also would be good to comment in documentation.

Thanks

Weijian-Espressif, 2021-04-19:
Duration:
Value             | Parameter Description

0x0000         | Scan continuously until explicitly disable
N = 0xXXXX | Scan durationRange: 0x0001 – 0xFFFFTime = N * 10 msTime Range: 10 ms to 655.35 s

Period:
Value            | Parameter Description
0x0000         | Periodic scanning disabled
N = 0xXXXX | Time interval from when the Controller started its last Scan_Duration until it begins the subsequent Scan_Duration.Range: 0x0001 – 0xFFFFTime = N * 1.28 secTime Range: 1.28 s to 83,884.8 s

If the Duration parameter is zero or both the Duration parameter and Period parameter are non-zero, the Controller shall continue scanning until scanning is disabled by the Host issuing an LE_Set_Extended_Scan_Enable command with the Enable parameter set to 0x00 (Scanning is disabled). The Period parameter is ignored when the Duration parameter is zero.

If the Duration parameter is non-zero and the Period parameter is zero, the Controller shall continue scanning until the duration specified in the Duration parameter has expired.

If both the Duration and Period parameters are non-zero and the Duration parameter is greater than or equal to the Period parameter, the Controller shall return the error code Invalid HCI Command Parameters (0x12).
When the Duration and Period parameters are non-zero, the Controller shall scan for the duration of the Duration parameter within a scan period specified by the Period parameter. After the scan period has expired, a new scan period shall begin and scanning shall begin again for the duration specified. The scan periods continue until the Host disables scanning.

chegewara, 2021-04-19:
Thanks for extensive explanation.
I think this issue can be closed, but still believe it would be nice to have at least basic info about params in documentation.

Thanks

chegewara, 2021-04-19:
@Weijian-Espressif 
I have one more question, about `esp_ble_gap_periodic_adv_set_params`, but i dont want to open new issue for that simple question.
```
static esp_ble_gap_periodic_adv_params_t periodic_adv_params = {
    .interval_min = 0x320, // 1000 ms interval
    .interval_max = 0x640, // 2000 ms
    .properties = 0, // Do not include TX power
};
```
The result is that advertising is exactly every 2000ms, at least periodic_sync_demo is reporting every 2000ms.
Is that correct that only `interval_max` is used and `interval_min` is ignored?

Weijian-Espressif, 2021-04-20:
@chegewara You are right, the controller always uses `interval_max`.

closed 2021-06-01 by Alvin1Zhang