espressif/esp-idf #6896: ESP-IDF 4.0 compilation terminated!

MRxing1, 2021-04-18:
Compiling the micropython firmware, and finally this error occurred！

OPY_ESP_IDF_4=1 -DMICROPY_ROM_TEXT_COMPRESSION=1 -c -MD -o build-GENERIC_SPIRAM//home/xkp/esp32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.o /home/xkp/esp32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c
/home/xkp/esp32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.c:39:21: fatal error: tcp_isn.h: No such file or directory
compilation terminated.
../../py/mkrules.mk:63: recipe for target 'build-GENERIC_SPIRAM//home/xkp/esp32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.o' failed
make: *** [build-GENERIC_SPIRAM//home/xkp/esp32/esp-idf/components/tcpip_adapter/tcpip_adapter_lwip.o] Error 1

Best wishes!

MRxing1, 2021-04-18:
But actually, I found that there is this file by looking it carefully. Why is the system always prompted to find it? This is something I really don't understand.

[image]

[image]

MRxing1, 2021-04-20:
[image]
How to add a path to this file? Please guide, thank you very much, best wishes!!!

MRxing1, 2021-04-20:
@igrr  Can you give me some guidance, thank you very much！

igrr, 2021-04-20:
@MRxing1 Could you please add the information requested in the issue template when you were opening the issue? In particular, commit IDs of ESP-IDF (and Micropython, since you're building that project), steps to reproduce the issue, and the build log. Please attach logs in text format or upload them to gist.github.com and post a link.

The path to tcp_isn directory in lwip port layer is already added in lwip component CMakeLists.txt:
https://github.com/espressif/esp-idf/blob/41efdb0b34864d434ce144e8530a4a78dfd7cd9f/components/lwip/CMakeLists.txt#L7

MRxing1, 2021-04-21:
OK, I'll pay attention to it next time. Thank you very much!

MRxing1, 2021-04-21:
@igrr 
I'll give you feedback on this. The following contents are added in micropypthon / ports / esp32 / makefile:

NC_ ESPCOMP += -I$(ESPCOMP)/lwip/port/esp32/tcp_ isn

After that, the compiler passed

MRxing1, 2021-04-21:
@igrr
Thank you again！！

igrr, 2021-04-23:
@MRxing1 good to know you have solved the problem.
FYI, Micrypython has recently started using CMake build system for the ESP32 chip (https://github.com/micropython/micropython/pull/6892) and now the list of source files and include directories is no longer hard-coded in micropython port makefiles. This makes it easier to switch between IDF versions in Micropython. This feature is part of micropython v1.15 release.

closed 2021-04-23 by igrr