espressif/esp-idf #6897: How do I use UART0 for debug logs and UART1 for CLI Console?

Darsh-Dev, 2021-04-19:
Hello,

In my project I want to use CLI Console provided in esp-idf demo examples.
It works fine I can run it on both the uarts UART0 as well as UART1. I can successfully change the console uart from menuconfig.
what I require is to use CLI console on UART1 and debug logs on UART0 and I'm unable to make it work. currently, I am getting both the CLI console and debug logs on same UART.
I don't know if this feature is already available or not.

Kindly, please help!!

projectgus, 2021-04-20:
Apologies for all the spammy messages, our syncing system created two JIRA tickets from this GitHub issue due to a bug.

Changing the console settings in menuconfig will change bootloader and logging output by default, as these are all routed to the same console output.

If you're including `esp_console.h` and using the API there to create a custom command line console for your project, then you can override the UART when creating the console.

```c
    esp_console_repl_t *repl = NULL;
    esp_console_repl_config_t repl_config = ESP_CONSOLE_REPL_CONFIG_DEFAULT();
    esp_console_dev_uart_config_t uart_config = {
        .channel = 1, // use UART1
        .baud_rate = CONFIG_ESP_CONSOLE_UART_BAUDRATE, // can also set a different baud rate if you want
        .tx_gpio_num = -1, // -1 will use the default UART1 GPIOs, but can use any pins
        .rx_gpio_num = -1,
    };
    ESP_ERROR_CHECK(esp_console_new_repl_uart(&uart_config, &repl_config, &repl));
```

The command REPL `repl` will use this UART config. See examples like [i2c_tools](https://github.com/espressif/esp-idf/tree/master/examples/peripherals/i2c/i2c_tools) for example code working with this interface.

However, standard output (i.e. printf) will still use the default standard output which is the console configured in menuconfig. If you want to split logs and "printf" from going to the same place then look at the `esp_log_set_vprintf` function which lets you point logging at a different destination.

BTW, the best place to ask this kind of "How do I..." question in the future is the https://esp32.com forums.)

Darsh-Dev, 2021-04-20:
@projectgus Thank you for the suggestion.
made the changes as you suggested in the **i2c_tools** example. still I'm not getting anything on UART1.  Whereas on UART0 I'm getting following logs.
[image]

o-marshmallow, 2021-05-19:
Hello @Darsh-Dev ,
We are currently checking this issue, which version of ESP-IDF are you using?

Darsh-Dev, 2021-05-19:
Hello @o-marshmallow,
 
Currently I'm using ESP-IDF v4.3-beta2.

Thank You!!

closed 2021-06-23 by Darsh-Dev