espressif/esp-idf #6899: idf.py create-project broken on ESP-IDF v4.2?

trusleymike, 2021-04-19:
Windows 10: IDF version:
----------------------------

  C:\Users\Mike\Desktop\esp-idf\examples>idf.py --version
  ESP-IDF v4.2-dirty

Asking for help with create-project (no options displayed)
----------------------------------------------------------------

  C:\Users\Mike\Desktop\esp-idf\examples>idf.py create-project --help
  Usage: idf.py create-project [OPTIONS]

     Execute targets that are not explicitly known to idf.py

  Options:
    --help  Show this message and exit.

Trying to create a new project under the examples folder:
---------------------------------------------------------------

  C:\Users\Mike\Desktop\esp-idf\examples>idf.py create-project mike
  Executing action: create-project
  CMakeLists.txt not found in project directory c:\users\mike\desktop\esp-idf\examples

Trying to create a new project under the idf folder:
--------------------------------------------------------

=====

C:\Users\Mike\Desktop\esp-idf>idf.py create-project mike
Executing action: create-project
Running cmake in directory c:\users\mike\desktop\esp-idf\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 c:\users\mike\desktop\esp-idf"...
-- The C compiler identification is unknown
-- The CXX compiler identification is unknown
-- The ASM compiler identification is unknown
-- Didn't find assembler
CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_C_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CC" or the CMake cache entry CMAKE_C_COMPILER to the full path to
  the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_CXX_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "CXX" or the CMake cache entry CMAKE_CXX_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

CMake Error at CMakeLists.txt:2 (project):
  No CMAKE_ASM_COMPILER could be found.

  Tell CMake where to find the compiler by setting either the environment
  variable "ASM" or the CMake cache entry CMAKE_ASM_COMPILER to the full path
  to the compiler, or to the compiler name if it is in the PATH.

-- Warning: Did not find file Compiler/-ASM
-- Configuring incomplete, errors occurred!
See also "C:/Users/Mike/Desktop/esp-idf/build/CMakeFiles/CMakeOutput.log".
See also "C:/Users/Mike/Desktop/esp-idf/build/CMakeFiles/CMakeError.log".
cmake failed with exit code 1

=====

dobairoland, 2021-04-19:
Hi @trusleymike. `idf.py create-project` is not available in v4.2.

I assume that you are reading the documentation for later ESP-IDF versions, i.e. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#start-a-new-project. You can notice that this subsection is missing from the documentation of [v4.2](https://docs.espressif.com/projects/esp-idf/en/v4.2/esp32/api-guides/build-system.html).

You can copy some of the simplest projects from https://github.com/espressif/esp-idf/tree/master/examples/get-started to start a new project.

trusleymike, 2021-04-19:
Thank you.  I am reading the latest documentation on the web site

However, I quote from https://github.com/espressif/esp-idf/releases:

  "ESP-IDF v3.3.5 is a bugfix update for ESP-IDF v3.3. IDF v4.2 is the latest stable release of ESP-IDF at time of writing."

Last week I installed via the Windows ESP-IDF Tools Installer and it displays ESP-IDF v4.2-dirty.

One of my colleagues has an older version and create-project works fine.

I am confused.  Why are there two version numbers: ESP-IDF v3.3 and IDF v4.2.  Are they different things?

Has create-project been dropped?  If not, should I uninstall and use Windows ESP-IDF Tools Installer to install again?

dobairoland, 2021-04-19:
The "latest" documentation is for the ESP-IDF "master" branch which is currently v4.4 and is under development. 

`idf.py create-project` is a new feature available since v4.3. 

You can read more about how versioning is done here: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/versions.html

closed 2021-04-19 by dobairoland