espressif/esp-idf #6900: stuck at E (234871) wifi:ieee80211_ioctl.c 1443

thedayu, 2021-04-19:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Module or chip used:  [ESP32]
- IDF version (run ``git describe --tags`` to find it):
- ESP-IDF v4.3-beta2-2-g9a2d25191-dirty
- Build System:         vscode extension 1.0.2, config with tag release/v4.3
- Operating System:     [Windows]
- Using an IDE: vscode
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description
project is building with UNICORE config, and it is basically WIFI example project , and it randomly stucks with log message:
(not often, somethings it runs days without any problem.)

```

E (234871) wifi:ieee80211_ioctl.c 1443

E (239851) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (239851) task_wdt:  - IDLE (CPU 0)
E (239851) task_wdt: Tasks currently running:
E (239851) task_wdt: CPU 0: wifi
E (239851) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4016C80A:0x3FFBE830 0x40082826:0x3FFBE850 0x40096DE2:0x3FFDDAA0 0x4009A8D6:0x3FFDDAC0 0x4009ADBC:0x3FFDDAF0 0x401DA7AE:0x3FFDDB10 0x401DA80E:0x3FFDDB30 0x40098A9A:0x3FFDDB50 0x40093541:0x3FFDDB80
0x4016c80a: task_wdt_isr at C:/Users/admin/esp/v4.3/esp-idf/components/esp_common/src/task_wdt.c:189

0x40082826: _xt_lowint1 at C:/Users/admin/esp/v4.3/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40096de2: wifi_rf_phy_enable at ??:?

0x4009a8d6: pm_dream at ??:?

0x4009adbc: pm_tbtt_process at ??:?

0x401da7ae: pm_update_next_tbtt at ??:?

0x401da80e: pp_timer_do_process at ??:?

0x40098a9a: ppTask at ??:?

0x40093541: vPortTaskWrapper at C:/Users/admin/esp/v4.3/esp-idf/components/freertos/port/xtensa/port.c:168

E (244851) task_wdt: Task watchdog got triggered. The following tasks did not reset the watchdog in time:
E (244851) task_wdt:  - IDLE (CPU 0)
E (244851) task_wdt: Tasks currently running:
E (244851) task_wdt: CPU 0: wifi
E (244851) task_wdt: Print CPU 0 (current core) backtrace

Backtrace:0x4016C80A:0x3FFBE830 0x40082826:0x3FFBE850 0x40096DE2:0x3FFDDAA0 0x4009A8D6:0x3FFDDAC0 0x4009ADBC:0x3FFDDAF0 0x401DA7AE:0x3FFDDB10 0x401DA80E:0x3FFDDB30 0x40098A9A:0x3FFDDB50 0x40093541:0x3FFDDB80
0x4016c80a: task_wdt_isr at C:/Users/admin/esp/v4.3/esp-idf/components/esp_common/src/task_wdt.c:189

0x40082826: _xt_lowint1 at C:/Users/admin/esp/v4.3/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:1105

0x40096de2: wifi_rf_phy_enable at ??:?

0x4009a8d6: pm_dream at ??:?

0x4009adbc: pm_tbtt_process at ??:?

0x401da7ae: pm_update_next_tbtt at ??:?

0x401da80e: pp_timer_do_process at ??:?

0x40098a9a: ppTask at ??:?

```
### Expected Behavior
do anything but don't stuck.  

### Steps to reproduce
seems random... no idea how to reproduce. 
connect to wifi, connect to tcp, send few bytes every 30 seconds to stay alive, and wait, (could be days).

AxelLin, 2021-04-19:
This is reported https://github.com/espressif/esp-idf/issues/6744#issuecomment-803760126
The fix is already in master, for v4.3 branch, needs to wait for the fix. 
(The wifi-lib v4.3 branch https://github.com/espressif/esp32-wifi-lib/tree/release/v4.3 includes the fix, maybe you can manually update submodule and test it.)

thedayu, 2021-04-20:
thanks a lot. I will update and test, though it's hard to reproduce on my devices.

Wasabinary, 2021-07-28:
Same problem here @thedayu. What's the ESP-IDF version to use to avoid this problem ?

Espressif-liuuuu, 2021-08-04:
> Same problem here @thedayu. What's the ESP-IDF version to use to avoid this problem ?

Hi, you could update your version to the newest v4.3 to avoid this problem.

Alvin1Zhang, 2021-08-09:
@thedayu Thanks for reporting, would you please help share if any updates for this issue? Thanks.

MihaelIsaev, 2021-08-21:
I just updated to v4.3 and don't see `wifi:ieee80211_ioctl.c 1443` anymore. Before that I seen it during OTA 14 times of 15 attempts. Now 0 times, so I hope it is gone.

Alvin1Zhang, 2021-08-23:
@MihaelIsaev Thanks for sharing the updates. @thedayu Do you have some updates regarding the testing? Thanks.

Alvin1Zhang, 2021-09-13:
@thedayu Thanks for reporting, do you have some updates regarding the testing? Thanks.

closed 2021-10-13 by Alvin1Zhang