espressif/esp-idf #6903: ESP32-C3FN4芯片启动WiFi连接会导致程序崩溃

liangzc1, 2021-04-20:
使用release/v4.3版本SDK的examples/wifi/getting_started/station工程，配置连接自己的路由器，连接过程中必现程序崩溃重启。硬件为开发板ESP32-C3-DevkitM-1，芯片为ESP32-C3FN4。若将该开发板芯片替换成ESP32-C3FH4则没有此问题。log如下
Leaving...
Hard resetting via RTS pin...
Executing action: monitor
Running idf_monitor in directory i:\esp\v4.3\esp-idf\examples\wifi\getting_started\station
Executing "I:\ESP\tools\4.3\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe I:\ESP\v4.3\esp-idf\tools/idf_monitor.py -p com13 -b 115200 --toolchain-prefix riscv32-esp-elf- --decode-panic backtrace --target esp32c3 i:\esp\v4.3\esp-idf\examples\wifi\getting_started\station\build\wifi_station.elf -m 'I:\ESP\tools\4.3\.espressif\python_env\idf4.3_py3.8_env\Scripts\python.exe' 'I:\ESP\v4.3\esp-idf\tools\idf.py' '-p' 'com13'"...
--- idf_monitor on com13 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1738
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2bf8
entry 0x403ce000
I (52) boot: ESP-IDF v4.3-beta2-2-g9a2d25191-dirty 2nd stage bootloader
I (52) boot: compile time 10:33:13
I (53) boot: chip revision: 3
I (56) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (63) boot.esp32c3: SPI Speed      : 80MHz
I (68) boot.esp32c3: SPI Mode       : DIO
I (73) boot.esp32c3: SPI Flash Size : 2MB
I (77) boot: Enabling RNG early entropy source...
I (83) boot: Partition Table:
I (86) boot: ## Label            Usage          Type ST Offset   Length
I (94) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (101) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (109) boot:  2 factory          factory app      00 00 00010000 00100000
I (116) boot: End of partition table
I (120) boot_comm: chip revision: 3, min. application chip revision: 0
I (128) esp_image: segment 0: paddr=00010020 vaddr=3c090020 size=15f28h ( 89896) map
I (150) esp_image: segment 1: paddr=00025f50 vaddr=3fc8de00 size=032f8h ( 13048) load
I (153) esp_image: segment 2: paddr=00029250 vaddr=40380000 size=06dc8h ( 28104) load
I (161) esp_image: segment 3: paddr=00030020 vaddr=42000020 size=83f24h (540452) map
I (249) esp_image: segment 4: paddr=000b3f4c vaddr=40386dc8 size=06f00h ( 28416) load
I (260) boot: Loaded app from partition at offset 0x10000
I (260) boot: Disabling RNG early entropy source...
I (271) cpu_start: Pro cpu up.
I (328) cpu_start: Pro cpu start user code
I (328) cpu_start: cpu freq: 160000000
I (328) cpu_start: Application information:
I (331) cpu_start: Project name:     wifi_station
I (336) cpu_start: App version:      v4.3-beta2-2-g9a2d25191-dirty
I (343) cpu_start: Compile time:     Apr 20 2021 10:33:00
I (349) cpu_start: ELF file SHA256:  79a7251740558b80...
I (355) cpu_start: ESP-IDF:          v4.3-beta2-2-g9a2d25191-dirty
I (362) heap_init: Initializing. RAM available for dynamic allocation:
I (369) heap_init: At 3FC95CC0 len 0002A340 (168 KiB): DRAM
I (375) heap_init: At 3FCC0000 len 0001F060 (124 KiB): STACK/DRAM
I (382) heap_init: At 50000000 len 00002000 (8 KiB): FAKEDRAM
I (389) spi_flash: detected chip: generic
I (393) spi_flash: flash io: dio
W (397) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (411) cpu_start: Starting scheduler.
I (420) wifi station: ESP_WIFI_MODE_STA
I (420) pp: pp rom version: 9387209
I (420) net80211: net80211 rom version: 9387209
I (440) wifi:wifi driver task: 3fc9f2cc, prio:23, stack:6656, core=0
I (440) system_api: Base MAC address is not set
I (440) system_api: read default base MAC address from EFUSE
I (450) wifi:wifi firmware version: bb5a818
I (450) wifi:wifi certification version: v7.0
I (450) wifi:config NVS flash: enabled
I (460) wifi:config nano formating: disabled
I (460) wifi:Init data frame dynamic rx buffer num: 32
I (470) wifi:Init management frame dynamic rx buffer num: 32
I (470) wifi:Init management short buffer num: 32
I (480) wifi:Init dynamic tx buffer num: 32
[... 84 lines cut ...]
3fc9f510: 0x00000000 0x2e617473 0x68747561 0x65646f6d 0x00000000 0x2e617473 0x00240701 0x00000000

ELF file SHA256: 79a7251740558b80
ELF file SHA256: 79a7251740558b80

Rebooting...
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
Saved PC:0x40382e56
0x40382e56: esp_restart_noos at I:/ESP/v4.3/esp-idf/components/esp32c3/system_api_esp32c3.c:137 (discriminator 1)

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd6100,len:0x1738
load:0x403ce000,len:0x894
load:0x403d0000,len:0x2bf8
entry 0x403ce000
I (57) boot: ESP-IDF v4.3-beta2-2-g9a2d25191-dirty 2nd stage bootloader
I (57) boot: compile time 10:33:13
I (57) boot: chip revision: 3
I (61) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (68) boot.esp32c3: SPI Speed      : 80MHz
I (73) boot.esp32c3: SPI Mode       : DIO
I (78) boot.esp32c3: SPI Flash Size : 2MB

TianaESP, 2021-04-20:
Hi @liangzc1,

您好，当前 ESP32-C3FN4 应暂未公开送样，能否烦请您帮忙查看开发板上模组丝印，确认有 H4，N4 的标示。针对出问题的模块，请尝试擦掉 flash，然后将 SDK 中的 build 文件全删除，再烧录固件连路由。

期待您的反馈，谢谢。
Tiana

liangzc1, 2021-04-20:
目前我司拿到的这个型号的芯片都是这个问题，不是某一个，照你说的测试了也是一样的结果

TianaESP, 2021-04-20:
@liangzc1 
是否方便将问题样品寄回给我们一些？
可以将问题反馈至我司对接您的商务。
或 将问题发送至 sales@espressif.com, 并提及该 Github 问题单号，我会和商务跟踪该问题，谢谢。
Tiana

liangzc1, 2021-04-20:
已经发送了邮件，请安排对接，对接后将对应芯片寄回贵司分析

Alvin1Zhang, 2021-04-26:
Thanks for reporting, feel free to reopen.

closed 2021-04-26 by Alvin1Zhang

trphoenix, 2021-07-03:
我也有同样的问题，也需要寄回么？

TianaESP, 2021-07-05:
@trphoenix  不需要寄回。请尝试：
1. 使用老版本的 release/v4.3 会存在这个问题，需要更新一下最新的版本。
2. 需要对新的 flash 进行一次 erase 操作。
谢谢。