espressif/esp-idf #6905: WPA2 ENT EAP example not connecting to ubiquity network. 4.1.1

Bradleyking4, 2021-04-20:
## Environment

- Development Kit:      [none]
- Module or chip used:  [ESP32-WROOM-32]
- IDF version v4.1 - > v4.1.1
- Build System:      [Make]
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r2) 8.2.0
- Operating System:   [Linux]
- Using an IDE?: [No]
- Power Supply:  [USB]

## Problem Description
After updating to v4.1.1 from v4.1 WPA ENT PEAP no longer connects to the access point,  Using either the examples program or ours. 

### Expected Behavior

```[0;32mI (2965) wpa: EAP-TLV: TLV Result - Success - EAP-TLV/Phase2 Completed[0m
[0;32mI (2975) wpa: >>>>>wpa2 FINISH
[0m
I (2995) wifi:connected with wpawep, aid = 2, channel 6, BW20, bssid = 3a:e8:29:5a:5a:e5
I (2995) wifi:security: WPA2-Ent, phy: bgn, rssi: -56
I (2995) wifi:pm start, type: 1

I (3045) wifi:AP's beacon interval = 102400 us, DTIM period = 1
[0;32mI (4965) example: ~~~~~~~~~~~[0m
[0;32mI (4965) example: IP:0.0.0.0[0m
[0;32mI (4965) example: MASK:0.0.0.0[0m
[0;32mI (4965) example: GW:0.0.0.0[0m
[0;32mI (4965) example: ~~~~~~~~~~~[0m
[0;32mI (5125) esp_netif_handlers: sta ip: 192.168.1.32, mask: 255.255.255.0, gw: 192.168.1.1[0m
[0;32mI (6965) example: ~~~~~~~~~~~[0m
[0;32mI (6965) example: IP:192.168.1.32[0m
[0;32mI (6965) example: MASK:255.255.255.0[0m
[0;32mI (6965) example: GW:192.168.1.1[0m
[0;32mI (6965) example: ~~~~~~~~~~~[0m
[0;32mI (8965) example: ~~~~~~~~~~~[0m
[0;32mI (8965) example: IP:192.168.1.32[0m
[0;32mI (8965) example: MASK:255.255.255.0[0m
[0;32mI (8965) example: GW:192.168.1.1[0m
[0;32mI (8965) example: ~~~~~~~~~~~[0m```

### Actual Behavior

```I (15219) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (15219) wifi:state: init -> auth (b0)
I (15239) wifi:state: auth -> assoc (0)
I (15249) wifi:state: assoc -> run (10)
[0;32mI (16909) example: ~~~~~~~~~~~[0m
[0;32mI (16909) example: IP:0.0.0.0[0m
[0;32mI (16909) example: MASK:0.0.0.0[0m
[0;32mI (16909) example: GW:0.0.0.0[0m
[0;32mI (16909) example: ~~~~~~~~~~~[0m
I (17319) wifi:state: run -> init (17c0)
I (17329) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
I (17329) wifi:new:<6,0>, old:<6,0>, ap:<255,255>, sta:<6,0>, prof:1
[0;32mI (18909) example: ~~~~~~~~~~~[0m
[0;32mI (18909) example: IP:0.0.0.0[0m
[0;32mI (18909) example: MASK:0.0.0.0[0m
[0;32mI (18909) example: GW:0.0.0.0[0m
[0;32mI (18909) example: ~~~~~~~~~~~[0m```

### Steps to reproduce

1. open wpa enterprise example 
2. change example configs, 
```
#
# Example Configuration
#
CONFIG_EXAMPLE_WIFI_SSID="wpawep"
# CONFIG_EXAMPLE_VALIDATE_SERVER_CERT is not set
# CONFIG_EXAMPLE_EAP_METHOD_TLS is not set
CONFIG_EXAMPLE_EAP_METHOD_PEAP=y
# CONFIG_EXAMPLE_EAP_METHOD_TTLS is not set
CONFIG_EXAMPLE_EAP_METHOD=1
CONFIG_EXAMPLE_EAP_ID="example@espressif.com"
CONFIG_EXAMPLE_EAP_USERNAME="fiffy"
CONFIG_EXAMPLE_EAP_PASSWORD="fiffy123"
# end of Example Configuration
```

// If possible, attach a picture of your setup/wiring here.
No wiring needed.

### Code to reproduce this issue
WPA ENT example program

## Debug Logs
I've attached logs of the default sdk config after changing the examples for PEAP, and one with verbose logging turned on.

## Other items if possible

[default-wpa4.1.1.log](https://github.com/espressif/esp-idf/files/6341043/default-wpa4.1.1.log)
[log-wpa4.1.1.log](https://github.com/espressif/esp-idf/files/6341044/log-wpa4.1.1.log)

[default-wpa4.1.log](https://github.com/espressif/esp-idf/files/6341046/default-wpa4.1.log)
[log-wpa4.1.log](https://github.com/espressif/esp-idf/files/6341047/log-wpa4.1.log)

[sdkconfig v4.1.1.txt](https://github.com/espressif/esp-idf/files/6341052/sdkconfig.v4.1.1.txt)
[sdkconfig v4.1.txt](https://github.com/espressif/esp-idf/files/6341053/sdkconfig.v4.1.txt)

sagb2015, 2021-04-21:
Hi @Bradleyking4 , Can you please help with supplicant logs? Need to compile with CONFIG_WPA_DEBUG_PRINT=y.

sagb2015, 2021-04-21:
Hi @Bradleyking4 , Can you please help with supplicant logs in 4.1.1 case? Need to compile with CONFIG_WPA_DEBUG_PRINT=y.

Bradleyking4, 2021-04-21:
[log-wpa4.1.1.log](https://github.com/espressif/esp-idf/files/6347645/log-wpa4.1.1.log)
Here is the updated file, sorry about that.

nishanth-radja, 2021-04-22:
Hi @Bradleyking4 , can we have the radius server logs,so that we can try to reproduce the Issue.Is the Radius server configured with PEAP-mschapv2?

Bradleyking4, 2021-05-16:
Here you go, please let me know if you require anything else.   With regards to, PEAP-mschapv2 it looks like a yes.

```
 Module: Instantiating module "mschap" from file /etc/freeradius/modules/mschap
  mschap {
        use_mppe = yes
        require_encryption = no
        require_strong = no
        with_ntdomain_hack = no
        allow_retry = yes
  }

 Module: Instantiating eap-peap
   peap {
        default_eap_type = "mschapv2"
        copy_request_to_tunnel = no
        use_tunneled_reply = yes
        proxy_tunneled_request_as_eap = yes
        virtual_server = "inner-tunnel"
        soh = no
   }
 Module: Linked to sub-module rlm_eap_mschapv2
 Module: Instantiating eap-mschapv2
   mschapv2 {
        with_ntdomain_hack = no
        send_error = no
   }
```

[4.1.1-fail.txt](https://github.com/espressif/esp-idf/files/6488609/4.1.1-fail.txt)
[4.1-success.txt](https://github.com/espressif/esp-idf/files/6488610/4.1-success.txt)

nishanth-radja, 2021-05-21:
Hi @Bradleyking4 After going through the logs, we found that in the 4.1.1 case the inner identity is blank.Which is not the case in 4.1
peap] EAPTLS_OK
[peap] Session established.  Decoding tunneled attributes.
[peap] Peap state WAITING FOR INNER IDENTITY
[peap] Identity -
[peap] Got inner identity ''
 
Can we pls have the Radius Config which is being used and which version of 4.1.1 is being used on the DUT.(Commit ID)

Alvin1Zhang, 2021-06-01:
@Bradleyking4 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

timmyhadwen, 2021-09-12:
Hi @Alvin1Zhang AND @nishanth-radja. I've taken over @Bradleyking4's work and we are still having this issue with 4.3.1 as well. We note that you found that the inner identity was blank. Can you advise on how we might be able to resolve that?

AxelLin, 2021-09-13:
@timmyhadwen 

Then can you provide the Radius Config (which was asked by https://github.com/espressif/esp-idf/issues/6905#issuecomment-845639079)

nishanth-radja, 2021-09-14:
@timmyhadwen Also can you also provide the radius server logs similar to the [https://github.com/espressif/esp-idf/issues/6905#issuecomment-841757744](comment). To make sure that it is the same issue in 4.3.1 as well

timmyhadwen, 2021-10-05:
@AxelLin We are using the UniFi internal RADIUS server. Looks like its based on freeradius. Attached config file. 

[radiusd.conf.txt](https://github.com/espressif/esp-idf/files/7283838/radiusd.conf.txt)

We can confirm that it is also broken on 4.3 and also on other networks. Namely we have tried Eduroam with 4.1 (success), 4.1.1 (failure) and 4.3 (failure).

timmyhadwen, 2021-10-05:
This is what im seeing from FreeRadius logs.

Tue Oct  5 18:29:09 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:13 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:15 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:19 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:23 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:25 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass
Tue Oct  5 18:29:28 2021 : Auth: Login incorrect: [tim/<via Auth-Type = EAP>] (from client client-5ef159a650b5653f9556baa0 port 0 cli 94-3C-C6-7D-34-E0) badpass

AxelLin, 2021-11-13:
@timmyhadwen Any update regarding comment https://github.com/espressif/esp-idf/issues/6905#issuecomment-938525075 ?

timmyhadwen, 2022-01-19:
@AxelLin @nishanth-radja Please find the files as requested and let me know of any further files you need. We are still having the problem on Eduroam networks as well, however we are unable to get the files from their config. Errors still persist on our test net.

[clients.conf.txt](https://github.com/espressif/esp-idf/files/7893448/clients.conf.txt)
[eap.conf.txt](https://github.com/espressif/esp-idf/files/7893449/eap.conf.txt)
[users.txt](https://github.com/espressif/esp-idf/files/7893450/users.txt)

nishanth-radja, 2022-01-20:
@timmyhadwen Thanks for the conf files,Using the eap.conf,I am trying to repro the issue locally.Can you point me to the IDF commit you are using.
I have tried on the latest 4.1.1 and 4.3 but did not see the issue. 
I see that in the eap.conf file "default_eap_type = md5" but the md5 is blank in the conf file and it connects using the peap eventually after NAK from STA.
can you also try changing it to "default_eap_type = peap" and try.
Can you pls share your commit ID?

timmyhadwen, 2022-02-14:
HI All, this seems to be resolved on latest master.

AxelLin, 2022-02-14:
> HI All, this seems to be resolved on latest master.

Does it work for you with stable branches?

AxelLin, 2022-05-11:
> HI All, this seems to be resolved on latest master.

@timmyhadwen 
Is this issue still valid or should be closed?
BTW, just wondering if it works for you on stable branches, e.g. v4.3.2 or v4.4.1?

Alvin1Zhang, 2022-05-11:
Thanks for reporting, sorry for slow turnaround, the fix on master branch is available at https://github.com/espressif/esp-idf/commit/190b31bb1b9b1e6e21feb038797aed6e14cb765b, thanks.

AxelLin, 2022-05-11:
> Thanks for reporting, sorry for slow turnaround, the fix on master branch is available at [190b31b](https://github.com/espressif/esp-idf/commit/190b31bb1b9b1e6e21feb038797aed6e14cb765b), thanks.

Above fix only available on master.
That fix was committed on Jan 12 which is several months ago, why stable branches do not have the fix?

timmyhadwen, 2022-05-20:
Thanks @Alvin1Zhang

AxelLin, 2022-06-06:
> Thanks for reporting, sorry for slow turnaround, the fix on master branch is available at [190b31b](https://github.com/espressif/esp-idf/commit/190b31bb1b9b1e6e21feb038797aed6e14cb765b), thanks.

@Alvin1Zhang 
This issue was reported on  v4.1.1, so I suppose stable branches need backport fix.
Could you confirm if stable branches need this fix? (If yes, why it takes such long time for backport fix?)

AxelLin, 2022-07-12:
@Alvin1Zhang 
The fix is in v4.3 branch but not available in v4.4 branch.

Alvin1Zhang, 2022-07-25:
Thanks for reporting, sorry for slow turnaround, fixes on release branches are

* release/4.4: https://github.com/espressif/esp-idf/commit/4908c4e27d56a0103dfdfad33a90114162f5f45a
* release/4.3: https://github.com/espressif/esp-idf/commit/e2ebb90df33ecd873c5548fe1ca11f4b68090242
* release/4.2: https://github.com/espressif/esp-idf/commit/8f2659e2ad2713b1f5dab8c115e000932a760420
* release/4.1: https://github.com/espressif/esp-idf/commit/ee50721f814b4601b898c9212b426181cd879250

Feel free to reopen.

closed 2022-07-25 by Alvin1Zhang