espressif/esp-idf #6906: Esp32c3 `int32_t` should be `int` and not `long int`

s-hadinger, 2021-04-20:
## Environment

- Development Kit:      [ESP32-C3-DevKitM-1]
- Module or chip used:  [ESP32C3]
- IDF version (run ``git describe --tags`` to find it): v3.40200.210118
- Build System:         [Platform.io]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): riscv32-esp-elf-c++ (crosstool-NG 1.24.0.123_64eb9ff) 8.4.0
- Operating System:     [macOS]
- Using an IDE?: [Yes, Platform.io]
- Power Supply:         [USB]

## Problem Description

When using ESP32 or ESP8266, `uint32_t` matches `unsigned int` and `int32_t` matches `int`.

But with RISC-V compiler the `uint32_t` now matches `unsigned long int` and `int32_t` matches `long int`.

This causes compilation failure of multiple Arduino libs, specifically in Tasmota.

A workaround is to add `-D__INT32_TYPE__=int -D__UINT32_TYPE__="unsigned int"` in the `build_flags` but it creates lots of warnings.

The core issue is that gcc builtins for `__INT32_TYPE__` is `long int` and `__UINT32_TYPE__` is `unsigned long int`.

I suggest to change the builtins in gcc.

### Expected Behavior

Same behavior as Xtensa based chips.

### Actual Behavior

Compilation error in Tasmota.

### Steps to reproduce

### Code to reproduce this issue

Example of code that compiles well on Esp8266/ESP32 but fails on RISCV EPS32C3

```cpp
#include <cstdint>

void test(int *a) {
  // do nothing
}

void setup() {
  int32_t i;
  test(&i);
}

void loop() {
}
```

## Debug Logs

Error when compiling with ESP32C3

```
Compiling .pio/build/tasmota32c3/src/main.o
src/main.cpp: In function 'void setup()':
src/main.cpp:10:8: error: invalid conversion from 'int32_t*' {aka 'long int*'} to 'int*' [-fpermissive]
   test(&i);
        ^~
src/main.cpp:4:16: note:   initializing argument 1 of 'void test(int*)'
 void test(int *a) {
           ~~~~~^
*** [.pio/build/tasmota32c3/src/main.o] Error 1
```

igrr, 2021-04-20:
Hi @s-hadinger, here we are following the defaults adopted by the upstream RISC-V toolchain. If there is a particular place in ESP-IDF APIs where the difference between these types is problematic, please let us know and we will address this.

s-hadinger, 2021-04-20:
Thanks. The problem comes from the numerous Arduino libraries that take as granted that `int32_t` is `int` and not `long int`. Esp-idf does not have this problem per se since it does not mix both types.

Is there a way to change this behavior in the upstream RISC-C toolchain?

This could be done either by changing the builtin defines for `__INT32_TYPE__` and `__UINT32_TYPE__` or by changing the definition of `__int32` in `toolchain-riscv32/riscv32-esp-elf/include/machine/_default_types.h`

igrr, 2021-04-20:
The change is actually pretty simple, `STDINT_LONG32` needs to be set to 0 in the target configuration file. Here is how it's done for Xtensa:
https://github.com/espressif/gcc/blob/d437985f6c8556a96ebb1eecb423bfd997f64b0c/gcc/config/xtensa/xtensa.h#L132-L134
Similar could be done for RISC-V: https://github.com/espressif/gcc/blob/esp_based_on_8_4_0/gcc/config/riscv/riscv.h.

(_default_types.h and other files will be generated during GCC build process)

I am mainly worried that changing the size types in Espressif's toolchain builds will cause incompatibilities with upstream RISC-V toolchain later on. Besides, choice to use `unsigned long` rather than `unsigned int` for `uint32_t` is not so uncommon. I've checked arm-none-eabi-gcc and it also uses `unsigned long` for `uint32_t` type. Do any of the libraries you see this problem with support ARM based Arduino boards (e.g. SAMD based)?

s-hadinger, 2021-04-20:
Thanks for the quick answer. Most libs don't have an issue, and I understand it is not a best practice to use `int`. In Tasmota we use tens of Arduino libs, I saw the problem in `IRremoteESP8266` and I temporarily fixed in with CLI defines.

Let's not rush in changing the upstream toolchain, I found a workaround for Tasmota by forcing an include file via CLI.

https://github.com/arendst/Tasmota/pull/11807

The workaround for Tasmota is clean enough, but I'm worried that some other people may have the same issue.

lbernstone, 2021-04-20:
If this is something that can be done during compile, it can be moved back into the arduino issue and addressed there.  Maybe a CONFIG_ entry to make it easier for @me-no-dev to set in his CI?

VojtechBartoska, 2022-04-27:
Hello @s-hadinger, is this still valid?

If yes, we can move it to Arduino issue tracker.

s-hadinger, 2022-04-27:
Yes, the issue is still there. We have a work-around in tasmota, by forcing an include with command line option and the following:

``` C

#ifdef __riscv

#undef __INT32_TYPE__
#define __INT32_TYPE__      int

#undef __UINT32_TYPE__
#define __UINT32_TYPE__     unsigned int

#endif // __riscv
```

igrr, 2022-04-27:
FYI, in IDF 5.0 we plan to change Xtensa toolchain to also use `unsigned long` for uint32_t, to be in line with the upstream GCC and with RISC-V.

s-hadinger, 2022-04-27:
Good to know, so it will be consistent between both architectures.

igrr, 2022-08-07:
Starting from IDF v5.0 (and toolchain release esp-2022r1-RC1), uint32_t type for Xtensa is defined as `unsigned long`, same as for RISC-V and same as for the upstream Xtensa GCC.

Migration guide link: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler

closed 2022-08-07 by igrr

baldhead69, 2022-12-06:
Hi @igrr,

The Migration guide link in broken.

Alvin1Zhang, 2022-12-06:
@baldhead69 Thanks for reporting, are you looking for this page https://docs.espressif.com/projects/esp-idf/en/latest/esp32/migration-guides/release-5.x/5.0/gcc.html#int32-t-and-uint32-t-for-xtensa-compiler? Thanks.

mathieucarbou, 2024-04-20:
@s-hadinger @igrr : what would be the correct way following such type def change to handle the printf warnings / format becoming incorrect ? Especially in the case of a library compatible with both Arduino 2 and 3 ?
Ref: https://github.com/espressif/arduino-esp32/issues/9546

igrr, 2024-04-20:
@mathieucarbou The recommended approach is to `#include <inttypes.h>` and use `PRIx32`, `PRIu32`, `PRId32` macros inside your format string when printing uint32_t and `PRId32` when printing int32_t values. This way your format strings will work both for toolchains where int32_t == int (like our older Xtensa GCC releases, and "arduino-esp32 2.x", or our new Clang-based toolchain)  and toolchains where int32_t == long (like all RISC-V GCC toolchains and newer Xtensa GCC toolchains).

(Although it works for some cases, I do not recommend redefining compiler built-in macros as suggested in the comment above.)

s-hadinger, 2024-04-20:
The problem was not about printing values, it was compile errors with arduino libraries.

See above (this fails to compile on RISCV but compiles fine on Xtensa):
```
#include <cstdint>

void test(int *a) {
  // do nothing
}

void setup() {
  int32_t i;
  test(&i);
}

void loop() {
}
```

igrr, 2024-04-20:
Right, but there's a type mismatch in this code. It fails to compile the same way e.g. on bare-metal ARM GCC: https://godbolt.org/z/YM8GqqoMP.

(I understand that your solution is a practical one for the purpose of getting a project such as Tasmota to compile, without having to fix such type mismatch issues in the libraries you depend upon. I just wouldn't recommend it to library maintainers, as eventually such type mismatch issues have to get fixed.)

mathieucarbou, 2024-04-20:
> @mathieucarbou The recommended approach is to `#include <inttypes.h>` and use `PRIx32`, `PRIu32`, `PRId32` macros inside your format string when printing uint32_t and `PRId32` when printing int32_t values. This way your format strings will work both for toolchains where int32_t == int (like our older Xtensa GCC releases, and "arduino-esp32 2.x", or our new Clang-based toolchain)  and toolchains where int32_t == long (like all RISC-V GCC toolchains and newer Xtensa GCC toolchains).
> 
> Here is an example commit: https://github.com/espressif/esp-idf/commit/fb7aafd3480841838be0e6af01bceeff79307ece
> 
> (Although it works for some cases, I do not recommend redefining compiler built-in macros as suggested in the comment above.)

Thanks I didn't know that. We'll do :+1:

diplfranzhoepfinger, 2024-05-30:
leethomason/tinyxml2#982