espressif/esp-idf #6907: problem about wifi:bcn_timout,ap_probe_send_start

niexuzhong, 2021-04-20:
- Development Kit:      ESP32-DevKitC
- Kit version (for DevKitC): v2|
- Module or chip used:  ESP32-WROOM-32
- IDF version v4.4-dev-473-gc49d03b8e
- Build System:        idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

- Operating System:    Linux

- Using an IDE?: No
- Power Supply:         USB

 I use esp_http_client to post an request to the server, then get the reply. after one cycle, It will pause for one seconds. Then another cycle is restart. I get right result by reply, but  after about 120-130 cycles, the error message will be display : wifi:bcn_timout,ap_probe_send_start, then the system is restart. I carefully review before issues about it. Then I do following procedures to solve it. but the problem is still exist:
1. I close the  bluetooth, but the error is still exist.
2. I check heap size , the occupied is unfixed after each cycle, So there should not exist memory leak.
3. I try to test the sample esp_http_client example with similar cycle. but the result is OK.
4. I don't have any exhausted operation(such as writing flash)at whole process.
so my question is:
  can any one give me some suggestions about it. because the restart  is too frequently., how can use some API to check if wifi error or ok. then I can wait the wifi connect again to start next data exchanged. Thank you!

AxelLin, 2021-04-20:
I also observe "wifi:bcn_timout,ap_probe_send_start" sometimes. (see https://www.esp32.com/viewtopic.php?f=13&t=20433)
But it does not cause system restart.
I'm wondering if you have any error log before system restart?
Or maybe you can get some hint by checking esp_reset_reason() during boot.

niexuzhong, 2021-04-20:
following is the log message:
--------------------------------------------------------------------------------
I (209966) wifi:bcn_timout,ap_probe_send_start
W (210466) wifi:m f probe req l=0

W (211466) wifi:m f probe req l=0

W (211476) wifi:m f null

W (211966) wifi:m f probe req l=0

I (212466) wifi:ap_probe_send over, resett wifi status to disassoc
I (212466) wifi:state: run -> init (c800)
W (212476) wifi:m f disassoc

I (212476) wifi:pm stop, total sleep time: 157481598 us / 208351062 us

W (212476) wifi:<ba-del>idx
W (212486) wifi:<ba-del>idx
I (212486) wifi:new:<6,0>, old:<6,2>, ap:<255,255>, sta:<6,2>, prof:1
I (212496) WiFi Station: Wi-Fi disconnected, trying to reconnect...
E (212496) esp-tls: [sock=54] delayed connect error: Software caused connection abort
E (212506) esp-tls: Failed to open new connection
E (212506) TRANSPORT_BASE: Failed to open a new connection
E (212516) HTTP_CLIENT: Connection failed, sock < 0
I (212526) HTTP_CLIENT: open socket error
E (212536) esp-tls: [sock=54] connect() error: Host is unreachable
E (212536) esp-tls: Failed to open new connection
E (212536) TRANSPORT_BASE: Failed to open a new connection
E (212546) HTTP_CLIENT: Connection failed, sock < 0
I (212546) HTTP_CLIENT: open socket error
E (212556) esp-tls: empty esp_tls parameter
E (212556) esp-tls: Failed to open new connection
E (212566) TRANSPORT_BASE: Failed to open a new connection
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a232f  PS      : 0x00060330  A0      : 0x801a2370  A1      : 0x3ffbc2e0  
0x401a232f: esp_tls_connect at    xxxxxx/esp-idf/components/tcp_transport/transport_ssl.c:109 (discriminator 3)
-----------------------------------------------------------------------------------------------------------------------------------------------------------
I add the esp_reset_reason function then I get the restart reason:
 restart reason is 4
Can these message provide useful information?
thank you!

AxelLin, 2021-04-20:
> ## Core 0 register dump:
> PC : 0x401a232f PS : 0x00060330 A0 : 0x801a2370 A1 : 0x3ffbc2e0
> 0x401a232f: esp_tls_connect at xxxxxx/esp-idf/components/tcp_transport/transport_ssl.c:109 (discriminator 3) [...]

Try git pull, v4.4-dev-960-gcf457d412a77 should have fixed this issue by 86bcd9fc21d0e442ff99ff8b6c1906dccc4a810c

niexuzhong, 2021-04-21:
I pull the latest the code. there are new error message:
I (301105) wifi:bcn_timout,ap_probe_send_start
E (301115) TRANSPORT_BASE: Failed to initialize new connection object
E (301115) HTTP_CLIENT: Connection failed, sock < 0
I (301115) HTTP_CLIENT: open socket error
E (301125) TRANSPORT_BASE: Failed to initialize new connection object
E (301135) HTTP_CLIENT: Connection failed, sock < 0
I (301135) HTTP_CLIENT: open socket error
E (301145) HTTP_CLIENT: Allocation failed
I (301145) HTTP_CLIENT: create http client error
I (301155) HTTP_CLIENT: url is http://192.168.1.110/mqtt/handlers/httpserver.ashx
E (301165) HTTP_CLIENT: Allocation failed
I (301165) HTTP_CLIENT: create http client error
I (301165) HTTP_CLIENT: url is http://192.168.1.110/mqtt/handlers/httpserver.ashx
E (301175) HTTP_CLIENT: Allocation failed
--------------------------------------------------------------------
then the above message repeated many times.but it is restart again after too many error:

E (305535) HTTP_CLIENT: Error allocate memory
Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x401a383f  PS      : 0x00060b30  A0      : 0x801a3865  A1      : 0x3ffbc300

niexuzhong, 2021-04-22:
please give me suggestion about it. should I add extra handler event to handle this error. Than which event can be added to  handle it？ Thank you1

espxiehang, 2021-04-22:
Hi @niexuzhong ,
Before the beacon timeout, we will throw the ‘WIFI_EVENT_STA_BEACON_TIMEOUT’ event.

AxelLin, 2021-04-22:
@espxi

> Hi @niexuzhong ,
> Before the beacon timeout, we will throw the ‘WIFI_EVENT_STA_BEACON_TIMEOUT’ event.

What do you expect the user application to do with ‘WIFI_EVENT_STA_BEACON_TIMEOUT’ event? (What is the purpose of this event?)

espxiehang, 2021-04-22:
@AxelLin ，Used to find beacon timeout problems in testing.
@niexuzhong  Can you provide the package capture file.

niexuzhong, 2021-04-22:
Dear espxiehang: how can I log the package capture file?Could your please to give an an link about detail operation description. Thank you！

espxiehang, 2021-04-22:
Hi @niexuzhong , Please refer to:
https://blog.csdn.net/flyingcys/article/details/82534737
https://www.cnblogs.com/mq0036/p/11187138.html

niexuzhong, 2021-04-23:
I try to use wireshark to capture the https://docs.espressif.com/projects/esp-idf/zh_CN/release-v4.3/esp32/api-guides/wireshark-user-guide.html, but the result is always failure. So I use an gateway running openwrt by tcpdump to capture the pack, Then I get following pack file. Can you analysis it? thank you!
[wireshark.zip](https://github.com/espressif/esp-idf/files/6362160/wireshark.zip)

niexuzhong, 2021-04-23:
Dear espxiehang, I just read the website https://docs.espressif.com/projects/esp-idf/zh_CN/release-v4.3/esp32/api-guides/wireshark-user-guide.html,  then I get the following pack file.Can such packs provide useful information? Thank you!@espxiehang
[testwireshark2.zip](https://github.com/espressif/esp-idf/files/6362278/testwireshark2.zip)

espxiehang, 2021-04-23:
Hi @niexuzhong , The signal of the router is too bad, so it is normal to have beacon timeout.
The RSSI in the beacon frame of the router is equal to - 128 dBm.
[image]

niexuzhong, 2021-04-23:
Hi @espxiehang as your suggestion, I will check if router cause such problem ,but I still have question about it:
1.I had changed several routers before, my customer also have same problem, we have tested it at 3-4 different router. maybe it is caused by poor PCB design
2.why the beacon rssi is -128dbm,but the rssi of frame is -31dbm?
3.Is there better solution to solve it(of course, I will improve the hardware design and get better router),but if there is any software to handler it, for example. If I get the timeout event, How can I do to remove it? can I disconnect the wifi or connect it again? Thank you!

espxiehang, 2021-04-23:
Hi @niexuzhong 
When esp32 does not receive a beacon for six seconds, it will generate a beacon timeout event.
Next, esp32 will send five "probe request" frames.
During this period, if beacon and proberesponse are not received, disassoc frame will be sent to disconnect WiFi connection.
You can call "esp_wifi_connect()" in the disconnect event.

2.why the beacon rssi is -128dbm,but the rssi of frame is -31dbm?
I don't know why it happened.

closed 2021-05-01 by niexuzhong