espressif/esp-idf #6908: 调用esp_wifi_sta_get_ap_info后，UDP接收数据丢失

Vinx911, 2021-04-20:
在UDP接受数据时，定时检测WIFI的rssi会导致UDP接收数据丢失。

代码如下：

```c
/* BSD Socket API Example

   This example code is in the Public Domain (or CC0 licensed, at your option.)

   Unless required by applicable law or agreed to in writing, this
   software is distributed on an "AS IS" BASIS, WITHOUT WARRANTIES OR
   CONDITIONS OF ANY KIND, either express or implied.
*/
#include <string.h>
#include <sys/param.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_wifi.h"
#include "esp_event.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_netif.h"
#include "protocol_examples_common.h"

#include "lwip/err.h"
#include "lwip/sockets.h"
#include "lwip/sys.h"
#include <lwip/netdb.h>

#define PORT CONFIG_EXAMPLE_PORT

static const char *TAG = "example";

static void wifi_rssi_task(void *pvParameters) 
{
    wifi_ap_record_t ap_info = {0};
    while (1) {
        esp_wifi_sta_get_ap_info(&ap_info);
        vTaskDelay(1000 / portTICK_PERIOD_MS);
        
        ESP_LOGI(TAG, "rssi = %d", ap_info.rssi);
    }
}

static void udp_server_task(void *pvParameters)
{
    char rx_buffer[128];
    char addr_str[128];
    int addr_family = (int)pvParameters;
    int ip_protocol = 0;
    struct sockaddr_in6 dest_addr;

    while (1) {
            struct sockaddr_in *dest_addr_ip4 = (struct sockaddr_in *)&dest_addr;
            dest_addr_ip4->sin_addr.s_addr = htonl(INADDR_ANY);
            dest_addr_ip4->sin_family = AF_INET;
            dest_addr_ip4->sin_port = htons(PORT);
            ip_protocol = IPPROTO_IP;

        int sock = socket(addr_family, SOCK_DGRAM, ip_protocol);
        if (sock < 0) {
            ESP_LOGE(TAG, "Unable to create socket: errno %d", errno);
            break;
        }
        ESP_LOGI(TAG, "Socket created");

        int err = bind(sock, (struct sockaddr *)&dest_addr, sizeof(dest_addr));
        if (err < 0) {
            ESP_LOGE(TAG, "Socket unable to bind: errno %d", errno);
        }
        ESP_LOGI(TAG, "Socket bound, port %d", PORT);

        while (1) {

            ESP_LOGI(TAG, "Waiting for data");
            struct sockaddr_storage source_addr; // Large enough for both IPv4 or IPv6
            socklen_t socklen = sizeof(source_addr);
            int len = recvfrom(sock, rx_buffer, sizeof(rx_buffer) - 1, 0, (struct sockaddr *)&source_addr, &socklen);

            // Error occurred during receiving
            if (len < 0) {
                ESP_LOGE(TAG, "recvfrom failed: errno %d", errno);
                break;
            }
            // Data received
            else {
                inet_ntoa_r(((struct sockaddr_in *)&source_addr)->sin_addr, addr_str, sizeof(addr_str) - 1);
                rx_buffer[len] = 0; // Null-terminate whatever we received and treat like a string...
                ESP_LOGI(TAG, "Received %d bytes from %s:", len, addr_str);
                ESP_LOGI(TAG, "%s", rx_buffer);
            }
        }

        if (sock != -1) {
            ESP_LOGE(TAG, "Shutting down socket and restarting...");
            shutdown(sock, 0);
            close(sock);
        }
    }
    vTaskDelete(NULL);
}

void app_main(void)
{
    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());

    /* This helper function configures Wi-Fi or Ethernet, as selected in menuconfig.
     * Read "Establishing Wi-Fi or Ethernet Connection" section in
     * examples/protocols/README.md for more information about this function.
     */
    ESP_ERROR_CHECK(example_connect());

    xTaskCreate(udp_server_task, "udp_server", 4096, (void*)AF_INET, 5, NULL);
    xTaskCreate(wifi_rssi_task, "wifi_rssi_task", 4096, 0, 5, NULL);
}
```

Alvin1Zhang, 2021-04-20:
@Vinx911 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, commit ID, hardware and etc. would help us debug further. Thanks.

xueyunfei998, 2021-04-21:
hi @Vinx911 
Can you test the two tasks on different cores? I also try to reproduce the problem with your code。

Vinx911, 2021-04-22:
> hi @Vinx911
> Can you test the two tasks on different cores? I also try to reproduce the problem with your code。

几个版本都进行了测试，都有这样的问题

MaxwellAlan, 2021-05-07:
hi @Vinx911 ,  try turn off modem sleep and test.
call  esp_wifi_set_ps(WIFI_PS_NONE); after wifi start.
thanks.

Vinx911, 2021-05-10:
非常感谢，问题已经解决

closed 2021-05-10 by Vinx911

Espressif-liuuuu, 2022-07-19:
hi @Vinx911 , The fix have been merge to master internally, https://github.com/espressif/esp-idf/commit/51f6091761be0c69b64d6310dcb72082dd5965d8 , which means you could keep PS_MIN_MODEM without issue. please check if you can find it on GitHub.