espressif/esp-idf #6910: How to access scan response data with Nimble stack

afcec, 2021-04-20:
Hi, I have posted this question on esp32.com but it is quite difficult to get answers there... 

so far I have been using **BLE with Nimble** stack taking the 'blecent' example as a reference for my development but always connecting to the remote devices and then subscribing for notifications.

Now I need to read the data sent by a device as a **scan response**. If I am not wrong, first I need to set 'passive' field from 'ble_gap_disc_params' struct to 0 to configure an **active scan**. But then I am quite lost and the documentation available at http://mynewt.apache.org/ is very poor.

In which GAP event type am I supposed to get the scan response data? Any example on how I can read it? Should I work with BLE_GAP_EVENT_DISC and parse event->disc.data?

Thanks

RahulEspressif, 2021-04-20:
Hi @afcec 

 Your understanding is correct. By default, in the existing blecent example provided as part of sdk, passive scan is enabled. As you pointed out, yes , there is a need to modify disc_param.passive as 0 in the blecent example's main.c file.  This will enable scan request. 

Also, BLE_GAP_EVENT_DISC event is generated for scan response data too. ( which is handled in the blecent_gap_event function).

I would like to point to the current header file for more information. i.e.  https://github.com/espressif/esp-nimble/blob/nimble-1.3.0-idf/nimble/host/include/host/ble_gap.h#L404   , we can see that event_type can be used to differentiate between different ADV PDU type. 

So, i think, in your modified code, if the event_type field ( disc->event.event_type ) value is checked, it should help to do the needed parsing for the data accordingly.

Thanks,
 Rahul

afcec, 2021-04-20:
Hi  @RahulEspressif ,

thank you very much for your prompt reply, I will give it a try.

Alvin1Zhang, 2022-05-10:
Thanks for reporting, will close due to short of feedback, feel free to reopen with more updates. Thanks.

closed 2022-05-10 by Alvin1Zhang

umut373, 2025-08-21:
@afcec I'm having the same problem right now. Did you solve the problem? If so, how did you solve it?