espressif/esp-idf #6911: Build error for esp32s3

twomas, 2021-04-20:
## Environment

- Development Kit:      [none]
- IDF version (run ``git describe --tags`` to find it):
    // v4.3-beta2-2-g9a2d251912
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32s3-elf-gcc.exe (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: PowerShell].
- Using an IDE?: [No]

## Problem Description

Config:
CONFIG_PM_ENABLE=y
CONFIG_IDF_TARGET="esp32s3"
CONFIG_IDF_TARGET_ESP32S3=y

[688/1313] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/controller/esp32s3/bt.c.obj
FAILED: esp-idf/bt/CMakeFiles/__idf_bt.dir/controller/esp32s3/bt.c.obj 
C:/DevOps/agt10/_work/1/s/SW/WifiBle/repo4/esp-idf/components/bt/controller/esp32s3/bt.c: In function 'btdm_sleep_enter_phase1_wrapper':
C:/DevOps/agt10/_work/1/s/SW/WifiBle/repo4/esp-idf/components/bt/controller/esp32s3/bt.c:774:28: error: implicit declaration of function 'btdm_lpcycles_2_us'; did you mean 'btdm_lpcycles_2_hus'? [-Werror=implicit-function-declaration]
     uint32_t us_to_sleep = btdm_lpcycles_2_us(lpcycles);
                            ^~~~~~~~~~~~~~~~~~
                            btdm_lpcycles_2_hus
cc1.exe: some warnings being treated as errors

TianaESP, 2021-04-21:
Hi @twomas,

Thanks for reporting the issue. May I know the hardware version you are using now, S3-beta2 or S3-beta3?
You can refer to the Samples Notes:
- ESP32-S3 Beta 2 (marked ‘Marlin9_1B'). Preview support is available in ESP-IDF branch release/v4.3 and the v4.3 release ONLY. 
- ESP32-S3 Beta 3 (marked 'Marlin9'). Preview support is available in the current ESP-IDF master branch but will be removed from ESP-IDF when the final chip is available.
Thanks.

twomas, 2021-04-21:
Hi @TianaESP

Now I'm using esp32 hw for my project.
So I only did a test build for esp32-s3 with my existing project
https://www.espressif.com/en/products/socs/esp32-s3

Do you know if it is possible to get a devkit with ESP32-S3 mounted?

TianaESP, 2021-04-21:
Hi @twomas

Please contact our sales team at [Contact Us](https://www.espressif.com/en/contact-us/sales-questions). 
Or by emailing **sales@espressif.com**. 
Thanks.

closed 2021-04-26 by Alvin1Zhang