espressif/esp-idf #6913: How can I use the sdkconfig in cmake?

0xFEEDC0DE64, 2021-04-20:
I started adding logs of custom Kconfig parameters to my project and now I need to do different things in cmake depending on some boolean Kconfig parameters.

How do I load the sdkconfig file in my main/CMakeLists.txt?

Any examples would be nice

0xFEEDC0DE64, 2021-04-20:
Ok looks like all the CONFIG_ macros were already accessible in CMake, but I cannot conditionally add dependencies to a component with this.

This example just doesnt work as expected:

```
if(CONFIG_HAS_SPECIAL_LIBRARY)
    set(optional_dep speciallib)
endif()

idf_component_register(
    INCLUDE_DIRS
        src
    SRCS
        ${headers}
        ${sources}
    REQUIRES
        cpputils
        cxx-ring-buffer
        date
        espchrono
        ${optional_dep}
)

if(CONFIG_GOE_CHARGING_LOGIC_INCLUDE_FRONIUS_INVERTER)
    message(WARNING "Adding special lib functionality")
    target_compile_options(${COMPONENT_TARGET}
        PUBLIC
            -DHAS_SPECIAL_LIB=1
    )
endif()
```

projectgus, 2021-04-21:
Hi @0xFEEDC0DE64,

Unfortunately, dependencies are expanded during an initial pass of the project when the sdkconfig variables are not available. This is needed to avoid a circular dependency (if a component is not added to the build then its config items are not added to the project's config, so we have to determine the project components first).

The CMakeLists.txt file is actually expanded twice this way - once during early expansion when idf_component_register is used to gather dependency tree information (and no sdkconfig is loaded) and then a second time when the component is actually registered (and config can be used). Therefore any changes to other parts of the components (apart from REQUIRES and PRIV_REQUIRES) that depend on config can be applied at this stage.

There is some description about the "early expansion" process [here in the docs](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#requirements-in-the-build-system-implementation), but I notice that it doesn't include the information about configuration. Will update this.

If I understand what you need to do correctly, the recommended approach is to register "speciallib" as a component always, but if disabled by sdkconfig then don't register any header or source files (the component will be registered as config-only in that case).

For example this is how the [bluetooth (bt) component](https://github.com/espressif/esp-idf/blob/master/components/bt/CMakeLists.txt) works.

projectgus, 2021-05-06:
Hi @0xFEEDC0DE64,

Does the above suggestion work for you?

Angus

Alvin1Zhang, 2021-05-27:
@0xFEEDC0DE64 Thanks for reporting, would you please help share if any updates for this issue? Thanks.

0xFEEDC0DE64, 2021-05-27:
Hi, sorry, was too busy finishing our firmware for production :)

I created cmake variables to add or remove certain linking dependencies and .h and .cpp files to our main project, which seems to be similar to the suggested solution. So yes, I can confirm that it worked.

Thank you all and best regards.

closed 2021-05-27 by 0xFEEDC0DE64

projectgus, 2021-06-07:
> There is some description about the "early expansion" process here in the docs, but I notice that it doesn't include the information about configuration. Will update this.

I came back to make this update but realised it's mentioned under the parent heading here:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/build-system.html#component-requirements

So will leave this part as-is for now, please let me know if something here is unclear.

gojimmypi, 2022-10-15:
If someone else considers "_hm, I wonder if I can use the CMake_ [check_symbol_exists](https://cmake.org/cmake/help/latest/module/CheckSymbolExists.html) _command?_", to more directly check a given header file, say `sdkconfig.h`, perhaps like this:

```
set(CMAKE_REQUIRED_LIBRARIES "sdkconfig")
enable_language("C")
include(CheckSymbolExists)
check_symbol_exists(CONFIG_DEBUG_MYLIB "sdkconfig.h" CONFIG_DEBUG_MYLIB )
message(">>>>> Found CONFIG_DEBUG_MYLIB = ${CONFIG_DEBUG_MYLIB }")
```

The answer is no, not really. Although it does work when inserted _after_ the `project(my_project)` line in the `CMakeFiles.txt`, it is not very helpful there in controlling project dependencies. In my case, I wanted to be able to turn on and off various components from the sdkconfig settings.

When inserted _before_ the `project(my_project)` line, there's no error, but `CONFIG_DEBUG_MYLIB` result is blank.

So, might as well just use `include($ENV{IDF_PATH}/tools/cmake/project.cmake)` (also only useful _after_  the `project(my_project)` line). 

It makes for some useful sanity check messages at compile time, but that's about it. 

Let me know if anyone has any other ideas.