espressif/esp-idf #6914: "region `iram0_0_seg' overflowed" (when optimizations are off) but IRAM seems to be nearly not used

TobiasBoeschMiele, 2021-04-21:
1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
       v4.4-dev-960-gcf457d412
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes, eclipse with esp-idf plugin
- Power Supply:         external 3.3V

## Problem Description

A project cannot be build **when optimization is turned off** (really off -O0) because of an overflowed IRAM segment. With optimization set to debug the project builds and runs and debugging works.

### Expected Behavior

The program will fit into IRAM since it is not full.

### Actual Behavior

Linker sees a filled up IRAM segment.

### Steps to reproduce

Impossible to reproduce - Hello world project does work with full optimization and I am not allowed to publish the project I work on.

## Debug Logs

```
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: [...].elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 18144 bytes
```

## Other items if possible

Application size analysis:
[image]

How do I find the cause of this problem?

igrr, 2021-04-21:
@TobiasBoeschMiele I suspect that this is a bug in the Eclipse plugin rather than a bug in the linker. "126kB available" doesn't look right. Could you please run `idf.py size` from the command line and paste its output as well here? Alternatively, if you can upload the .map file or email it to me (ivan at %companyname%.com), that would help us diagnose the issue quicker.

TobiasBoeschMiele, 2021-04-21:
> Could you please run `idf.py size` from the command line

I tried, but I think this command needs a working build. When the optimization is off, the build will not be successful and this command fails with the build error. Should I share the commands output when optimization is not turned off?

TobiasBoeschMiele, 2021-04-21:
> `idf.py size` with optimization set to Debug `-Og`

```
Total sizes:
 DRAM .data size:   16196 bytes
 DRAM .bss  size:   21624 bytes
Used static DRAM:   37820 bytes (  86760 available, 30.4% used)
Used static IRAM:  108763 bytes (  22309 available, 83.0% used)
      Flash code:  522083 bytes
    Flash rodata:  133172 bytes
Total image size:~ 780214 bytes (.bin may be padded larger)
```

TobiasBoeschMiele, 2021-04-21:
It seems IRAM is already filled up to more than 80 %. That means I have to somehow reduce the IRAM size.

TobiasBoeschMiele, 2021-04-21:
This is the component based analysis of size:

(Real names of some components were replaced by `[numer]`)
```
Per-archive contributions to ELF file:
            Archive File DRAM .data & .bss & other   IRAM   D/IRAM Flash code & rodata   Total
                 libbt.a        164  12803       0    487        0     243869    74459  331782
           libbtdm_app.a        773   2573       0  27818        0      69835     4551  105550
                  libc.a          4      0       0      0        0      80624     4734   85362
                libphy.a       1436    715       0   7790        0      32263        0   42204
             libdriver.a       1040    248       0   5740        0      23387    10705   41120
           libfreertos.a       3112   1044       0  17186        0        370     4190   25902
          libnvs_flash.a          0     24       0      0        0      14196     2924   17144
          libspi_flash.a       1538    294       0   9574        0       1962     1802   15170
         libesp_system.a        245    206       0   3080        0       6075     3885   13491
                libhal.a        136      0       0   6521        0       4846     1935   13438
            libcoexist.a       1804     33       0   3707        0       4190      325   10059
          libapp_trace.a       1570   1977       0   6442        0          0        0    9989
               libheap.a       1965      8       0   5227        0       1408      777    9385
     libesp_hw_support.a        176     45       0   4058        0       3995      878    9152
                lib[1].a         40    654       0    406        0       5481      938    7519
         libesp_common.a          0      0       0      0        0         53     6957    7010
                libgcc.a          8     12       0    104        0       5543      872    6539
                libvfs.a        308     48       0      0        0       5640      517    6513
                libsoc.a          0      0       0    285        0        520     4145    4950
                lib[2].a        113     52       0      0        0       2322     1813    4300
             libstdc++.a          8     20       0      0        0       2517     1044    3589
          libesp_timer.a         32     24       0   1500        0       1180      743    3479
           libesp_wifi.a         92     27       0    242        0       1732     1336    3429
             libnewlib.a        190    272       0   1021        0        667      151    2301
                librtc.a          0      4       0   2227        0          0        0    2231
              libefuse.a         64      4       0      0        0       1230      807    2105
 libbootloader_support.a          0      0       0    965        0        594        0    1559
                liblog.a          8    272       0    262        0        855      144    1541
        libesp_ringbuf.a          0      0       0    905        0          0      538    1443
             libxtensa.a       1024      0       0     16        0        239       35    1314
            libpthread.a         16     12       0    239        0        535      354    1156
              libesp32.a          8     40       0    406        0        220      260     934
         libapp_update.a          1     12       0    194        0        175      453     835
            libesp_ipc.a          0     56       0    188        0        464       89     797
                lib[3].a          0     40       0      0        0        713        0     753
             libxt_hal.a          0      0       0    515        0          0       32     547
                lib[4].a          0      0       0      0        0        139       42     181
            libesp_rom.a          0      0       0    112        0          0        0     112
                   (exe)          0      0       0      3        0          3       12      18
                libcxx.a          0      0       0      0        0         11        0      11
             libesp_pm.a          0      0       0      0        0          8        0       8
               libcore.a          0      0       0      0        0          0        0       0
            libesp_eth.a          0      0       0      0        0          0        0       0
          libesp_event.a          0      0       0      0        0          0        0       0
          libesp_netif.a          0      0       0      0        0          0        0       0
               liblwip.a          0      0       0      0        0          0        0       0
         libmbedcrypto.a          0      0       0      0        0          0        0       0
            libmbedtls.a          0      0       0      0        0          0        0       0
           libmbedx509.a          0      0       0      0        0          0        0       0
               libmesh.a          0      0       0      0        0          0        0       0
           libnet80211.a          0      0       0      0        0          0        0       0
                 libpp.a          0      0       0      0        0          0        0       0
      libtcpip_adapter.a          0      0       0      0        0          0        0       0
     libwpa_supplicant.a          0      0       0      0        0          0        0       0
```

igrr, 2021-04-21:
> When the optimization is off, the build will not be successful and this command fails with the build error. Should I share the commands output when optimization is not turned off?

You're right, `idf.py size` is indeed not working when the build fails... However the .map file should still be generated by the linker, so you can try to run the following after the failed `idf.py build`:
```
python $IDF_PATH/tools/idf_size.py build/your_app_name.map
```

If that works, please also run 
```
python $IDF_PATH/tools/idf_size.py build/your_app_name.map --json
```

and post the output you get — it's the JSON output that is consumed by Eclipse.

With regards to reducing IRAM usage, you can try enabling [`CONFIG_FREERTOS_PLACE_FUNCTIONS_INTO_FLASH`](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/kconfig.html#config-freertos-place-functions-into-flash) — it places FreeRTOS function which aren't used in ISRs into flash instead of IRAM.

TobiasBoeschMiele, 2021-04-21:
*Note: I used `%IDF_PATH%` on windows.*

```
somewhere>python %IDF_PATH%/tools/idf_size.py build/[appname].map
Total sizes:
 DRAM .data size:   19268 bytes
 DRAM .bss  size:   21656 bytes
Used static DRAM:   40924 bytes (  83656 available, 32.8% used)
Used static IRAM:    1027 bytes ( 130045 available, 0.8% used)
      Flash code:  715291 bytes
    Flash rodata:  160148 bytes
Total image size:~ 895734 bytes (.bin may be padded larger)
```

while the build fails with:
```
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: [appname].elf section `.iram0.text' will not fit in region `iram0_0_seg'
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: IRAM0 segment data does not fit.
c:/espressif_esp_idf_tools/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: region `iram0_0_seg' overflowed by 18144 bytes
```

This looks very similar to the eclipse plugin calculations in the [initial comment](https://github.com/espressif/esp-idf/issues/6914#issue-863559155)

igrr, 2021-04-21:
Alright, so it's an idf_size.py issue then. I think that it simply doesn't consider the possibility of an overflow yet, hence the calculation is incorrect. We will fix this, thank you for all the details.

TobiasBoeschMiele, 2021-04-21:
You're welcome.
I think solving the overflow issue (which I think will persist after fixing the `idf.size.py` issue) has to wait for fixing the `idf.size.py` issue?

TobiasBoeschMiele, 2021-09-27:
This issue persists (veryfied). I cannot reopen it.

igrr, 2021-09-27:
@TobiasBoeschMiele could you please clarify which issue you are seeing? The issue that IRAM segment overflows at -O0 level, or that idf.py size doesn't work when overflow happens? The linked commit should have fixed the latter issue.

TobiasBoeschMiele, 2021-09-28:
I made a mistake and did not update before testing the issue. I will test this again right now.

closed 2021-09-28 by TobiasBoeschMiele

ESP-Marius, 2023-04-11:
On latest master we've added a kconfig option for using parts of SRAM1 as IRAM in https://github.com/espressif/esp-idf/commit/5cbd311ecf0b381a0711ef8a96980a67b30a135f which will help with this issue.

We've also focused on reducing IRAM usage/making placement optional for several components. For an full overview of all such option see [Optimizing-iram-usage](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/performance/ram-usage.html#optimizing-iram-usage). All new options for saving IRAM we add will be listed in this chapter.