espressif/esp-idf #6915: Failed to resolve component 'esp_tls'

YangJiawei, 2021-04-21:
## Environment

- Development Kit:      [ESP32-Wrover-Kit]
- Kit version (for WroverKit/PicoKit/DevKitC): [v4.1]
- Module or chip used:  [ESP32-WROVER-B]
- IDF version : v4.2 - c40f2590bf
- Build System:   [CMake|idf.py]
- Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [ubuntu]
- Using an IDE?: [VSCode + Espressif IDFv1.0.2 ]
- Power Supply:         [USB]

## Problem Description
使用`hello_world`例程，
`>ESP-IDF:Create new ESP-IDF Component`添加`demo`组件
tree：
```
.
├── CMakeLists.txt
├── components
│   └── demo
│       ├── CMakeLists.txt
│       ├── demo.c
│       └── include
│           └── demo.h
├── debug.log
├── main
│   ├── CMakeLists.txt
│   ├── component.mk
│   └── hello_world_main.c
├── Makefile
├── README.md
└── sdkconfig
```
demo.c
```cpp
#include <stdio.h>
#include "demo.h"

#include "nvs_flash.h"
#include "esp_tls.h"

void func(void)
{

}

```
`components/demo/CMakeLists.txt`
```cpp
idf_component_register(SRCS "demo.c"
                    INCLUDE_DIRS "include"
                    REQUIRES nvs_flash esp_tls )
```

## Debug Logs
`Build project`或者`idf.py build`提示如下：
```
> Executing task: cmake -G Ninja .. <

-- Building ESP-IDF components for target esp32
-- Checking Python dependencies...
Python requirements from /home/yjw/esp/esp-idf/requirements.txt are satisfied.
CMake Error at /home/yjw/esp/esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'esp_tls'.
Call Stack (most recent call first):
  /home/yjw/esp/esp-idf/tools/cmake/build.cmake:212 (__build_resolve_and_add_req)
  /home/yjw/esp/esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  /home/yjw/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/yjw/demo/ESP32/hello_world/build/CMakeFiles/CMakeOutput.log".
终端进程“/bin/bash '-c', 'cmake -G Ninja ..'”已终止，退出代码: 1。

终端将被任务重用，按任意键关闭。

```
```
Executing action: all (aliases: build)
Running cmake in directory /home/yjw/demo/ESP32/hello_world/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/yjw/demo/ESP32/hello_world"...
-- Building ESP-IDF components for target esp32
CMake Error at /home/yjw/esp/esp-idf/tools/cmake/build.cmake:186 (message):
  Failed to resolve component 'esp_tls'.
Call Stack (most recent call first):
  /home/yjw/esp/esp-idf/tools/cmake/build.cmake:212 (__build_resolve_and_add_req)
  /home/yjw/esp/esp-idf/tools/cmake/build.cmake:426 (__build_expand_requirements)
  /home/yjw/esp/esp-idf/tools/cmake/project.cmake:395 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/home/yjw/demo/ESP32/hello_world/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
```
引用`nvs_flash`通过，但是引用`esp_tls`依赖失败

AxelLin, 2021-04-21:
REQUIRES esp-tls instead of esp_tls.

YangJiawei, 2021-04-21:
> 需要esp-tls而不是esp_tls。

OMG!!!!!!!!
```
├── esp_adc_cal
├── esp_common
├── espcoredump
├── esp_eth
├── esp_event
├── esp_gdbstub
├── esp_hid
├── esp_http_client
├── esp_http_server
├── esp_https_ota
├── esp_https_server
├── esp_ipc
├── esp_local_ctrl
├── esp_netif
├── esp_ringbuf
├── esp_rom
├── esp_serial_slave_link
├── esp_system
├── esp_timer
├── esp-tls
├── esptool_py
├── esp_websocket_client
├── esp_wifi
```
愚蠢的我竟然因为这个问题弄了一天！！！

closed 2021-04-21 by YangJiawei