espressif/esp-idf #6916: Flash encryption and secure boot together process

PranavD10, 2021-04-21:
Hi,
I have queries about flash encryption and secure boot process. If I implement secure boot first in my application and then flash encryption, does that means I probably broke the bootloader as it will get encrypted by flash encryption, and the secure bootloader digest will be different in that case.
Thanks for your input in advance.

Alvin1Zhang, 2021-04-21:
@PranavD10 Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

PranavD10, 2021-04-21:
Thank for replying,
I am using esp32 wroom dev kit v1 and using secure boot v1.
I shall explain in detail.
The secure boot process calculates the bootloader digest and store it at 0x0. Now at this point my bootloader is in plaintext format. After that i enable flash encryption, the flash encryption process will encrypt data in place for bootloader as well. So on next boot as secure boot will verify the digest of the encrypted bootloader with the plaintext bootloader digest which is already stored at 0x0. Which wont match? Am i correct or something i didnt get ??

negativekelvin, 2021-04-21:
https://github.com/espressif/esp-idf/blob/1cb31e50943bb757966ca91ed7f4852692a5b0ed/components/bootloader_support/src/esp32/flash_encrypt.c#L268-L279

projectgus, 2021-04-22:
Hi @PranavD10,

Yes, for Secure Boot V1 enabling Secure Boot first and then enabling Flash Encryption later is an issue. Although the digest can be encrypted, when you enable Flash Encryption in menuconfig and recompile the bootloader then you end up with a different bootloader binary, so the old digest isn't valid even as plaintext.

If you are using Secure Boot V1 in "reflashable" mode, or you otherwise pre-burned the Secure Boot V1 secret key and have a copy of it, then you could generate a new digest for the new bootloader, flash it, and then boot it to enable flash encryption. But if you don't have this key then the bootloader can't be updated at all.

(Secure Boot V2 improves this situation because both bootloader and app are signed, so it's possible to update them in the same way. But this requires ESP32 ECO3 or a newer chip.)

Does that all make sense?

PranavD10, 2021-04-22:
Thanks for replying @negativekelvin and @projectgus,
@negativekelvin that part tells us about encrypting the bootloader digest and iv at 0x0, I am concern about digest of encrypted bootloader.
@projectgus if I enable the secure boot and compile, the bootloader binaries will be different. But if I enable both options in menuconfig i.e. Flash encryption and secure boot ( both using host generated keys ). does that resolves the issue of different bootloader binaries ?
Also if I have all the keys on the host .i.e flash encryption, secure boot sign and secure bootloader key, Does that means I have to encrypt the bootloader and then calculate the digest and store it at 0x0.
Also what documentation should I follow If I want to perform flash encryption, and secure boot manually without first boot in production environment?
Thank you

projectgus, 2021-04-23:
Hi @PranavD10,

I don't know what ESP-IDF version you're using, but I'm going to assume V4.x.

We don't have a documented workflow for pre-encrypting everything on the host at the moment, it's quite complex and we're encouraging people to use Development mode flash encryption for local testing instead of pre-encrypting on the host (see also #5037). However it's possible, you can do it like this:

1. Generate 256-bit random keys for flash encryption and secure boot.
1. [Pre-burn your flash encryption key](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/flash-encryption.html#using-host-generated-key) using `espefuse.py --port PORT burn_key flash_encryption my_flash_encryption_key.bin`
1. [Pre-burn your Secure boot key using `espefuse.py --port PORT burn_key secure_boot_v1 my_secure_boot_key.bin`
([Alternatively you can use the Reflashable option which generates a key derived from the Signing key](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/security/secure-boot-v1.html#re-flashable-software-bootloader).
1. Configure your project for Flash Encryption and Secure Boot as per the docs, flash it in plaintext, allow it to do the first boot.
1. Instead of generating new keys, the first boot will use the keys you already pre-burned.

With this method, remaining secure efuses are still burned as part of the first boot of the device when it self-encrypts. This is recommended as it's harder to end up in a situation where not everything is correctly set.

Then, after all security features are enabled, to update the bootloader with a new binary you need to manually run espsecure.py to generate a new Secure Boot digest, encrypt the whole thing, and then flash it:

```
espsecure.py digest_secure_bootloader --keyfile my_secure_boot_key.bin --output bootloader-digest.bin build/bootloader.bin
espsecure.py encrypt_flash_data --keyfile my_flash_encryption_key.bin --address 0x0 -o bootloader-digest-encrypted.bin bootloader-digest.bin
esptool.py -P PORT -b BAUD write_flash 0x0 bootloader-digest-encrypted.bin
```

Note: the digest contains both the digest and the bootloader, so only one file needs to be flashed.

To encrypt the app locally and then flash, use the same `espsecure.py encrypt_flash_data` command but change the address to match the app flashing address, then flash the ciphertext file with esptool.py.

Can you tell us a bit more about the problem you're trying to solve? Is this for a factory workflow? If you can give more details then I may be able to give some alternative suggestions.

PranavD10, 2021-04-24:
Hi @projectgus ,
Thanks for the detailed answer. It resolves my issue about its implementation. I am looking for a method that can be implemented for production. which will help to manage device keys and firmware.
Also There is no tool by espressif to manage all these data. Is there any other third party tool available for that.
Do I have to create another issue to ask this or its okay to be off topic on this issue. 
Thank you

closed 2021-04-26 by PranavD10