espressif/esp-idf #6917: Secure boot and flash encryption complete process on fresh device for production.

PranavD10, 2021-04-21:
Hi,
How do I create the production grade device using flash encryption, secure boot, using host generated flash encryption key, secure boot signing key and secure bootloader digest. Also what are the efuse bits i need to burn and protect in order for UART access disable and bootloader locking.
Thanks

Alvin1Zhang, 2021-04-21:
Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

PranavD10, 2021-04-26:
Hi,
I have resolved my queries about the process in other thread. Thanks to @projectgus .

closed 2021-04-26 by PranavD10

roysG, 2021-07-07:
can you share the link for the other thread? thanks

PranavD10, 2021-07-10:
Here it is... [6916](https://github.com/espressif/esp-idf/issues/6916)