espressif/esp-idf #6919: LWIP lwipopts.h is outdated on ESP32

Harvie, 2021-04-21:
This file seems to be old
https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h

Please update it with something more recent, eg.:
https://github.com/espressif/esp-lwip/blob/master/src/include/lwip/opt.h

-------

Here is one example why this matters:

OLD version:
```C
#define LWIP_TCPIP_CORE_LOCKING         0
```
It is not possible to override defines by buildsystem. Build fails on invalid redefine.
This is problem for most defines in that file.

NEW version:
```C
#if !defined LWIP_TCPIP_CORE_LOCKING || defined __DOXYGEN__
#define LWIP_TCPIP_CORE_LOCKING         1
#endif
```
Redefinition is prevented and therefore it is possible to parametrize the LWIP by buildsystem.

AxelLin, 2021-04-22:
@Harvie 
Does it work for you with setting LWIP_TCPIP_CORE_LOCKING=1 ?
I just tried it on v4.3-beta3, then the device failed to join AP if LWIP_TCPIP_CORE_LOCKING=1.

Harvie, 2021-04-22:
@AxelLin 
Not really sure, i was not able to build with LWIP_TCPIP_CORE_LOCKING=1 because in buildsystem i use, i can't edit files,  i can only specify -D parameters and that currently causes redefine issue.

Reason i am looking into this is that i am writing my own netif driver for LWIP and while according to LWIP mailing list
i have everything setup correcty, it seems that there is some kind of data race.
But that is not point of this issue. I just want all #defines to be protected by #ifdefs, so i can experiment freely by modifying CFLAGS.

AxelLin, 2021-04-22:
What's your IDF version? (run git describe --tags to find it)

Harvie, 2021-04-22:
v4.2, it has LWIP v2.1.2. Which is latest STABLE release of LWIP.
I use mongoose os buildsystem (mos tool) to build it, which kinda limits me. Meaning i can only specify CFLAGS, but cannot modify header files to change config.

david-cermak, 2021-04-22:
@Harvie Thanks for reporting this issue. Indeed, the `LWIP_TCPIP_CORE_LOCKING` should be used in IDF's port in future. Due to some historical reasons, esp-lwip still has this option set to `0` and IDF has to care about thread safety manually, typically using `esp_netif` component.

We're already testing this option, but it will take some time to get all the components and usecases properly tested and profiled, so that it could be safely used in stable IDF releases. Here's an experimental patch, which let users enable this option.
[lwip-optimization-config-option-LWIP_TCPIP_CORE_LOCKING.patch.txt](https://github.com/espressif/esp-idf/files/6358440/lwip-optimization-config-option-LWIP_TCPIP_CORE_LOCKING.patch.txt)

Harvie, 2021-04-22:
> Thanks for reporting this issue. Indeed, the LWIP_TCPIP_CORE_LOCKING should be used in IDF's port in future

Good to know. But that was not my point. My point is that currently outdated header file is used, which does not allow users to override the settings. Whether LWIP_TCPIP_CORE_LOCKING is used or not is not currently my complaint.

Also i've discussed my issue with lwip team and it seems that they synchronize using FIFO queue in case that MUTEX is disabled. So it's not critical to me. But i would like the header file to be updated with one from currently used LWIP version as it does obviously lag behind.

I am not sure why this file is even in there, because LWIP ships with such file per se. Anyway if esp-idf ships with second separate copy of that header file, i think it would make sense to at least keep both files at the same version of LWIP.

david-cermak, 2021-04-22:
> I am not sure why this file is even in there,

These two files have *very* different meaning. The lwip's internal https://github.com/espressif/esp-lwip/blob/master/src/include/lwip/opt.h define default options for generic lwip stack.
The IDF's https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h define specific options for the IDF port of lwip, that is tested and verified. Some options could be configured using Kconfig. Some options could be even overridden by injecting a macro from the build system, but this is not recommended and unsupported use of IDF.

Harvie, 2021-04-22:
Currenly i am injecting these:

```
  LWIP_NUM_NETIF_CLIENT_DATA 1
  BRIDGEIF_MAX_PORTS 3
```

Unlike core locking, this seems to build without throwing redefine error. But haven't yet tested if bridging actualy works. However i will need that, so hopefuly it will be OK.

AxelLin, 2021-04-23:
> We're already testing this option, but it will take some time to get all the components and usecases properly tested and profiled, so that it could be safely used in stable IDF releases. Here's an [...]

Hi @david-cermak 
The patch works in my test, thanks.

Below adds some missing #if !LWIP_TCPIP_CORE_LOCKING guard.
https://gist.github.com/AxelLin/06556f114a25462f44e36f2f35c92e51

AxelLin, 2021-04-26:
> Here's an experimental patch, which let users enable this option.
> [lwip-optimization-config-option-LWIP_TCPIP_CORE_LOCKING.patch.txt](https://github.com/espressif/esp-idf/files/6358440/lwip-optimization-config-option-LWIP_TCPIP_CORE_LOCKING.patch.txt)

Hi @david-cermak 
I notice an issue with your patch:
After apply your patch and set CONFIG_LWIP_TCPIP_CORE_LOCKING=n.
ping will always fail as below:
E (18793) ping_sock: send error=0

I added debug code and found:
  sent != (ssize_t)ep->icmp_pkt_size
  (sent is -1, ep->icmp_pkt_size is 72)

It works if set CONFIG_LWIP_TCPIP_CORE_LOCKING=y.

AxelLin, 2021-04-26:
@david-cermak 

You patch changes the logic in esp_netif_lwip_ipc_call when !LWIP_TCPIP_CORE_LOCKING.

In the original code, if (g_lwip_task == xTaskGetCurrentTaskHandle()) it will call fn(&msg);
but in your patch it just return ESP_ERR_INVALID_STATE.
This causes ping failure.

BTW, I don't know the reason you set LWIP_NETCONN_SEM_PER_THREAD=1 when !LWIP_TCPIP_CORE_LOCKING.
LWIP_NETCONN_SEM_PER_THREAD was default 0 in original code.

Attached my changes for components/esp_netif/lwip/esp_netif_lwip.c for reference:
[fix-esp_netif_lwip-core-locking-support.diff.txt](https://github.com/espressif/esp-idf/files/6376455/fix-esp_netif_lwip-core-locking-support.diff.txt)

david-cermak, 2021-04-30:
@AxelLin Thanks a lot for sharing this patch. This makes much sense and looks cleaner!

As you see, we're testing the `CONFIG_LWIP_TCPIP_CORE_LOCKING=y` option so far, and didn't pay much attention to the compatible mode. Thanks for catching the ping issue in this experimental change.

david-cermak, 2021-05-06:
> LWIP_NETCONN_SEM_PER_THREAD was default 0 in original code.

[`LWIP_NETCONN_SEM_PER_THREAD=1`](https://github.com/espressif/esp-idf/blob/639e7ad494d93fc82159f9fe854041bc43a96d5f/components/lwip/port/esp32/include/lwipopts.h#L1034) on 639e7ad494d93fc82159f9fe854041bc43a96d5f

AxelLin, 2021-05-06:
> [`LWIP_NETCONN_SEM_PER_THREAD=1`](https://github.com/espressif/esp-idf/blob/639e7ad494d93fc82159f9fe854041bc43a96d5f/components/lwip/port/esp32/include/lwipopts.h#L1034) on [...]

Hmm, this is strange.

Just to confirm if LWIP_NETCONN_SEM_PER_THREAD is set:

In components/lwip/lwip/src/api/api_lib.c
Find the code around #if LWIP_NETCONN_SEM_PER_THREAD
and I type some random character.

Then I build my application, it does not hit copmile error.
i.e. I'm sure LWIP_NETCONN_SEM_PER_THREAD is 0

BTW, I don't see how  LWIP_NETCONN_SEM_PER_THREAD can be changed by 639e7ad, it looks not related at all.

david-cermak, 2021-05-06:
It's the same on the [latest master](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h#L1034), btw.
...and it was like that from the [first public version](https://github.com/espressif/esp-idf/commit/bd6ea4393c7)  (if you `git blame` the line, just picked 639e7ad494d93fc82159f9fe854041bc43a96d5f since that was the version you used in the referred issue.

AxelLin, 2021-05-06:
> It's the same on the [latest master](https://github.com/espressif/esp-idf/blob/master/components/lwip/port/esp32/include/lwipopts.h#L1034), btw.
> ...and it was like that from the [first public version](https://github.com/espressif/esp-idf/commit/bd6ea4393c7) (if you `git blame` the line, just picked [639e7ad](https://github.com/espressif/esp-idf/commit/639e7ad494d93fc82159f9fe854041bc43a96d5f) since that was the version you used in the referred issue.

Ah, I see.
My bad, just mis-reading.

Harvie, 2021-05-19:
BTW had any of you tried to setup LWIP bridge between two interfaces? I need it, but was not able to get that to work so far...

david-cermak, 2021-05-21:
@AxelLin Sorry for that, will close this issue once merged. Should be safe, but we're currently getting some issues with setting `CONFIG_LWIP_TCPIP_CORE_LOCKING=y` and wanted to resolve these (basically setting reasonable stack sizes for system tasks)

@Harvie We discussed the bridge interface when implementing the `esp_netif` and theoretically this should work even with IDF, but TBH haven't really tried that. You can raise a feature request (as a separate issue) and explain your use-case(s), reasoning and benefits for (IoT) users (if more people comment/upvote, this could be implemented)

AxelLin, 2021-07-30:
> @AxelLin Sorry for that, will close this issue once merged. Should be safe, but we're currently getting some issues with setting `CONFIG_LWIP_TCPIP_CORE_LOCKING=y` and wanted to resolve these (basically setting reasonable stack sizes for system tasks)

Do you have target timeline to support CONFIG_LWIP_TCPIP_CORE_LOCKING=y?
(As I said, it won't be tested by users if the config setting is not available. Then it's difficult to make it really stable and usable.)

AxelLin, 2021-09-08:
CONFIG_LWIP_TCPIP_CORE_LOCKING option is added by 0d07569fff5d41ecbadd7606cb9b3c6224597841.

Alvin1Zhang, 2021-09-08:
@AxelLin Thanks for the notes. @Harvie  Thanks for reporting, feel free to reopen.

closed 2021-09-08 by Alvin1Zhang