espressif/esp-idf #6921: BT_BLE_42_FEATURES_SUPPORTED need IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3 in 4.3

GeneKong, 2021-04-22:
I think BLE 4.2 is support in all esp32 version, why in 4.3 version is depends on (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))?
```
config BT_BLE_50_FEATURES_SUPPORTED
    bool "Enable BLE 5.0 features"
    depends on (BT_BLUEDROID_ENABLED && (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))
    default y
    help
        This enables BLE 5.0 features, this option only support esp32c3/esp32s3 chip

config BT_BLE_42_FEATURES_SUPPORTED
    bool "Enable BLE 4.2 features"
    depends on (BT_BLUEDROID_ENABLED && (IDF_TARGET_ESP32C3 || IDF_TARGET_ESP32S3))
    default n
    help
        This enables BLE 4.2 features.
```

Weijian-Espressif, 2021-04-22:
These two options were added for ESPC3 and ESPS3. ESP32 only supports 4.2, so 4.2 is enabled by default

GeneKong, 2021-04-22:
> SP32 only supports 4.2, so 4.2 is enabled by de

No, this option is not enable, so some function is wrong.

Weijian-Espressif, 2021-04-22:
ESP32 enable 4.2 features by default

Weijian-Espressif, 2021-04-22:
If you get any error, please provide your log, thanks.

GeneKong, 2021-04-22:
OK, I found this in espidf

So there have some other place to open this malloc?

 

 
&mdash;

GeneKong, 2021-04-22:
OK, I found this in espidf

[image]

[image]

[image]

So there have some other place to open this malloc?

GeneKong, 2021-04-22:
> If you get any error, please provide your log, thanks.

Just some compile is error, no esp_ble_adv_params_t type found

Weijian-Espressif, 2021-04-22:
please refer
` #if (UC_BT_BLE_42_FEATURES_SUPPORTED == TRUE || BLE_50_FEATURE_SUPPORT == FALSE)
#define BLE_42_FEATURE_SUPPORT   TRUE
#else
#define BLE_42_FEATURE_SUPPORT   FALSE
#endif`

GeneKong, 2021-04-22:
OK, bad trace

closed 2021-04-22 by GeneKong