espressif/esp-idf #6922: The socket does not close

kormilicinkostia, 2021-04-22:
## Environment

- Module or chip used: ESP32-WROOM-32UE(16MB)
- IDF version V4.1
- Build System: CMake
- Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2019r2) 8.2.0
- Operating System: Linux
- environment type: VS Code
- Using an IDE?: VS Code
- Power Supply: external 3.3V
## Problem Description

I am using the example https://github.com/espressif/esp-idf/tree/v4.1/examples/protocols/sockets/tcp_server to start a tcp server on the esp32 side.
I am connecting to the server from a mobile phone. After a successful connection, I turn off wifi on the phone. After that, the esp32 code hangs on the recv function (line 35). Therefore, the socket is not closed and it is not possible to reconnect to esp32.

### Expected Behavior

After disabling wifi on the client side, the socket on the esp32 side is closed.

### Actual Behavior

After disconnecting wifi on client side, socket on esp32 side does not close.

### Steps to reproduce

1. Enable tcp server on esp32 side
2. Connect from a mobile phone to a tcp server
3. Disable wifi on mobile

yuanjianmin, 2021-04-23:
@kormilicinkostia hi, you can try the tcp_server example in master branch. When disabling the wifi on the client side, server can't detect the TCP connection logic whether is on. We need depend on TCP keep-alive mechanism to detect it. You can follow this code https://github.com/espressif/esp-idf/blob/master/examples/protocols/sockets/tcp_server/main/tcp_server.c#L134 and try.

kormilicinkostia, 2021-04-23:
The suggested solution works. Thank you so much.

closed 2021-04-23 by kormilicinkostia