espressif/esp-idf #6923: Seek more information about the nvs flash memory

ankit-thealchemist, 2021-04-22:
----------------------------- Delete below -----------------------------

## Environment

- Development Kit:    ESP32-DevKitC
- Kit version (for DevKitC): [v4]
- Module or chip used:  ESP32-WROOM-32D
- IDF version (run ``git describe --tags`` to find it):
    // v4.2
-
## Problem Description

Hello, I was implementing the ble mesh Scene server. I am storing the scene in the nvs flash using the Non-volatile storage library. I also enable the option "Store BLE Mesh configuration persistently". I am storing the all values of scenes in the default partition "nvs". So far every thing goes fine. I implemented the reset the node by calling the nvs_flash_erase() function in the event "    ESP_BLE_MESH_NODE_PROV_RESET_EVT". I surprised to find that all the data related to ble mesh has been deleted as expected , but the data related to scene is still in the flash memory. I have to delete it using the function nvs_erase_all(). As per my knowledge we are storing all the data into the flash in the nvs partition. while erasing the whole partition using the function nvs_flash_erase() the data related to scene should also be deleted.  What I am missing here?

InfiniteYuan, 2021-04-23:
@ankit-thealchemist Currently, scene-related data is not stored in the ble mesh protocol stack. What API are you using to store scene data?

ankit-thealchemist, 2021-04-23:
@InfiniteYuan Thanks for the speedy reply. My apologies I wasn't clear.  Actually its not the scene related issue its more about storing the values into the nvs flash and nvs flash working .   

Suppose we save the value  in the nvs flash using the key value pair. I create the handle and open the  the name space using the function nvs_open(), its description is "Open non-volatile storage with a given namespace from the default NVS partition.". So what I assuming that I am saving the value in the "nvs" partition by using the function nvs_set_blob(). Now when I try to erase the partition using the function nvs_flash_erase(), it should delete all the name space and the value stored in the nvs partition, but its not happening. I can still access the data in the namespace.  I have to delete the data using the function nvs_erase_all().  My question is it not enough to use nvs_flash_erase() to format the partition, do we have to delete the multiple namespace individually in a given partition?
I want to have deeper understanding of the nvs flash.

igrr, 2021-04-23:
> Now when I try to erase the partition using the function nvs_flash_erase(), it should delete all the name space and the value stored in the nvs partition, but its not happening. I can still access the data in the namespace

Just to check, are you de-initializing NVS (nvs_close, nvs_flash_deinit) before erasing the partition and, re-initializing it afterwards?

Could you post the code snippet which demonstrates the problem, please?

ankit-thealchemist, 2021-04-23:
@igrr thanks for the speedy reply. I just de-initialize the nvs first and then uses the function nvs_flash_erase(), now its working fine.
Is it mandatory to de initialize the nvs memory before erasing?  Also the notes in the  nvs_flash_erase() function in documentation mention that "If the partition is initialized, this function first de-initializes it. Afterwards, the partition has to be initialized again to be used."  also in the return type its mention, ESP_OK on success
ESP_ERR_NOT_FOUND if there is no NVS partition labeled “nvs” in the partition table

different error in case de-initialization fails (shouldn’t happen).

I am using the function ESP_ERROR_CHECK(nvs_flash_erase()) but not getting any error.  I am getting confuse.

igrr, 2021-04-23:
nvs_flash_erase indeed de-initializes NVS automatically starting from commit https://github.com/espressif/esp-idf/commit/98b1da9e6034dcfb1cd95c30a14001036aab3f59, in particular to prevent issues like this. This change is part of v4.2 release. Are you using exactly the v4.2 tag version, as you indicated in the initial post? What is the output of `git describe` executed in IDF repository?

If you are indeed on v4.2 tag, then I can't explain this behavior, as the commit linked above contains a test case for this specific scenario. Please post the code snippet you use to reproduce this issue, we will investigate. Thank you.

If you are on an older version (such as v4.2-beta) then it is recommended to update to the stable release (v4.2 at the time of writing) or use the suggested approach with deinitializing NVS before erasing.

ankit-thealchemist, 2021-04-25:
@igrr sorry for the delayed reply!

Yes it is v4.2
[image]

This is how I am storing the values:

nvs_handle_t myHandle;
            esp_err_t err;
            err = nvs_open(SceneStorage, NVS_READWRITE, &myHandle);
            if (err != ESP_OK)
            {
                ESP_LOGE(bleTag, "cannot open the nvs storage due to %s", esp_err_to_name(err));
            }

            size_t required_size = 0;
            err = nvs_get_blob(myHandle, NvsSceneStorage, NULL, &required_size);
            // checking whether the blob is initiated or not, if not intiated it will intitate it  and fill with 0
            if (required_size == 0)
            {
                struct nvs_storage_scene *vSceneStorage = calloc(NumberOfSceneCount, sizeof(struct nvs_storage_scene));
                if (vSceneStorage == NULL)
                {
                    ESP_LOGE(bleTag, "Stack is full. Cant get the free space");
                    return;
                }
                err = nvs_set_blob(myHandle, NvsSceneStorage, vSceneStorage, NumberOfSceneCount * sizeof(struct nvs_storage_scene));
                if (err != ESP_OK)
                {
                    ESP_LOGE(bleTag, "nvs storage problem due to %s", esp_err_to_name(err));
                }
                free(vSceneStorage);
                nvs_commit(myHandle);
                if (err != ESP_OK)
                {
                    ESP_LOGE(bleTag, " nvs storage due to %s", esp_err_to_name(err));
                }
            }
            required_size = NumberOfSceneCount * sizeof(struct nvs_storage_scene);
            struct nvs_storage_scene *vSceneStorage = malloc(required_size);
            if (vSceneStorage == NULL)
            {
                ESP_LOGE(bleTag, "Stack is full. Cant get the free space");
                return;
            }
            err = nvs_get_blob(myHandle, "sceneDatabase", vSceneStorage, &required_size);
            if (err != ESP_OK)
            {
                ESP_LOGE(bleTag, "nvs storage problem due to %s", esp_err_to_name(err));
            }
            memcpy(vSceneStorage + i, &vStore_scene, sizeof(struct nvs_storage_scene));
            err = nvs_set_blob(myHandle, "sceneDatabase", vSceneStorage, required_size);
            free(vSceneStorage);

            if (err != ESP_OK)
            {
                ESP_LOGE(bleTag, " nvs storage due to %s", esp_err_to_name(err));
            }

            err = nvs_commit(myHandle);
            if (err != ESP_OK)
            {
                ESP_LOGE(bleTag, " nvs storage due to %s", esp_err_to_name(err));
            }
            nvs_close(myHandle);
            break; // break for the main for loop for processing the scene storage
        }   

and this is how i am trying to erase the partition

case ESP_BLE_MESH_NODE_PROV_RESET_EVT:
        ESP_LOGI(tag, "ESP_BLE_MESH_NODE_PROV_RESET_EVT");
        ESP_LOGI(tag, "Resetting the node to the unprovisioned state");
        ESP_ERROR_CHECK(nvs_flash_deinit());
        ESP_ERROR_CHECK(nvs_flash_erase());
        ESP_LOGI(tag, "Restarting the device....");
        readDataFromNvs();
        vTaskDelay(pdMS_TO_TICKS(10000));
        esp_restart();
        break;