espressif/esp-idf #6924: Esp-Idf new project => no serial port found.

lgLindstrom, 2021-04-22:
I have not yet managed to install a serial port on my new Laptop.  Yet I tried to create a new project assuming that I just would like to write some code. 

The command "New Project" gave me the following error ( and stopped )
 
`No serial ports found`

I guess this will be solved if I manage to install a serial port but I still think it should be possible to create a project.

I am running winndows 10 on a Laptop with only USB-C connectors. I have a USB-C to Micro-USB cable connected to a DevkitC board.

All my efforts to install FTDI drives have failed miserably. I have tried all  install variants from the instructions from Silicon Labs.

Can anyone with a similar Laptop (no com ports) give me some directions?

dobairoland, 2021-04-22:
@lgLindstrom I'm sorry but I think you posted this in the wrong place. ESP-IDF doesn't have any "New Project" command you are mentioning. ESP-IDF can build projects and a board is required only at the final step for flashing. Please refer to the documentation: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/get-started/index.html

lgLindstrom, 2021-04-22:
Ahh you are correct. I should posted at  VsCode Espress IDF extension page. 
I am so  sorry,, I close this issue.

closed 2021-04-22 by lgLindstrom