espressif/esp-idf #6926: MCPWM example fails and problems using MCPWM input capture after merge branch 'feature/update_spi_programming_guide_on_c3'

TobiasBoeschMiele, 2021-04-23:
## Environment

- Development Kit:      none
- Kit version (for WroverKit/PicoKit/DevKitC): 
- Module or chip used:  ESP32-WROOM-32
- IDF version (run ``git describe --tags`` to find it):
       v4.4-dev-960-gcf457d412
- Build System:         CMake
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     Windows
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: Yes, eclipse with esp-idf plugin
- Power Supply:         external 3.3V

## Problem Description

I use the input capture module and since https://github.com/espressif/esp-idf/commit/4028d4e083c6bc36be4f9e81967a78e42b6cfd86 it does not trigger an interrupt anymore. (I did not search for the cause in the stack. I don't have time to do this currently - I currently just stay away from that commit.
I found out that the example https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_basic_config also doesn't work anymore:

- fails already with a wrong GPIO configuration message on boot on UART)
- does not log input captured edges

right after the commit named above.
I believe there is a relationship between my problems and the problem of the example.

Since there were changes in definitions, has the configuration procedure changed for MCPWM? Is this documented somewhere?

Can someone please try to reproduce this issue with the example?

### Expected Behavior

Example works and MCPWM input capture creates interrupts on incoming edges.

### Actual Behavior

I found out that the example https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_basic_config also doesn't work anymore:

- fails already with a wrong GPIO configuration message on boot on UART)
- does not log input captured edges

right after the commit named above. Before that commit input capture module logs captured edges from the test signal when a loopback cable is used.

### Steps to reproduce

1. Use the MCPWM example with a current commit (Example: https://github.com/espressif/esp-idf/commit/cf457d412a7748139b77c94e7debe72ee86199af it just have to be after https://github.com/espressif/esp-idf/commit/4028d4e083c6bc36be4f9e81967a78e42b6cfd86)
2. Enable input capture
3. Connect loopback cable to used GPIO (test signal and input capture)
4. Look for a wrong GPIO configuration message on boot on UART
5. Look for missing logs for captured edges

### Code to reproduce this issue

https://github.com/espressif/esp-idf/tree/master/examples/peripherals/mcpwm/mcpwm_basic_config

## Debug Logs

(Altered GPIO used here: GPIO test signal uses GPIO 27 instead of 12)

Before https://github.com/espressif/esp-idf/commit/4028d4e083c6bc36be4f9e81967a78e42b6cfd86
```
[...]
I (0) cpu_start: Starting scheduler on APP CPU.
Testing MCPWM...
initializing mcpwm gpio...
intializing test signal...
Configuring Initial Parameters of mcpwm...
I (10) gpio: GPIO[27]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldow
n: 1| Intr:0
CAP0 : 40278240 us
CAP0 : 195316 us
CAP0 : 199992 us
CAP0 : 200000 us
CAP0 : 200000 us
[...]
```

After https://github.com/espressif/esp-idf/commit/4028d4e083c6bc36be4f9e81967a78e42b6cfd86
```
[...]
Testing MCPWM...
intializing test signal...
initializing mcpwm gpio...
I (336) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 1| Intr:0
E (336) gpio: io_num=34 can only be input
E (346) gpio: io_num=34 can only be input
Configuring Initial Parameters of mcpwm...
```

## Other items if possible

Alvin1Zhang, 2021-04-23:
@TobiasBoeschMiele Thanks for the very detailed report, we will look into.

L-KAYA, 2021-04-25:
@TobiasBoeschMiele 
We have tested the example mcpwm_basic_config, it works well on my computer.
Here's my suggestions:
+  Try to use the newest version of idf. It seems no fails in MCPWM example. But there are indeed two bugs. You can fix one of them by replacing the gpio pin 34 by other available pins like 33 & 22 in line 52 & 53 (the definition of GPIO_FAULT1_IN and GPIO_FAULT2_IN) in mcpwm_basic_config_example.c file. Another bug is in line 162, the parameter's type should be (void* args) referring to the format of isr function (it is not important in this case).
+ Guarantee that the capture function is enabled by the definition in line 30 while switching branches.
+ Check if you connect the right pins. The original wave generate pin is 12 and the capture pin is 23/25/26.
Thanks for your report again. 
Looking forward for your feedback.

TobiasBoeschMiele, 2021-04-26:
> We have tested the example mcpwm_basic_config, it works well on my computer.

Thanks for that. What does it mean, when it works? Have you connected two GPIO using a loopback cable and have you seen the output `CAP0 : 200000 us`? (IO23 and IO12 with default settings)

> * Try to use the newest version of idf.

As said above I did that already without success. I even nailed it down to the failing merge commit.

> * It seems no fails in MCPWM example

There are fails and they were named by my debug logs (IO34 errors) and you named them in your comment (IO34 config errors). They are just irrelevant for the issue I want to get more clarity into. (Missing input capture functionality)

> * You can fix one of them by replacing the gpio pin 34 by other available pins like 33 & 22 in line 52 & 53 (the definition of GPIO_FAULT1_IN and GPIO_FAULT2_IN) in mcpwm_basic_config_example.c file.

Did that. Of course that solves the IO34 config issue - but for me it solves only that issue - the missing input capture funcitonality is still present.

> * Another bug is in line 162, the parameter's type should be (void* args) referring to the format of isr function (it is not important in this case).

Changed and it is obvious - this does not help solving the issue.

> * Guarantee that the capture function is enabled by the definition in line 30 while switching branches.

I JUST change the esp-idf commit (commits were mentioned above) and the input capture functionality using a loopback cable on CAP0 and the configured IO works fine with one commit and does not with the other. So if this this setting would be not configured correctly it wouldn't work at all.

> * Check if you connect the right pins. The original wave generate pin is 12 and the capture pin is 23/25/26.

In general same as above.
I additionally reverted to the original example without changin the GPIO. So I use pins 12 and 23 now. I changed the loopback cable connection to pins to 12 and 23. I reverted the stack to commit d01c951d463ee833b67d4696b7e99ec5731a0bec and build it and I made sure the input capture works on and then JUST moved the esp-idf stack to 4028d4e083c6bc36be4f9e81967a78e42b6cfd86 and it doesn't work anymore. (Edges on pin 23 are not detected)

I also build the example on a computer of a collegue of mine with a current commit of the esp-idf stack. I kopied the binary files (app, bootloader and partition table) in my project and used idf.py flash to only flash the binary without build.
That software also does not has input capture functionality.

So either I (we already - me and my collegue) did something wrong with the build or config, or you did another test that I do.
Can you please print a full boot up log of the example (UART log) with the working input capture logging? That log should also contain the commit on boot up.

Here are mine:
d01c951d463ee833b67d4696b7e99ec5731a0bec 
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6708
load:0x40078000,len:14224
ho 0 tail 12 room 4
load:0x40080400,len:3672
0x40080400: _init at ??:?

entry 0x40080678
I (28) boot: ESP-IDF v4.4-dev-530-gd01c951d4-dirty 2nd stage bootloader
I (28) boot: compile time 11:49:35
I (30) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=07460h ( 29792)
 map
I (124) esp_image: segment 1: paddr=00017488 vaddr=3ffb0000 size=02a3ch ( 10812)
 load
[... 105 lines cut ...]
I (217) cpu_start: Application information:
I (221) cpu_start: Project name:     mcpwm_basic_config
I (227) cpu_start: App version:      1
I (232) cpu_start: Compile time:     Apr 26 2021 12:10:59
I (238) cpu_start: ELF file SHA256:  b172d32b613228eb...
I (244) cpu_start: ESP-IDF:          v4.4-dev-534-g4028d4e08-dirty
I (251) heap_init: Initializing. RAM available for dynamic allocation:
I (258) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (264) heap_init: At 3FFB3270 len 0002CD90 (179 KiB): DRAM
I (270) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (277) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (283) heap_init: At 4008B14C len 00014EB4 (83 KiB): IRAM
I (290) spi_flash: detected chip: generic
I (294) spi_flash: flash io: dio
W (298) spi_flash: Detected size(4096k) larger than the size in the binary image
 header(2048k). Using the size in the binary image header.
I (312) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
Testing MCPWM...
intializing test signal...
initializing mcpwm gpio...
I (337) gpio: GPIO[12]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldo
wn: 1| Intr:0
Configuring Initial Parameters of mcpwm...
```

TobiasBoeschMiele, 2021-06-14:
This still prevents me from doing an esp-idf update.
Can someone please tell me give me an update regarding this topic?

suda-morris, 2021-06-15:
> I reverted the stack to commit d01c951 and build it and I made sure the input capture works on

We reverted back the MCPWM gpio configuration on master, which will be synced to Github soon. For now, can you try if this change can help?
```c
esp_err_t mcpwm_gpio_init(mcpwm_unit_t mcpwm_num, mcpwm_io_signals_t io_signal, int gpio_num)
{
    if (gpio_num < 0) { // ignore on minus gpio number
        return ESP_OK;
    }

    ESP_RETURN_ON_FALSE(mcpwm_num < SOC_MCPWM_GROUPS, ESP_ERR_INVALID_ARG, TAG, MCPWM_GROUP_NUM_ERROR);
    ESP_RETURN_ON_FALSE(GPIO_IS_VALID_GPIO(gpio_num), ESP_ERR_INVALID_ARG, TAG, MCPWM_GPIO_ERROR);

    if (io_signal <= MCPWM2B) { // Generator output signal
        ESP_RETURN_ON_FALSE(GPIO_IS_VALID_OUTPUT_GPIO(gpio_num), ESP_ERR_INVALID_ARG, TAG, MCPWM_GPIO_ERROR);
        gpio_set_direction(gpio_num, GPIO_MODE_OUTPUT);
        int operator_id = io_signal / 2;
        int generator_id = io_signal % 2;
        esp_rom_gpio_connect_out_signal(gpio_num, mcpwm_periph_signals.groups[mcpwm_num].operators[operator_id].generators[generator_id].pwm_sig, 0, 0);
    } else if (io_signal <= MCPWM_SYNC_2) { // External sync input signal
        gpio_set_direction(gpio_num, GPIO_MODE_INPUT);
        int ext_sync_id = io_signal - MCPWM_SYNC_0;
        esp_rom_gpio_connect_in_signal(gpio_num, mcpwm_periph_signals.groups[mcpwm_num].ext_syncers[ext_sync_id].sync_sig, 0);
    } else if (io_signal <= MCPWM_FAULT_2) { // Fault input signal
        gpio_set_direction(gpio_num, GPIO_MODE_INPUT);
        int fault_id = io_signal - MCPWM_FAULT_0;
        esp_rom_gpio_connect_in_signal(gpio_num, mcpwm_periph_signals.groups[mcpwm_num].detectors[fault_id].fault_sig, 0);
    } else if (io_signal >= MCPWM_CAP_0 && io_signal <= MCPWM_CAP_2) { // Capture input signal
        gpio_set_direction(gpio_num, GPIO_MODE_INPUT);
        int capture_id = io_signal - MCPWM_CAP_0;
        esp_rom_gpio_connect_in_signal(gpio_num, mcpwm_periph_signals.groups[mcpwm_num].captures[capture_id].cap_sig, 0);
    }
    gpio_hal_iomux_func_sel(GPIO_PIN_MUX_REG[gpio_num], PIN_FUNC_GPIO);
    return ESP_OK;
}
```

TobiasBoeschMiele, 2021-06-16:
Thanks for that suggestion. I tried to use the change in the current master commit ([1d7068e4be430edd92bb63f2d922036dcf5c3cc1](https://github.com/espressif/esp-idf/commit/1d7068e4be430edd92bb63f2d922036dcf5c3cc1)).
Compiling fails with:
`driver/mcpwm.c:102:5: error: implicit declaration of function 'ESP_RETURN_ON_FALSE'`

Can I replace that macro with something else?

closed 2021-07-16 by espressif-bot

TobiasBoeschMiele, 2021-07-27:
I can confirm the example works again with

- the example code from 6df89fa and
- a recent esp-idf stack (c69f0ec3292de2b9df5554405744296333d0feb2)

Updating the stack to c69f0ec3292de2b9df5554405744296333d0feb2 also makes my app work again.

Thank you for your help.