espressif/esp-idf #6928: Wi-Fi reconnection cause TCP/IP connection drop even with ethernet connected

mcarrier-nov, 2021-04-23:
## Environment

- Module or chip used:  [ESP32-WROOM-32U]
- IDF version [4.1.1]
- Build System:         [idf.py]
- Compiler version: [8.4.0]
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command Prompt].
- Using an IDE?: [No (please give details)]
- Power Supply:         [external 5V]

## Problem Description

Actual setup:
We are using both Wi-Fi and ethernet (lan8720). 
We establish connection to remote HTTP server using WebSocket.
It works perfectly if connected to the access point with ethernet cable connected, both have access to the same network.
Now, if I stop the access point or remove the antenna on the board it will cause Wi-Fi disconnection and force WebSocket to reconnect to the server using the ethernet connection this time and that's fine. 
But every time Wi-Fi attempt to reconnect it cause TCP/IP connection to get dropped. The attempt by itself is enough to cause disconnection.
This problem happens only on remote server (internet) but doesn't happens if the server is hosted locally.

I tried using IDF 4.2 to test without success, I also tried to change network priority to ethernet.

My hypothesis being that somehow network priority change cause some change to LWIP level that break existing connection when Wi-Fi attempt to connect. I tried to make two sub-net network with a gateway to see if connection will be dropped like when we are connected to a remote server (internet).
Tried also to check the network traffic using Wireshark and I see no evidence that connection loss have anything to do with remote peer. It seems like the ESP32 just forget about the connection.

If anybody have any idea or hypothesis I could test feel free to submit it.

### Expected Behavior
 - Wi-Fi reconnection attempt shouldn't cause disconnection on the ethernet interface
### Actual Behavior
- It drop TCP/IP connection

## Other items if possible
- [X ] sdkconfig file (attach the sdkconfig file from your project folder)

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6364900/sdkconfig.zip)

yuanjianmin, 2021-04-23:
@mcarrier-nov Could you please check the traffic connect to remote server, which interface the data go through, Wi-Fi or ethernet. You can use Wireshark to capture the TCP/IP connection and see the source IP address belonging to which interface.

mcarrier-nov, 2021-04-23:
I can confirm it went through the ethernet correctly when it is disconnected from the access point because it is able to connect to the server for a few seconds until wi-fi connection attempt happens again and cause connection drop.
If I deactivate the wi-fi (I can do it while running without recompiling) the websocket connection works flawlessly.

yuanjianmin, 2021-04-25:
@mcarrier-nov Okay, Could you please open the `Debug` level log(idf.py menuconfig->Component config->Log output->Default log verbosity (Debug)) and paste the log. Also, you can checkout branch to master and test this.

mcarrier-nov, 2021-04-27:
@yuanjianmin 
Unfortunately I can't compile my code with "master" because modification required to compile are too big and I'm not allowed to continue. We will deal with the problem another way for now unfortunately.

closed 2021-04-27 by mcarrier-nov

pdemianczuk07, 2021-05-12:
Hi, @mcarrier-nov  do you solve the problem? I have also problem with Eth (w5500)

mcarrier-nov, 2021-05-13:
> Hi, @mcarrier-nov did you solve the problem? I have also problem with Eth (w5500).
> in my issue , when I set debug_level > Info, the problem didn't show.

Hi, unfortunately I didn't solve the problem.
I'm on tight schedule so they decided to drop the functionality for now.