espressif/esp-idf #6930: GPIO for antenna configuration is not working anymore

aircable, 2021-04-23:
- Module or chip used: ESP32-U4WDH and ESP32-D0WD-V3
   - IDF version v4.4-dev-960
   - Build System: idf.py
   - Compiler version xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0

## Problem Description

This code snippet  uses GPIOs for showing with LEDs when the Wifi transmitter/receiver are active.

```
        config_gpio.gpio_cfg[0].gpio_select = 1; 
        config_gpio.gpio_cfg[0].gpio_num = LIGHTS_GPIO_LED_BLUE;       // Antenna TX switch -- blue LED
        config_gpio.gpio_cfg[1].gpio_select = 1; 
        config_gpio.gpio_cfg[1].gpio_num = LIGHTS_GPIO_LED_RED;      // Antenna RX switch -- red LED
        config_gpio.gpio_cfg[2].gpio_select = 0; 
        config_gpio.gpio_cfg[3].gpio_select = 0; 

        wifi_ant_config_t m_ant;
        m_ant.rx_ant_mode = WIFI_ANT_MODE_ANT1; 
        m_ant.rx_ant_default = WIFI_ANT_MODE_ANT1; 
        m_ant.tx_ant_mode = WIFI_ANT_MODE_ANT0; 
        m_ant.enabled_ant0 = 1;
        m_ant.enabled_ant1 = 2;
        if(( err = esp_wifi_set_ant_gpio( &config_gpio )) != ESP_OK ) {
            fprintf( stderr, "%s: wifi gpio: %s\n", TAG, esp_err_to_name( err ));
        }
        if(( err = esp_wifi_set_ant( &m_ant )) != ESP_OK ) {
            fprintf( stderr, "%s: wifi ant: %s\n", TAG, esp_err_to_name( err ));
        }
```

With the latest IDF-4.4, the LEDs are being "captured" by the esp_wifi_set_ant_gpio function. Previous status is being reset and both LEDs go out after the function call. But neither transmit or receive events will turn on any LEDs. 

I can confirm that it worked before on IDF-4.3 but not on IDF-4.4. Double confirmed by installing idf-4.3-1197.

## Debug Logs

There are no error messages.

ojousima, 2022-02-24:
Is this issue being worked on?

mhdong, 2022-03-03:
hi @aircable  Could you tell me the value of the macro LIGHTS_GPIO_LED_BLUE and LIGHTS_GPIO_LED_RED.

aircable, 2022-03-13:
#define LIGHTS_GPIO_LED_RED                GPIO_NUM_25
#define LIGHTS_GPIO_LED_BLUE               GPIO_NUM_5

MaxwellAlan, 2022-07-11:
Hi @aircable 
GPIO 2 & 25 are both using for Antenna and not real connect to ESP32-D0WD-V3 board's pins. And we test on master branch, these apis works, maybe you can refer to https://github.com/MaxwellAlan/esp32_dual_ant_demo or some more details from this PR: https://github.com/espressif/arduino-esp32/pull/6841

Alvin1Zhang, 2022-10-18:
Thanks for reporting, will close due to short of feedback, feel free to reopen. Thanks.

closed 2022-10-18 by Alvin1Zhang