espressif/esp-idf #6935: Build failure in esp_netif_lwip_ppp.c if CONFIG_LWIP_IPV6 is not set

AxelLin, 2021-04-25:
## Environment

Module or chip used: ESP32-WROOM-32E
IDF version: v4.4-dev-960-gcf457d412a77
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux
Power Supply: USB

## Problem Description

Build failure if CONFIG_LWIP_IPV6 is not set

### Expected Behavior

It should be able to build with CONFIG_LWIP_IPV6=n

### Steps to reproduce

Use pppos_client example to reproduce the build issue:

1. cd examples/protocols/pppos_client and add below line to CMakeLists.txt
   set(EXCLUDE_COMPONENTS "coap" "asio")

2. $ idf.py menuconfig
   Component Config -> LWIP -> Enable IPv6 (Uncheck it)

3. idf.py build
   Then got below build error:
```
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c: In function 'on_ppp_status_changed':
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:76:54: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
                 evt.ip_info.ip.addr = pppif->ip_addr.u_addr.ip4.addr;
                                                      ^~~~~~
                                                      addr
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:77:49: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
                 evt.ip_info.gw.addr = pppif->gw.u_addr.ip4.addr;
                                                 ^~~~~~
                                                 addr
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:78:59: error: 'ip_addr_t' {aka 'struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
                 evt.ip_info.netmask.addr = pppif->netmask.u_addr.ip4.addr;
                                                           ^~~~~~
                                                           addr
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:82:43: error: 'ip_addr_t' {aka 'const struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
                     ns1.addr = (*dest_ip).u_addr.ip4.addr;
                                           ^~~~~~
                                           addr
/home/axel/esp/esp-idf/components/esp_netif/lwip/esp_netif_lwip_ppp.c:86:43: error: 'ip_addr_t' {aka 'const struct ip4_addr'} has no member named 'u_addr'; did you mean 'addr'?
                     ns2.addr = (*dest_ip).u_addr.ip4.addr;
                                           ^~~~~~
                                           addr
```

ESP-YJM, 2021-04-25:
@AxelLin Thanks for reporting this issue. Yes, there will indeed be compilation problems, because IPv6 macros are not well isolated. I will look into and give you a patch.

AxelLin, 2021-04-25:
@ESP-YJM 
I currently fix it by below changes:
https://gist.github.com/AxelLin/08072602efc5a7b682d9a2bff69b29af
You may have a better fix.

ESP-YJM, 2021-04-26:
@AxelLin Sorry for the late reply, you can try to see if this patch can solve your problem.
[0001-ppp-Fix-disable-IPv6-will-make-esp_netif_lwip_ppp-bu.zip](https://github.com/espressif/esp-idf/files/6374007/0001-ppp-Fix-disable-IPv6-will-make-esp_netif_lwip_ppp-bu.zip)

AxelLin, 2021-04-26:
Hi @ESP-YJM 
It works, thank you.
BTW, Hope the fix also applies other v4.x branches.

ESP-YJM, 2021-04-26:
@AxelLin Yes, we will backport this patch to other v4.x branch which include IPv6 disable function.

closed 2021-06-06 by espressif-bot

AxelLin, 2021-06-07:
Just to remind v4.3-rc also needs this fix, hope the official v4.3 release can include the fix.

AxelLin, 2021-06-15:
Unfortunately, the v4.3 release does not include this fix.

ESP-YJM, 2021-06-15:
I'm sorry that the fix not synchronize to release/v4.3. The fix has already created internal MR for release/v4.3. I will promote the integration of the MR, please be patient.

AxelLin, 2021-06-15:
@ESP-YJM,
this is not a single case.
I even remind it here: https://www.esp32.com/viewtopic.php?f=25&t=21325#p77852
But **all** the fixes are not included in official release.
I think there are something wrong in the process of taking patches. Especially for trivial bug fixes.

ESP-YJM, 2021-06-15:
@AxelLin I think this fixes have not caught up with the v4.3 release version published, and this fix may be included in the v4.3.1 release version.