espressif/esp-idf #6939: UARTs available in ESP32-WROVER-E

carlessole, 2021-04-26:
Hello,
We are starting a project using a ESP32-WROVER-E module and we would like to know how many UARTs are available and which pins can be used for that.

After looking the [datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf#page=20) it seems that there is only one UART available (the UART 0) but I'm pretty sure that two other ones (as it seems so in [esp32 chip datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf ) ) must be available:

[image]

In some forums we have seen that **ANY GPIO** can be used as a UART pin. Is that true? Is there no limitations?
And how it has to be configured? Just using [uart_set_pin](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv412uart_set_pin11uart_port_tiiii) function?

Lot of thanks,

Carles

negativekelvin, 2021-04-26:
Yes using uart_set_pin almost any gpio are usable

https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html

carlessole, 2021-04-27:
Thanks @negativekelvin 
And so, 3 UARTs can be used, right?

AshUK, 2021-04-27:
There are three UARTs they are all useable. However unlike other microcontrollers there are some limitations. I recommend checking out technical reference manual to make sure your design will work seamlessly. Here are some issue we have come across.

https://github.com/espressif/esp-idf/issues/2317 https://github.com/espressif/esp-idf/issues/6152 https://github.com/espressif/esp-idf/issues/6186

General questions like this are better suited to the esp32 forums.

carlessole, 2021-04-27:
Thanks @AshUK .
We have already check the technical reference manual but we are not quite sure which are the recommended pins as in the [ESP-WROVER-E Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf#page=9) just the UART 0 is documented:

[image]

Thanks

AshUK, 2021-04-27:
@carlessole its in Table 1: Pin Description, labeled; U0RXD, U0TXD, U1RXD, U1TXD, U2RXD, U2TXD.

carlessole, 2021-04-27:
@AshUK, which document are you talking about? I am looking at [ESP-WROVER-E Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf)

And there, I think Pin Description is in table 3:
[image]

And there is nothing about U1RXD, U1TXD, U2RXD, U2TXD. There just appear U0RXD, U0TXD:

[image]

AshUK, 2021-04-27:
@carlessole you are right it is no included probably for verbosity. its in the ESP-32 Data sheet.

Section 12.1 of the WROVER-E data sheet has a list of must read documents. I highly recommend reading through all the documents and resources.

AshUK, 2021-04-27:
If the pin you wish to use is not expose by the module you can use IO multiplexing to assign it to any pin. This is explain in the ESP-32 TRM Chapter 4. Also see https://www.esp32.com/viewtopic.php?t=3569

carlessole, 2021-04-27:
@AshUK thanks for the quick responses.

We already read them, but for the ESP32-WROVER-E we have the doubt if the recommended UART1 and UART2 pins are available.
I will try to explain what we found:
On [ESP32 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf) table 1:
[image]

So it seems we must use for UART 2 (pins 25,26,32,33) and for UART 1 (pins 28,29,30,31)

But then on [ESP32-WROVER-E Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf#page=20) (page 20) is like these pins (25..33) are not available in the module (and used by flash and PSRAM)

Thanks

carlessole, 2021-04-27:
> 
> 
> If the pin you wish to use is not exposed by the module you can use IO multiplexing to assign it to any pin. This is explain in the ESP-32 TRM Chapter 4. Also see https://www.esp32.com/viewtopic.php?t=3569

Ok, I will take a look and see if it solves our problem. Thanks

carlessole, 2021-04-27:
From [ESP32 Datasheet](https://www.espressif.com/sites/default/files/documentation/esp32_datasheet_en.pdf#page=35)
> ESP32 has three UART interfaces, i.e., UART0, UART1 and UART2, which provide asynchronouscommunication (RS232 and RS485) and IrDA support, communicating at a speed of up to 5 Mbps. UARTprovides hardware management of the CTS and RTS signals and software flow control (XON and XOFF). All ofthe interfaces can be accessed by the DMA controller or directly by the CPU

[image]

And from [https://www.esp32.com/viewtopic.php?t=3569](https://www.esp32.com/viewtopic.php?t=3569):

> The issue is that there are two ways of mapping pins: the GPIO mux and the IO_MUX. The GPIO mux allows you to map a peripheral to any pin, but it has the disadvantage that it introduces an APB clock (80MHz) tick delay in the signal; for some purposes this can be disadvantageous. Because of this, we also have the IO_MUX, which can map the signals of some peripherals to a very limited amount of IO pins; this reduces flexibility but because the IO_MUX does not have the delay, it can be useful for some purposes.

So I understand that ideally it is better to work with the mapping done by IO_MUX, that means using pins (25..33). But as in ESP32-WROVER-E these pins are used for other "internal" proposes (flash, PSRAM) it is needed to map other pins to the UARTS, using GPIO mux. This second way (GPIO mux) must work but we could experience some unexpected behaviour (maybe) introduced bytthat APB clock.

carlessole, 2021-04-27:
> 
> 
> If it helps we use a WROVER-E module and make use of all three UARTs and we use the following the pins:
> 
> UART0 is unchanged as exposed by the module.
> 
> UART1_TX GPIO_NUM_27
> UART1_RX GPIO_NUM_14
> 
> UART2_TX GPIO_NUM_2
> UART2_RX GPIO_NUM_15

Thanks! We are gonna try these ones for our prototype

closed 2021-06-03 by carlessole

carlessole, 2021-11-02:
Hello @Andres-San ,

Yes. It has worked for me. My project is currently using following UART setup:

[image]

Which pins are you trying to use?
Have in mind that some GPIO's (>= GPIO_NUM_34) can just be used as input (and so they will not work as a UART pin):
[image]

Andres-San, 2021-11-03:
@carlessole Thanks! Im on the Schematic yet, but those same doubts assaulted me every night haha. 

After reading and reading, i have come to the conclusion that the person who wrote "Any GPIO" in the ESP32 DataSheet is wrong (or atleast missleading), because the IO_MUX shows otherwise. 

For instance GPIO 15 cant be used as I2C on any of its functions (red), but MTD0, GPIO0, GPIO2 and GPIO4 can (blue). 

[image]

Is this the same conclusion you've come to? 

Best from Chile!

negativekelvin, 2021-11-03:
@Andres-San gpio mux has limited options but gpio matrix can connect almost any pin to almost any function.  Research gpio matrix.

carlessole, 2021-11-03:
Hi @Andres-San,

As said here: https://www.esp32.com/viewtopic.php?t=3569

"Any GPIO" is true always that you hardware (or module) is not using it for some other reasons. So, if you have the ESP- WROVER module you won't have all the GPIO available because some of them will be routed internally to SPI RAM and other peripherals.

But all the GPIOs accessible from outside de ESP-WROVER can be used as a GPIO if not other use is desired (as JTAG or whatever). Take a look here (https://www.espressif.com/sites/default/files/documentation/esp32-wrover-e_esp32-wrover-ie_datasheet_en.pdf#page=20):

[image]

Andres-San, 2021-11-03:
@negativekelvin and @carlessole 

Thank you so much for your clarification!, effectively now im missing some direct IO_MUX pins, because i have to use x3 UART, but only one is "wired" directly via IO_MUX (red) into Wrover-E, and for the rest i'll have to use GPIO_Matrix right? 

[image]

Is this correct? if it is, now i'll have to look further on how to use GPIO matrix on code. 

Best!

carlessole, 2021-11-04:
Hi @Andres-San ,

I don't know If I have inderstood you ok, but if your intention is to use GPIO9/10 (UART1) and GPIO16/17(UART2) that is not possible because those GPIOs are not available outside the WROOVER-E module.

You need to choose any GPIO available and free. And to use them is it as easy as (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html#_CPPv412uart_set_pin11uart_port_tiiii):

[image]

[image]

Instead of GPIO4/0 you can choose the one that you prefer (if it is available on WROOVER-E):
[image]

For example:

GPIO 23, GPIO22, GPIO19, GPIO18.....

**/!\ GPIO number and PIN number is not the same. You need to use GPIO number.**

Andres-San, 2021-11-04:
Hi @carlessole,

Yes thank you for your patience!  I fully understand now =)

Best from Chile

igrr, 2023-05-16:
@markusb237 If you want to use GPIO35, please pass 35 as the 3rd argument to this function:
```c++
#include <Arduino.h>

void setup() {
  Serial2.begin(115200,SERIAL_8N1,35,12);   // will use GPIO35 as RXD and GPIO12 as TXD
}
```

(All APIs in Arduino and ESP-IDF use GPIO numbers; In software, we never use the numbers of pins/pads of the chip/module.)

markusb237, 2023-05-16:
Doh! - Of course, how could I miss this...
Should explain why 7 leads to errors.

Thanks!

Regards