espressif/esp-idf #6942: ESP32-S2 is restarted after waking up from light sleep

EduardShaid, 2021-04-26:
I am using the latest version 4.3-beta3.
I have a code where I first turn off WiFi using the "wifi_stop()" function and then turn on light-sleep using the "light_sleep()" function. After waking up from light sleep (waking up by UART), I configure WiFi as STA and try to scan the network using the esp_wifi_scan_start() function. At this moment, the chip is restarted (reason rst: 0xc (RTC_SW_CPU_RST)).

pseudocode:

```
wifi_stop()
light_sleep()
...
(waking up by UART)
...

(init STA)
esp_wifi_scan_start() 
(here restart)

```

Tell me how to cope with this problem?

Espressif-liuuuu, 2021-04-27:
@EduardShaid  Hi, thanks for your reporting. Could you please provide us with specifical names of the functions you have used?

EduardShaid, 2021-04-27:
I give the exact sequence of commands:

// I turn off Wifi
`wifi_stop(); `

// go to light sleep
```
char buf[32] = {0};
    esp_sleep_disable_wakeup_source(ESP_SLEEP_WAKEUP_ALL);
    ESP_ERROR_CHECK( uart_set_wakeup_threshold(UART1, 3) );
    ESP_ERROR_CHECK( esp_sleep_enable_uart_wakeup(UART1) );
    esp_light_sleep_start();

    esp_sleep_wakeup_cause_t cause = esp_sleep_get_wakeup_cause();
    const char *cause_str;
    switch (cause) {
    	case ESP_SLEEP_WAKEUP_UART:
    		cause_str = "UART1";
      break;
    	default:
    		cause_str = "unknown";
    		sprintf(buf, " *%d\r\n", cause);
    		esp_at_port_write_data((uint8_t *)buf, strlen(buf));
    }
    sprintf(buf, "Woke up from: %s\r\n", cause_str);
    esp_at_port_write_data((uint8_t *)buf, strlen(buf));
```

// here I wake up when I enter a character on UART1
//  then I move WiFi into the active state of STA
```
wifi_config_t wifi_config = {
	    	.sta = {
	          		.ssid = "",
	           		.password = "",
				.channel = 0,
	            	.scan_method = WIFI_ALL_CHANNEL_SCAN,
	            	.sort_method = WIFI_CONNECT_AP_BY_SIGNAL,
	            	.threshold.rssi = 0,
			.listen_interval = 0						
	            }
	};

	ESP_ERROR_CHECK( esp_wifi_set_storage(WIFI_STORAGE_RAM) );
	ESP_ERROR_CHECK( esp_wifi_set_mode(WIFI_MODE_STA) );
	ESP_ERROR_CHECK( esp_wifi_set_config(ESP_IF_WIFI_STA, &wifi_config) );
	ESP_ERROR_CHECK( esp_wifi_set_ps(WIFI_PS_NONE) );			
	ESP_ERROR_CHECK( esp_wifi_start() );
```
// I give a command to scan the AP and after it I immediately get a restart!
```
wifi_scan_config_t scan_config = { 0 };
    	scan_config.show_hidden = 1;
    	scan_config.channel = 0;
    	scan_config.scan_type = WIFI_SCAN_TYPE_PASSIVE;
    	scan_config.scan_time.passive = 200;
    	ESP_ERROR_CHECK( esp_wifi_scan_start(&scan_config, false) );
```

…
rst:0xc (RTC_SW_CPU_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
…

Espressif-liuuuu, 2021-04-27:
@EduardShaid  Many thx! BTW is everything allright if you remove all of the part of codes entering and leaving lightsleep?

EduardShaid, 2021-04-27:
Yes, if I don't use the part of the code that puts the chip into sleep mode, then everything works fine. I can repeat this code periodically(turn off and then turn on WiFi) and it works (I scan nearby APs without problems).

Espressif-liuuuu, 2021-04-27:
Understood, though I did not reproduce your problem in my place. 
May I see your log and how do you repeat your code?  Since you called esp_wifi_scan_start(&scan_config, **false**) and it did not block your code, Im afraid that somewhere cause reboot after esp_wifi_scan_start returns.

EduardShaid, 2021-04-27:
There is actually nothing in the log, only the list of scanned APs is visible and how the chip is reset (in sleep mode).
I repeat the above code periodically by submitting my AT commands.
The esp_wifi_scan_start() function is non-blocking - I get the AP list via the wifi_event_handler() callback function in the case of WIFI_EVENT_SCAN_DONE.
The problem is, of course, difficult to find so quickly. I thought that someone might have faced such a problem and be able to give a search direction.

EduardShaid, 2021-04-29:
I tried to debug the program and this is what it turned out to be:

after I am in sleep mode, the program goes into restart when the "esp_wifi_start()" function is called.
And the "panic_abort()" function is called which is called from the "abort()" function, which in turn is called from the "lock_acquire_generic()" function.

In the "lock_acquire_generic()" function itself, the "abort()" function is called here:

/ * In ISR Context * /
if (mutex_type == queueQUEUE_TYPE_RECURSIVE_MUTEX) {
   **abort ();**
}

projectgus, 2021-05-03:
Hi @EduardShaid,

Is it possible to get the entire backtrace that shows exactly where the call to lock_acquire_generic() came from, please? This is necessary in order to find out the root cause, it looks like possibly a function was called in an ISR context when it shouldn't have been.

If using the "idf.py monitor" command then the full backtrace should be automatically decoded on the console by the idf_monitor tool. Otherwise it is possible (although more difficult) to run the command manually, [see here for details](https://docs.espressif.com/projects/esp-idf/en/v4.3-beta3/esp32s2/api-guides/tools/idf-monitor.html?highlight=backtrace#automatic-address-decoding).

EduardShaid, 2021-05-03:
I transferred the project to the latest version ESP_IDF 4.2.1 and this problem disappeared, that is, now I turn WiFi on and off after sleeping without restarting the chip.
It is likely that in version ESP_IDF 4.3-beta3 there is an error inside the Wifi stack related to the call of the kernel synchronization primitives(mutex) from the ISR.

Thank you.

AxelLin, 2022-02-18:
> It is likely that in version ESP_IDF 4.3-beta3 there is an error inside the Wifi stack related to the call of the kernel synchronization primitives(mutex) from the ISR.

@EduardShaid 
Did you try v4.3.2 release? Just wondering if this still an issue in v4.3+ branch.

jack0c, 2022-04-14:
@EduardShaid @AxelLin It is possible that it was a  bug from Wi-Fi Stack. That is calling sleep locks in TBTT interrupt. But this behavior no longer exists anymore.  I think this is fixed in 4.3 version. I suggest close the issue. Please reopen it if you still find the issue.

closed 2022-04-14 by jack0c

jack0c, 2022-04-14:
Another possibility is this is same as https://github.com/espressif/esp-idf/issues/7576. I will keep https://github.com/espressif/esp-idf/issues/7576 open.