espressif/esp-idf #6943: SPI Slave not Realtime

nikhilavrio, 2021-04-26:
## Environment

- Module or chip used:  ESP32-WROVER-IE
- IDF version: v4.2-351-gce2c615bb
- Build System:        idf.py
- Compiler version: 8.4.0
- Operating System:     macOS
- Power Supply:        external 3.3V

## Problem Description

I am interfacing with ADC to capture waveforms. ADC sends data as SPI Master at 8Mhz / 4Mhz (configurable) to ESP32 (SPI Slave). Waveform is a 50Hz sine wave connected to signal generator.

The data received in ESP32 is then plotted. I am getting spikes every  3-4 cycle.

Plotting the SPI Protocol data of ADC (using a protocol analyzer) results in a perfect sine wave.

So there is definitely something wrong with SPI Slave driver. Can you help ?

SPI Mode (as per datasheet of ADC) should be 3

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

nikhilavrio, 2021-04-26:
Here is Logic Analyzer screenshots.
[image]
[image]
[image]
[image]

Alvin1Zhang, 2021-04-27:
Thanks for the very detailed report.

Icarus113, 2021-05-24:
Hi @nikhilavrio ,

For SPI protocol, master determines when to start/finish a transaction, no matter if slave has prepared ready TX/RX/both buffer(s) or not. So usually there is a synchronization way to let master know slave is ready. 

Therefore, if you directly use SPI slave driver to receive data from ADC without synchronization, you may not get complete data in some conditions. Let's take an easy example:
1. Slave side prepares a 28-Byte buffer.
2. Master lowers down the CS, the transaction starts.
3. Slave side called `spi_slave_transmit`. When the slave-prepared-buffer is actually loaded to the SPI hardware, we assume that at this time, master has already sent 10 Bytes data to the SPI bus.
4. Slave side gets the rest 18 Bytes data from the SPI bus.
5. Master finish sending all the data, and it sets CS to high. 
6. `spi_slave_transmit` will return due to the low CS. 

Finally, for the 28-Byte buffer, the first 18 Bytes data are actually got from the master, whereas the rest 10 Bytes of the buffer are **unchanged**. At this point, if you plot all the buffer, you'll get a wrong plot. 

However, you can check the `trans_len`  of `spi_slave_transaction_t` indicating the actual received bytes per transaction. With respect to the above situation, the `trans_len` would be 18*8 (in bits).

For synchronization, the goal is to make sure that slave has actually loaded its buffer to the SPI hardware (time point A) before master lowers down the CS line. We have a callback to indicate the time point A  - `post_setup_cb` in `spi_slave_interface_config_t` will be called at this time.

closed 2022-07-24 by ginkgm