espressif/esp-idf #6945: Crash when using Wi-Fi + UART after initializing UART driver

nathanieloliveira, 2021-04-26:
## Environment

- Development Kit: NodeMCU
- Kit version: v1.2
- Module or chip used: ESP32-WROVER-E, 16 MB Flash, 4MB SPIRAM. Was soldered onto NodeMCU board.
- IDF version: v4.2
- Build System: idf.py
- Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System: Linux
- Using an IDE?: No
- Power Supply: USB

## Problem Description

A Crash occurs after installing UART driver when trying to allocate memory inside the wi-fi task.

### Expected Behavior

UART + Wi-Fi should not cause a crash

### Actual Behavior

Crashes outside user code when trying to allocate memory.

### Steps to reproduce

1. Start ESP32
2. Have it connect to a wifi network
3. Crash happens some time after uart initialization

Setup is only the devkit connected to a usb port on a computer.

### Code to reproduce this issue

Gist link: https://gist.github.com/nathanieloliveira/2f2b24185ba5c691fe1906def2e5957a/raw/0005b01b08bf9a0d8f7948cd353e864c51920663/main.c

## Debug Logs
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4
load:0x3fff0034,len:7100
load:0x40078000,len:13684
load:0x40080400,len:4632
0x40080400: _init at ??:?

entry 0x400806f4
I (29) boot: ESP-IDF v4.2 2nd stage bootloader
I (29) boot: compile time 15:41:16
I (29) boot: chip revision: 3
I (32) boot_comm: chip revision: 3, min. bootloader chip revision: 0
I (39) boot.esp32: SPI Speed      : 80MHz
I (44) boot.esp32: SPI Mode       : DIO
I (48) boot.esp32: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 otadata          OTA data         01 00 0000f000 00002000
I (84) boot:  2 phy_init         RF data          01 01 00011000 00001000
I (91) boot:  3 factory          factory app      00 00 00020000 00200000
I (99) boot:  4 ota_0            OTA app          00 10 00220000 00400000
I (106) boot:  5 ota_1            OTA app          00 11 00620000 00400000
I (114) boot: End of partition table
I (118) boot: Defaulting to factory image
I (123) boot_comm: chip revision: 3, min. application chip revision: 0
I (130) esp_image: segment 0: paddr=0x00020020 vaddr=0x3f400020 size=0x18460 ( 99424) map
I (175) esp_image: segment 1: paddr=0x00038488 vaddr=0x3ffbdb60 size=0x04e28 ( 20008) load
I (183) esp_image: segment 2: paddr=0x0003d2b8 vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at /home/nth/monitor20/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (184) esp_image: segment 3: paddr=0x0003d6c4 vaddr=0x40080404 size=0x02954 ( 10580) load
I (195) esp_image: segment 4: paddr=0x00040020 vaddr=0x400d0020 size=0x7477c (477052) map
0x400d0020: _stext at ??:?

I (370) esp_image: segment 5: paddr=0x000b47a4 vaddr=0x40082d58 size=0x127bc ( 75708) load
0x40082d58: psram_enable at /home/nth/monitor20/esp-idf/components/esp32/spiram_psram.c:935

I (414) boot: Loaded app from partition at offset 0x20000
I (414) boot: Disabling RNG early entropy source...
I (415) psram: This chip is ESP32-D0WD
I (419) spiram: Found 64MBit SPI RAM device
I (424) spiram: SPI RAM mode: flash 80m sram 40m
I (429) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (436) cpu_start: Pro cpu up.
I (440) cpu_start: Application information:
I (445) cpu_start: Project name:     esp32uartwifibug
I (451) cpu_start: App version:      1
I (455) cpu_start: Compile time:     Apr 26 2021 18:00:46
I (461) cpu_start: ELF file SHA256:  f8476f5bbebac9d3...
I (467) cpu_start: ESP-IDF:          v4.2
I (472) cpu_start: Starting app cpu, entry point is 0x40081934
0x40081934: call_start_cpu1 at /home/nth/monitor20/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (1371) spiram: SPI SRAM memory test OK
I (1371) heap_init: Initializing. RAM available for dynamic allocation:
I (1372) heap_init: At 3FFAE6E0 len 0000F480 (61 KiB): DRAM
I (1378) heap_init: At 3FFC7700 len 00018900 (98 KiB): DRAM
I (1384) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1391) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1397) heap_init: At 40095514 len 0000AAEC (42 KiB): IRAM
I (1403) cpu_start: Pro cpu start user code
I (1408) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (1429) spi_flash: detected chip: gd
I (1429) spi_flash: flash io: dio
I (1429) esp_core_dump_uart: Init core dump to UART
I (1432) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (1443) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
I (1503) wifi:wifi driver task: 3ffb1f30, prio:23, stack:6656, core=0
I (1503) system_api: Base MAC address is not set
I (1503) system_api: read default base MAC address from EFUSE
I (1523) wifi:wifi firmware version: 1865b55
I (1523) wifi:wifi certification version: v7.0
I (1523) wifi:config NVS flash: enabled
I (1523) wifi:config nano formating: disabled
I (1533) wifi:Init data frame dynamic rx buffer num: 32
I (1533) wifi:Init management frame dynamic rx buffer num: 32
I (1543) wifi:Init management short buffer num: 32
I (1543) wifi:Init static tx buffer num: 16
I (1553) wifi:Init tx cache buffer num: 32
I (1553) wifi:Init static rx buffer size: 1600
I (1553) wifi:Init static rx buffer num: 10
I (1563) wifi:Init dynamic rx buffer num: 32
I (1563) wifi_init: rx ba win: 6
I (1573) wifi_init: tcpip mbox: 32
I (1573) wifi_init: udp mbox: 6
I (1573) wifi_init: tcp mbox: 6
I (1583) wifi_init: tcp tx win: 5744
I (1583) wifi_init: tcp rx win: 5744
I (1593) wifi_init: tcp mss: 1440
I (1593) wifi_init: WiFi IRAM OP enabled
I (1603) wifi:Set ps type: 1

I (1603) main: WIFI SSID inpulse (2.4 GHz) PASS comprometimento
I (1683) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (1693) wifi:mode : sta (10:52:1c:87:20:c4)
I (1813) wifi:new:<2,0>, old:<1,0>, ap:<255,255>, sta:<2,0>, prof:1
I (1813) wifi:state: init -> auth (b0)
I (1813) wifi:state: auth -> assoc (0)
I (1823) wifi:state: assoc -> run (10)
I (1833) wifi:connected with inpulse (2.4 GHz), aid = 8, channel 2, BW20, bssid = 04:d4:c4:b8:02:a0
I (1833) wifi:security: WPA2-PSK, phy: bgn, rssi: -29
I (1833) wifi:pm start, type: 1

I (1893) wifi:AP's beacon interval = 102400 us, DTIM period = 3
I (2483) esp_netif_handlers: sta ip: 192.168.0.38, mask: 255.255.255.0, gw: 192.168.0.1
I (2483) main: got ip:192.168.0.38
I (2483) main: connected!!
I (3483) uart: queue free spaces: 2
I (3483) main: end of main
CORRUPT HEAP: multi_heap.c:439 detected at 0x3f80dc20

abort() was called at PC 0x4008eac3 on core 0
0x4008eac3: multi_heap_assert at /home/nth/monitor20/esp-idf/components/heap/multi_heap_platform.h:67

Backtrace:0x4008b401:0x3ffca0e0 0x4008b9c9:0x3ffca100 0x400900a1:0x3ffca120 0x4008eac3:0x3ffca190 0x4008ee59:0x3ffca1b0 0x40083842:0x3ffca1e0 0x40083866:0x3ffca210 0x400900a9:0x3ffca230 0x4008354d:0x3ffca250 0x4011e28d:0x3ffca270 0x40127e35:0x3ffca2a0 0x40091ae5:0x3ffca320
0x4008b401: panic_abort at /home/nth/monitor20/esp-idf/components/esp_system/panic.c:341

0x4008b9c9: esp_system_abort at /home/nth/monitor20/esp-idf/components/esp_system/system_api.c:106

0x400900a1: abort at /home/nth/monitor20/esp-idf/components/newlib/abort.c:46

0x4008eac3: multi_heap_assert at /home/nth/monitor20/esp-idf/components/heap/multi_heap_platform.h:67

0x4008ee59: multi_heap_malloc_impl at /home/nth/monitor20/esp-idf/components/heap/multi_heap.c:439

0x40083842: heap_caps_malloc at /home/nth/monitor20/esp-idf/components/heap/heap_caps.c:145

0x40083866: heap_caps_malloc_default at /home/nth/monitor20/esp-idf/components/heap/heap_caps.c:183

0x400900a9: malloc at /home/nth/monitor20/esp-idf/components/newlib/heap.c:32

0x4008354d: wifi_malloc at /home/nth/monitor20/esp-idf/components/esp_wifi/esp32/esp_adapter.c:73

0x4011e28d: esf_buf_alloc at ??:?

0x40127e35: wdevProcessRxSucDataAll at ??:?

0x40091ae5: ppTask at ??:?

ELF file SHA256: f8476f5bbebac9d3
```

## Other items if possible

- sdkconfig file: https://gist.github.com/nathanieloliveira/2f2b24185ba5c691fe1906def2e5957a/raw/0005b01b08bf9a0d8f7948cd353e864c51920663/sdkconfig
- elf file: 
[esp32uartwifibug.elf.zip](https://github.com/espressif/esp-idf/files/6380426/esp32uartwifibug.elf.zip)

- coredump: 
[coredump.txt](https://github.com/espressif/esp-idf/files/6380430/coredump.txt)

negativekelvin, 2021-04-26:
```
#define CAPNO_UART_PIN_TX      GPIO_NUM_17
#define CAPNO_UART_PIN_RX      GPIO_NUM_16
```

Those are psram pins

nathanieloliveira, 2021-04-28:
> ```
> #define CAPNO_UART_PIN_TX      GPIO_NUM_17
> #define CAPNO_UART_PIN_RX      GPIO_NUM_16
> ```
> 
> Those are psram pins

This was the problem. Thank you for your help.

closed 2021-04-28 by nathanieloliveira