espressif/esp-idf #6946: IDF_TARGET=esp32c3 'make monitor' failed

wonderxue, 2021-04-27:
## Environment

- Module or chip used:  [ESP32C3]
- IDF version : v4.4-dev-960-gcf457d412
- Build System:         [CMake]
- Compiler version: 8.4.0
- Operating System:     [Linux(wsl)]
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

when i use `make monitor` ,errors occurs. informations as follows
```
Scanning dependencies of target monitor                                                                                                                                 
Note: python /home/cxk/esp32/esp-idf/tools/idf_monitor.py will search for a serial port. To specify a port, set the ESPPORT environment variable.                       
Note: python /home/cxk/esp32/esp-idf/tools/idf_monitor.py will attempt to set baud rate automatically. To specify a baud rate, set the ESPBAUD environment variable.    
usage: idf_monitor - a serial output monitor for esp-idf [-h] [--port PORT]                                                                                                                                                      
.......
idf_monitor - a serial output monitor for esp-idf: error: argument elf_file: can't open '2;/home/cxk/esp32/esp-idf/examples/get-started/hello_world/build/hello-world.elf': [Errno 2] 
No such file or directory: '2;/home/cxk/esp32/esp-idf/examples/get-started/hello_world/build/hello-world.elf'                                             
CMake Error at run_serial_tool.cmake:50 (message):         
python /home/cxk/esp32/esp-idf/tools/idf_monitor.py failed                                                                                                                                                                                                                                                                                                                                                                                                                                                            make[3]: *** [CMakeFiles/monitor.dir/build.make:76: CMakeFiles/monitor] Error 1                                                                                         make[2]: *** [CMakeFiles/Makefile2:2194: CMakeFiles/monitor.dir/all] Error 2                                                                                            make[1]: *** [CMakeFiles/Makefile2:2201: CMakeFiles/monitor.dir/rule] Error 2                                                                                           make: *** [Makefile:313: monitor] Error 2     
```
And when i set `CONFIG_ESP32C3_REV_MIN` as `3`, it changes into `can't open ‘3;/home/cxk/esp32/esp-idf/examples/get-started/hello_world/build/hello-world.elf'`

dobairoland, 2021-04-27:
The `make` build system supports ESP32 only. For other chips please use `idf.py` or `cmake`.

The right way to achieve what you intend is to run:
```sh
idf.py set-target esp32c3
idf.py build flash
idf.py monitor
```
Please see
https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html for more information.

wonderxue, 2021-04-28:
> The `make` build system supports ESP32 only. For other chips please use `idf.py` or `cmake`.
> 
> The right way to achieve what you intend is to run:
> 
> ```shell
> idf.py set-target esp32c3
> idf.py build flash
> idf.py monitor
> ```
> 
> Please see
> https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/get-started/index.html for more information.

wonderxue, 2021-04-28:
sorry for closing this issue by misoperation. I run `cmake` to generate `Makefile`, then run `make flash`. It does work but when i run `make monitor`, errors occur @dobairoland. And obviously the directory is not correct

dobairoland, 2021-04-28:
Yes, I see what is the issue now. Please use the above steps until we fix this. I'm sorry for the inconvenience.

closed 2021-05-04 by espressif-bot