espressif/esp-idf #6948: RFCOMM / SPP-Acceptor Deadlock

wylerAgy, 2021-04-27:
## Environment

- Development Kit:     ESP32-DevKitC
- Kit version v4
- Module or chip used:  ESP32-WROOM-32D
- IDF version V4.2 (Tested even with v4.4-dev-960-gcf457d412, same problem)
- Build System:      
- Compiler version  8.4.0
- Operating System:    Windows
- Using an IDE?: Yes Visual Studio Code
- Power Supply:         USB

## Problem Description

After some  RFCOMM-Connections initiated by a Silicon Labs BT121-A Module the ESP32 SPP-Server stops working and the ESP32 have to be restarted. 

In the Log-Output we see that there is a difference between BT121 and other Modules in closing the RFCOMM-Connection. 

Example:

If closing a PC-Connection, we get the Log-msg:

```
W (794163) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Closed (res: 19)
```

Closing a BT121-Connection results in:
```
W (5310483) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
```

By the way, we've also contacted Silicon Labs cause of this Problem and here is they're Response:

*I would also like to point out that the problem you have reported  is in fact a potential security vulnerability of the ESP32 bluetooth  module. It seems to us that there is a specific disconnection sequence  which causes the the chip to hang and it might be a bug on it's side  which potentially can be used for a security violation. Some kind of  coincidence caused that the BT121 stack uses that exact sequence which  exposes the weakness of the ESP32.*

### Expected Behavior
we need unlimited RFCOMM Connections with all Modules, specially with the BT121 from Silicon Labs

### Actual Behavior
After max 20 RFCOMM Connections it's not possible to Connect  again and the ESP32 have to be restarted. 
### Steps to reproduce

1. Start the BT_SPP_ACCEPTOR Example on ESP32
2. use a BT121-A Module/Eval-Board with BGTool from Silicon Labs to start a RFCOMM-Connection to the ESP32
3. just close and reconnect the RFCOMM-Connection in the BGTool/BT121 
4. after appr.20 Connections it's not possible to start any further RFCOMM-connection. 
5. esp32 restart needed to reenable RFCOMM-Connections.

### Code to reproduce this issue
just use the BT_SPP_ACCEPTOR example-code, no changes, just build and flash...

## Debug Log for a working PC-Connection

```
W (794003) BT_APPL: new conn_srvc id:26, app_id:255
I (794003) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
W (794163) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Closed (res: 19)
I (794163) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
W (794323) BT_RFCOMM: rfc_find_lcid_mcb LCID reused LCID:0x40 current:0x0
W (794333) BT_RFCOMM: RFCOMM_DisconnectInd LCID:0x40
```
## Debug Logs with NON-working BT121-Connections
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7100 
load:0x40078000,len:13212
ho 0 tail 12 room 4      
load:0x40080400,len:4568
0x40080400: _init at ??:?

entry 0x400806f4
I (30) boot: ESP-IDF v4.2-dirty 2nd stage bootloader
I (30) boot: compile time 11:37:30
I (31) boot: chip revision: 1
I (34) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (41) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (50) boot.esp32: SPI Flash Size : 4MB
I (54) boot: Enabling RNG early entropy source...
I (60) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (71) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot: End of partition table
I (97) boot_comm: chip revision: 1, min. application chip revision: 0
I (104) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x17c10 ( 97296) map
I (150) esp_image: segment 1: paddr=0x00027c38 vaddr=0x3ffbdb60 size=0x0335c ( 13148) load
I (156) esp_image: segment 2: paddr=0x0002af9c vaddr=0x40080000 size=0x00404 (  1028) load
0x40080000: _WindowOverflow4 at C:/Users/agysi/esp/esp-idf/components/freertos/xtensa/xtensa_vectors.S:1730

I (158) esp_image: segment 3: paddr=0x0002b3a8 vaddr=0x40080404 size=0x04c70 ( 19568) load
I (175) esp_image: segment 4: paddr=0x00030020 vaddr=0x400d0020 size=0x64274 (410228) map
0x400d0020: _stext at ??:?

I (331) esp_image: segment 5: paddr=0x0009429c vaddr=0x40085074 size=0x1032c ( 66348) load
0x40085074: r_ld_fm_sket_isr at ??:?

I (372) boot: Loaded app from partition at offset 0x10000
I (372) boot: Disabling RNG early entropy source...
I (373) cpu_start: Pro cpu up.
I (376) cpu_start: Application information:
I (381) cpu_start: Project name:     bt_spp_acceptor_demo
I (387) cpu_start: App version:      1
I (392) cpu_start: Compile time:     Apr 27 2021 11:37:20
I (398) cpu_start: ELF file SHA256:  e9e94c16522ed763...
I (404) cpu_start: ESP-IDF:          v4.2-dirty
I (409) cpu_start: Starting app cpu, entry point is 0x40081878
0x40081878: call_start_cpu1 at C:/Users/agysi/esp/esp-idf/components/esp32/cpu_start.c:287

I (0) cpu_start: App cpu up.
I (419) heap_init: Initializing. RAM available for dynamic allocation:
I (426) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (432) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (438) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (444) heap_init: At 3FFCA058 len 00015FA8 (87 KiB): DRAM
I (451) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (457) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (463) heap_init: At 400953A0 len 0000AC60 (43 KiB): IRAM
I (470) cpu_start: Pro cpu start user code
I (488) spi_flash: detected chip: generic
I (489) spi_flash: flash io: dio
I (489) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (519) BTDM_INIT: BT controller compile version [3723d5b]
I (519) system_api: Base MAC address is not set
I (519) system_api: read default base MAC address from EFUSE
I (629) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (1179) SPP_ACCEPTOR_DEMO: ESP_SPP_INIT_EVT
I (1199) SPP_ACCEPTOR_DEMO: event: 10
I (1199) SPP_ACCEPTOR_DEMO: ESP_SPP_START_EVT
W (44679) BT_APPL: new conn_srvc id:26, app_id:255
I (44689) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
W (51099) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
I (51099) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
W (54449) BT_APPL: new conn_srvc id:26, app_id:255
I (54449) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
W (56369) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
I (56369) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
W (60319) BT_APPL: new conn_srvc id:26, app_id:255
I (60319) SPP_ACCEPTOR_DEMO: ESP_SPP_SRV_OPEN_EVT
W (62119) BT_RFCOMM: port_rfc_closed RFCOMM connection in state 2 closed: Peer connection failed (res: 16)
I (62119) SPP_ACCEPTOR_DEMO: ESP_SPP_CLOSE_EVT
E (74019) BT_HCI: command_timed_out hci layer timeout waiting for response to a command. opcode: 0x40f
```

In this example the ESP32 died after only 3 Connections, after getting the Error(74019) a Module-Restart is needed.

Wth-Esp, 2021-05-06:
Hi @wylerAgy 

Could you please update the v4.2.1 release version? I have a fix commit for connection issue on v4.2.1 but not on v4.2.
And other valuable fixes are also on  v4.2.1 , so I think it's good to update IDF to v4.2.1.

Thanks

wylerAgy, 2021-05-06:
Hi @Wth-Esp

Wth-Esp, 2021-05-06:
Hi @wylerAgy 

I have a fix for Classic BT's connection issue on release `v4.2.1`, could you please upgrade the IDF version and try again?

Thanks

wylerAgy, 2021-05-06:
I've created the previous log with 4.1.2 ?

I'v re-Configured the ESP-IDF Extension for VisualStudioCode and especially selected v4.2.1 from dropdown...
[image]

Wth-Esp, 2021-05-06:
@wylerAgy Let me find the BT121 dev kit and reproduce, please wait my update.

Wth-Esp, 2021-05-07:
@wylerAgy  I cannot connect BT121-A via BlueGiga, and the log is here, im I right with the dev kit configuration?
[image]

Wth-Esp, 2021-05-07:
I cannot connect to BT121 with my android phone. And my phone says that BT121 reject encription.

wylerAgy, 2021-05-10:
I'm also working with a dev-kit here.
[https://www.silabs.com/development-tools/wireless/bluetooth/dkbt-bluetooth-smart-development-kit](https://www.silabs.com/development-tools/wireless/bluetooth/dkbt-bluetooth-smart-development-kit)

 Assume you missed to enable Bondable mode on BT121? 
Switch to the Security manager Tab and Set Bondable mode to on.
[image]

Wth-Esp, 2021-05-10:
And I find that I cannot kick the reconnection button in BGTool.
[image]

And also cannot find any valuable information from the air log.
[image]

This may need more debugs, also more time. I'm not very sure whether the issue is on the ESP32 side.

Thanks

Wth-Esp, 2021-05-10:
SDP is Service Discovery Profile, the main function of SDP is to discover profiles on the peer device and handle the discover request from the peer device.

`W (74559) BT_SDP: SDP - Rcvd L2CAP disc, unknown CID: 0x46`

Maybe delay 2s after disconnect and reconnect to ESP32 can help a lot.

wylerAgy, 2021-05-10:
This is confusing? Why none of the Devices i have here is able to reconnect but on your side it works?

I'v tested with a CAT S31 and a ASUS BT-Dongle and the BT121. I always have to restart the ESP32 to get the SPP back working.

Any suggestions?

wylerAgy, 2021-05-12:
I've switched the Debug Log Level from HCI-Layer to DEBUG in hope to see more and now i've detected this:

After the failure occurs, it seems as it's only possible to connect, but RFCOMM is not started/involved. In the ESP-log i only see HCI-Logs but no more RFCOMM-logs? 

This is a Log from a working RFCOMM Connect/Close Sequence, you see HCI and RFCOMM logs:

```
D (519875) BT_HCI: Receive packet event_code=0x4

D (519875) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (519875) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (519885) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (519895) BT_HCI: HCI Enqueue Comamnd opcode=0x409

D (519895) BT_HCI: Receive packet event_code=0xf

D (519905) BT_HCI: Receive packet event_code=0x3

D (519905) BT_HCI: HCI Enqueue Comamnd opcode=0x41f

D (519915) BT_HCI: reassemble_and_dispatch

D (519915) BT_HCI: Receive packet event_code=0xf

D (519915) BT_HCI: Receive packet event_code=0x1c

D (519925) BT_HCI: Receive packet event_code=0x1b

D (519925) BT_HCI: HCI Enqueue Comamnd opcode=0x41d

D (519935) BT_HCI: Receive packet event_code=0xf

D (519935) BT_HCI: Receive packet event_code=0xc

D (519945) BT_HCI: HCI Enqueue Comamnd opcode=0x40f

D (519955) BT_HCI: Receive packet event_code=0xf

D (519955) BT_HCI: Receive packet event_code=0x1d

D (519955) BT_HCI: HCI Enqueue Comamnd opcode=0x80d

D (519965) BT_HCI: Receive packet event_code=0xe

D (519965) BT_HCI: btu_hcif_command_complete_evt

D (519975) BT_HCI: HCI Enqueue Comamnd opcode=0x80d

D (519975) BT_HCI: Receive packet event_code=0xe

D (519985) BT_HCI: btu_hcif_command_complete_evt

D (519985) BT_HCI: Receive packet event_code=0x13

D (519995) BT_HCI: HCI Enqueue Comamnd opcode=0xc35

D (519995) BT_HCI: reassemble_and_dispatch

D (520005) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (520015) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (520015) BT_HCI: btu_free_quick_timer Unable to find expected alarm in hashmap
D (520025) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (520035) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap
D (520055) BT_HCI: Receive packet event_code=0x13

D (520055) BT_HCI: Receive packet event_code=0x13

D (520055) BT_HCI: HCI Enqueue Comamnd opcode=0xc35

D (520055) BT_HCI: reassemble_and_dispatch
[... 384 lines cut ...]

D (890785) BT_HCI: Receive packet event_code=0xf

D (890805) BT_HCI: Receive packet event_code=0x23

D (890805) BT_HCI: HCI Enqueue Comamnd opcode=0x40f

D (890805) BT_HCI: Receive packet event_code=0xf

D (890805) BT_HCI: Receive packet event_code=0x1d

D (890815) BT_HCI: HCI Enqueue Comamnd opcode=0x80d

D (890815) BT_HCI: Receive packet event_code=0xe

D (890825) BT_HCI: btu_hcif_command_complete_evt

D (917745) BT_HCI: Receive packet event_code=0x5

D (917745) BT_HCI: btu_free_timer Unable to find expected alarm in hashmap

```

@Wth-Esp this seems as this matches your observation as you said you're still able to Connect with your Phone, but in fact there is no working RFCOMM-Connection in this Situation! 
So please try to reproduce this with your Setup but try to establish a working RFCOMM-connection instead of only connecting. I assume you get the same result as me? Please verify this.

xiongweichao, 2021-06-11:
Hi, @wylerAgy 

Can you try the patch attached below and check if it solves your issue?  Please let us know if the change works for you.
[fix_spp_acceptor_deadlock_master.zip](https://github.com/espressif/esp-idf/files/6636263/fix_spp_acceptor_deadlock_master.zip)

Thanks

wylerAgy, 2021-06-15:
Hi @xiongweichao
I've applied your patch to my VS Code IDF master setup and now it seems to work. I'm able to make multiple(tested 100+) RFCOMM-Connections between the BT121 and the ESP32 without any Problems and the ESP32 keeps on working...

closed 2021-06-28 by espressif-bot