espressif/esp-idf #6950: esp_netif_deinit 不能释放“tiT”任务

YangJiawei, 2021-04-27:
Development Kit: [ESP32-Wrover-Kit]
Kit version (for WroverKit/PicoKit/DevKitC): [v4.1]
Module or chip used: [ESP32-WROVER-B]
IDF version : v4.2 - c40f259
Build System: [CMake|idf.py]
Compiler version :xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: [ubuntu]
Using an IDE?: [VSCode + Espressif IDFv1.0.2 ]
Power Supply: [USB]

## Problem Description

使用`esp_netif_init();`时会创建名为`tiT`的进程。使用`esp_netif_deinit();`并不能把进程删除。

因为要从AP模式转为sta模式，把之前的wifi连接都注销了。
```
    esp_event_loop_delete_default();
    esp_wifi_disconnect();
    esp_wifi_stop();
    esp_wifi_deinit();
    esp_netif_deinit();
```
使用上面的语句删除了`sys_evt`,`wifi`进程，
但是发现堆栈里面还存在`tiT`进程。导致切换成AP一直连接，但是不能连接成功。

请问怎样把之前的sta模式的所有进程全部删除!?

ESP-YJM, 2021-04-28:
@YangJiawei `tiT` 是 tcpip task, 无论是 AP 模式还是 STA 模式都需要  tcpip task,所以可以不需要关闭。请问不关闭这个 task 会有问题吗？

YangJiawei, 2021-04-29:
@ESP-YJM 只是习惯性的认为创建的进程在结束时就应该都注销掉，改了一下代码注销掉了进程，再从sta切换为ap发现还是不能连接，应该不是这个进程的问题。
`sta`模式切换到`ap`，先清理了之前的连接，之后再去进行ap初始化，可以出现热点，但是连接不上，获取不到ip。现在不知道问题出在哪里……
```
void wifi_clean(void)
{
    esp_event_loop_delete_default();
    esp_wifi_disconnect();
    esp_wifi_stop();
    esp_wifi_deinit();
    //esp_netif_deinit();
    mj_esp_netif_deinit();
}
```
```
void wifi_init_ap(void)
{
    ESP_ERROR_CHECK(esp_netif_init());                  
    ESP_ERROR_CHECK(esp_event_loop_create_default());  
    esp_netif_create_default_wifi_ap();                 
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT(); 
    ESP_ERROR_CHECK(esp_wifi_init(&cfg));              
    wifi_config_t wifi_config = {
        .ap = {
            .ssid = WIFI_AP_NAME,
            .ssid_len = strlen(WIFI_AP_NAME),
            .max_connection = 4,
            .authmode = WIFI_AUTH_OPEN 
        }};
    ESP_ERROR_CHECK(esp_wifi_set_mode(WIFI_MODE_AP));                 
    ESP_ERROR_CHECK(esp_wifi_set_config(ESP_IF_WIFI_AP, &wifi_config)); 
    ESP_ERROR_CHECK(esp_wifi_start());                               
    esp_wifi_connect();
}
```
```
I (13176) wifi:flush txq
I (13176) wifi:stop sw txq
I (13186) wifi:lmac stop hw txq
I (13186) wifi:Deinit lldesc rx mblock:10
I (13206) wifi:wifi driver task: 3ffbe044, prio:23, stack:6656, core=0
I (13226) wifi:wifi firmware version: 1865b55
I (13226) wifi:wifi certification version: v7.0
I (13226) wifi:config NVS flash: enabled
I (13226) wifi:config nano formating: disabled
I (13226) wifi:Init data frame dynamic rx buffer num: 32
I (13236) wifi:Init management frame dynamic rx buffer num: 32
I (13236) wifi:Init management short buffer num: 32
I (13246) wifi:Init dynamic tx buffer num: 32
I (13246) wifi:Init static rx buffer size: 1600
I (13256) wifi:Init static rx buffer num: 10
I (13256) wifi:Init dynamic rx buffer num: 32
I (13256) wifi_init: rx ba win: 6
I (13266) wifi_init: tcpip mbox: 32
I (13266) wifi_init: udp mbox: 6
I (13276) wifi_init: tcp mbox: 6
I (13276) wifi_init: tcp tx win: 5744
I (13276) wifi_init: tcp rx win: 5744
I (13286) wifi_init: tcp mss: 1440
I (13286) wifi_init: WiFi IRAM OP enabled
I (13296) wifi_init: WiFi RX IRAM OP enabled
I (13316) wifi:mode : softAP (a8:03:2a:eb:fc:29)
I (13316) wifi:Total power save buffer number: 16
I (13316) wifi:Init max length of beacon: 752/752
I (13316) wifi:Init max length of beacon: 752/752
I (58646) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (58656) wifi:station: 20:f4:78:0b:45:1f join, AID=1, bgn, 20
I (58766) wifi:new:<1,1>, old:<1,0>, ap:<1,1>, sta:<255,255>, prof:1
I (58766) wifi:station: 7c:94:2a:b3:78:81 join, AID=2, bgn, 40U
I (58786) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (58786) wifi:station: 7c:94:2a:b3:78:81 join, AID=2, bgn, 40U
I (94876) wifi:station: 20:f4:78:0b:45:1f leave, AID = 1, bss_flags is 134259, bss:0x3ffb8e20
I (94876) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (96396) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (96396) wifi:station: 20:f4:78:0b:45:1f join, AID=1, bgn, 20
I (132486) wifi:station: 20:f4:78:0b:45:1f leave, AID = 1, bss_flags is 134243, bss:0x3ffb8e20
I (132486) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (133876) wifi:new:<1,1>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (133876) wifi:station: 20:f4:78:0b:45:1f join, AID=1, bgn, 20
```

YangJiawei, 2021-04-29:
是不是MAC地址的问题？

ESP-YJM, 2021-04-29:
@YangJiawei 确认下sta切换成 ap时，之前注销的`tiT`进程有没有起来？

YangJiawei, 2021-04-29:
@ESP-YJM 有的，注销和不注销都尝试过，进程`tiT`都没问题。应该不是这个问题。
但是测试的时候，单独`sta`工作没问题，单独`ap`也没问题，但是从`sta`切换到`ap`手机就连接不到热点了。

程序设这样写的，如果`sta`连接失败则切换到`ap`模式。日志如下：
```
I (0) cpu_start: App cpu up.
I (510) heap_init: Initializing. RAM available for dynamic allocation:
I (516) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (522) heap_init: At 3FFB7EA8 len 00028158 (160 KiB): DRAM
I (529) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (535) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (541) heap_init: At 400955BC len 0000AA44 (42 KiB): IRAM
I (548) cpu_start: Pro cpu start user code
I (566) spi_flash: detected chip: generic
I (567) spi_flash: flash io: dio
I (567) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
mj_nvs_get_str:abcdefgh
mj_nvs_get_str:12345678
I (676) wifi:wifi driver task: 3ffc077c, prio:23, stack:6656, core=0
I (676) system_api: Base MAC address is not set
I (676) system_api: read default base MAC address from EFUSE
I (696) wifi:wifi firmware version: 1865b55
I (696) wifi:wifi certification version: v7.0
I (696) wifi:config NVS flash: enabled
I (696) wifi:config nano formating: disabled
I (706) wifi:Init data frame dynamic rx buffer num: 32
I (706) wifi:Init management frame dynamic rx buffer num: 32
I (716) wifi:Init management short buffer num: 32
I (716) wifi:Init dynamic tx buffer num: 32
I (726) wifi:Init static rx buffer size: 1600
I (726) wifi:Init static rx buffer num: 10
I (726) wifi:Init dynamic rx buffer num: 32
I (736) wifi_init: rx ba win: 6
I (736) wifi_init: tcpip mbox: 32
I (746) wifi_init: udp mbox: 6
I (746) wifi_init: tcp mbox: 6
I (746) wifi_init: tcp tx win: 5744
I (756) wifi_init: tcp rx win: 5744
I (756) wifi_init: tcp mss: 1440
I (766) wifi_init: WiFi IRAM OP enabled
I (766) wifi_init: WiFi RX IRAM OP enabled
I (866) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (866) wifi:mode : sta (a8:03:2a:eb:fc:28)
I (866) WIFI_EVENT事件 : 连接wifi
I (2916) WIFI_EVENT事件：: 重试连接AP
I (2926) WIFI_EVENT事件：: 连接AP失败
I (4966) WIFI_EVENT事件：: 重试连接AP
I (4966) WIFI_EVENT事件：: 连接AP失败
I (7016) WIFI_EVENT事件：: 重试连接AP
I (7016) WIFI_EVENT事件：: 连接AP失败
I (9066) WIFI_EVENT事件：: 重试连接AP
I (9066) WIFI_EVENT事件：: 连接AP失败
I (11116) WIFI_EVENT事件：: 重试连接AP
I (11116) WIFI_EVENT事件：: 连接AP失败
I (13166) WIFI_EVENT事件：: 连接AP失败
I (13166) 标志bits状态：: 连接失败
I (13166) wifi:flush txq
I (13166) wifi:stop sw txq
I (13176) wifi:lmac stop hw txq
I (13176) wifi:Deinit lldesc rx mblock:10
I (13196) wifi:wifi driver task: 3ffbe044, prio:23, stack:6656, core=0
I (13216) wifi:wifi firmware version: 1865b55
I (13216) wifi:wifi certification version: v7.0
I (13216) wifi:config NVS flash: enabled
I (13216) wifi:config nano formating: disabled
I (13216) wifi:Init data frame dynamic rx buffer num: 32
I (13226) wifi:Init management frame dynamic rx buffer num: 32
I (13226) wifi:Init management short buffer num: 32
I (13236) wifi:Init dynamic tx buffer num: 32
[... 49 lines cut ...]
I (706) wifi:Init data frame dynamic rx buffer num: 32
I (706) wifi:Init management frame dynamic rx buffer num: 32
I (716) wifi:Init management short buffer num: 32
I (716) wifi:Init dynamic tx buffer num: 32
I (726) wifi:Init static rx buffer size: 1600
I (726) wifi:Init static rx buffer num: 10
I (726) wifi:Init dynamic rx buffer num: 32
I (736) wifi_init: rx ba win: 6
I (736) wifi_init: tcpip mbox: 32
I (746) wifi_init: udp mbox: 6
I (746) wifi_init: tcp mbox: 6
I (746) wifi_init: tcp tx win: 5744
I (756) wifi_init: tcp rx win: 5744
I (756) wifi_init: tcp mss: 1440
I (766) wifi_init: WiFi IRAM OP enabled
I (766) wifi_init: WiFi RX IRAM OP enabled
I (876) phy: phy_version: 4500, 0cd6843, Sep 17 2020, 15:37:07, 0, 0
I (876) wifi:mode : softAP (a8:03:2a:eb:fc:29)
I (876) wifi:Total power save buffer number: 16
I (886) wifi:Init max length of beacon: 752/752
I (886) wifi:Init max length of beacon: 752/752
I (13166) wifi:new:<1,0>, old:<1,1>, ap:<1,1>, sta:<255,255>, prof:1
I (13176) wifi:station: 20:f4:78:0b:45:1f join, AID=1, bgn, 20
I (13376) esp_netif_lwip: DHCP server assigned IP to a station, IP is: 192.168.4.2
```

ESP-YJM, 2021-04-29:
@YangJiawei 你应该少调用了 `esp_wifi_clear_default_wifi_driver_and_handlers(wifi_netif)`,具体函数使用已经 wifi模式切换你可以参考 https://github.com/espressif/esp-idf/blob/master/examples/common_components/protocol_examples_common/connect.c#L329。

YangJiawei, 2021-04-29:
应该是'esp_netif_init'存在问题，改用`tcpip_adapter_init`

ESP-YJM, 2021-04-29:
> esp_netif_init

这个函数不存在问题，只是使用上的问题。`tcpip_adapter_init` 不是推荐的函数。

YangJiawei, 2021-04-30:
@ESP-YJM 你可以简单尝试以下，比如让AP+STA共同模式工作，如果是`esp_netif_init`进行初始化，那么热点就不能连接，如果更改为`tcpip_adapter_init`就可以的。

ESP-YJM, 2021-05-06:
`tcpip_adapter_init` 的话是以前版本的函数，只针对于 wifi 的适配函数，`esp_netif_init` 是 当前版本的函数，针对于 wifi，以太网，SLIP，PPP等网络接口写的组件，涉及功能比较多，建议按照我们示例里使用。你说的`tcpip_adapter_init就可以的` 应该只是功能都封装好了，而使用 `esp_netif_init` 满足你的需求还需要再调用其他函数。另外sta切换ap或者ap切换sta，可以只涉及 wifi层的函数调用，像 netif_init，event这些函数可以不需要注销再初始化。

YangJiawei, 2021-05-13:
@ESP-YJM 使用的是AP和STD共存的模式，暂时先应用`tcpip_adapter_init`吧，有时间再去慢慢查。另外，咱俩名字是差一个字嘛！！！！

Alvin1Zhang, 2021-06-01:
Thanks for reporting, the fix is available at https://github.com/espressif/esp-idf/commit/c798906f4ae578b6ca16d8867d2b508224ba117f, feel free to reopen.

closed 2021-06-01 by Alvin1Zhang