espressif/esp-idf #6953: False-positive warning about app partitions are too small

AxelLin, 2021-04-28:
## Environment

Module or chip used: ESP32-WROOM-32E
IDF version: v4.4-dev-1183-g9d34a1cd42f6
Build System: idf.py
Compiler version: xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
Operating System: Linux
Power Supply: USB

## Problem Description

After pull the latest esp-idf code, I hit below issue:
On my testing device, I have a dedicated partition for testing firmware.
(I have set BOOTLOADER_APP_TEST)
The testing firmware is much smaller than the normal application, so it has a small partition.

The problem is when I build my application, I always got warning:
app partitions are too small for binary my-app.bin

I'm wondering if the idf.py build can ignore the partition size checking against SubType: "test" partition.
(It's good to have such partition size checking, but if it shows false-positive warnings, I will just ignore it all the time.
Then it's possible to ignore real warning.)

dobairoland, 2021-05-06:
Hi @AxelLin. I'm sorry but it is not clear to me why do you think this is a false positive.

> The testing firmware is much smaller than the normal application, so it has a small partition.

> app partitions are too small for binary my-app.bin

The application is bigger (not smaller) than the partition.

Can you please provide more information for reproducing this?

AxelLin, 2021-05-06:
Hi @dobairoland ,
As I said, I have a small partition for test app (with SubType "test").
I use esptool.py to download a testing firmware to this partition. (the testing firmware is independently build).
The warning is because my normal app is much bigger, but the tool seems checks all partitions and found the image size is too big and cannot fit into the test app partition. 
The "test" partition is not for downloading "normal" application. So I think the tool should  ignore testing SubType "test" partition.

 ./tools/unit-test-app/partition_table_unit_test_two_ota.csv as example shows "test" SubType.

closed 2021-05-06 by AxelLin

dobairoland, 2021-05-06:
@AxelLin Ok, thank you for the explanation. 

I think the current behavior is correct. The application of the current project has to fit into the "test" partition as well. It is used as a fallback partition. 

Please use a custom partition type for this use case (https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/partition-tables.html#type-field). You will not get a warning in that case.

> ./tools/unit-test-app/partition_table_unit_test_two_ota.csv as example shows "test" SubType.

You can notice there that factory and the OTA partitions have the same type.