espressif/esp-idf #6955: ESP32 ECO3 Highlevel interrupt irq wdt timeout

sicanins, 2021-04-28:
## Environment

- Module or chip used:  ESP32-WROVER-E
- IDF version: v4.3-beta2
- Using an IDE?: Yes

## Problem Description

I am using a highlevel 5 interrupt. It serves the UART to fix the misalignment between break irq and fifo.
That works so far, but after some minutes, the system runs into a interrupt wdt timeout.
Meanwhile we tested a ton of things, but everything seems to be right. 
Really need some advice here.

We made sure its NOT stuck in our code by setting a pin hi/lo at the beginning/end of the interrupt.
Also, after setting the wdt timeout to 10s, its possible to connect the debugger. Everything looks even normal, code is executed on both cores, the lower IRQ handlers seem to be called on the CPU0 even.

I dont know how to check the irq enable flags of the CPU, didn't find the doc for this.

Is this related to the errata 3.15 maybe? Or would this behave differently?

### Actual Behavior
```
Guru Meditation Error: Core  0 panic'ed (Interrupt wdt timeout on CPU0).

Core  0 register dump:
PC      : 0x40188a46  PS      : 0x00060334  A0      : 0x800d9a23  A1      : 0x3ffbd340
0x40188a46: esp_pm_impl_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_pm/pm_impl.c:802

A2      : 0x00000000  A3      : 0x00060320  A4      : 0x800922d5  A5      : 0x3ffbe540
A6      : 0x00000000  A7      : 0x3ff4e0c4  A8      : 0x8014f56e  A9      : 0x3ffbd330
A10     : 0x00000003  A11     : 0x00060323  A12     : 0x00060320  A13     : 0x3ffbe540
A14     : 0x00000000  A15     : 0x3ff4e0c4  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40188a43:0x3ffbd340 0x400d9a20:0x3ffbd360 0x40097eb1:0x3ffbd380 0x4009a201:0x3ffbd3a0
0x40188a43: cpu_ll_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_pm/pm_impl.c:800

0x400d9a20: esp_vApplicationIdleHook at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40097eb1: prvIdleTask at m:\mcu_workspace\controllerm4\build/../esp-idf/components/freertos/tasks.c:3835 (discriminator 1)

0x4009a201: vPortTaskWrapper at m:\mcu_workspace\controllerm4\build/../esp-idf/components/freertos/port/xtensa/port.c:171

Core  1 register dump:
PC      : 0x40188a46  PS      : 0x00060c34  A0      : 0x800d9a23  A1      : 0x3ffcc860
0x40188a46: esp_pm_impl_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_pm/pm_impl.c:802

A2      : 0x3ffc7bfd  A3      : 0x00060a20  A4      : 0x00000014  A5      : 0x000000a5
A6      : 0x000000a5  A7      : 0x00060023  A8      : 0x8014f56e  A9      : 0x3ffcc850
A10     : 0x00000003  A11     : 0x00060e23  A12     : 0x00060e20  A13     : 0x000000a5
A14     : 0x000000a5  A15     : 0x00060b23  SAR     : 0x0000001d  EXCCAUSE: 0x00000005
EXCVADDR: 0x00000000  LBEG    : 0x00000000  LEND    : 0x00000000  LCOUNT  : 0x00000000

Backtrace:0x40188a43:0x3ffcc860 0x400d9a20:0x3ffcc880 0x40097eb1:0x3ffcc8a0 0x4009a201:0x3ffcc8c0
0x40188a43: cpu_ll_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/hal/esp32/include/hal/cpu_ll.h:183
 (inlined by) esp_pm_impl_waiti at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_pm/pm_impl.c:800

0x400d9a20: esp_vApplicationIdleHook at m:\mcu_workspace\controllerm4\build/../esp-idf/components/esp_common/src/freertos_hooks.c:63

0x40097eb1: prvIdleTask at m:\mcu_workspace\controllerm4\build/../esp-idf/components/freertos/tasks.c:3835 (discriminator 1)

0x4009a201: vPortTaskWrapper at m:\mcu_workspace\controllerm4\build/../esp-idf/components/freertos/port/xtensa/port.c:171
```

### Steps to reproduce

The IRQ handler looks like this (cut the midddle part as it does not contribute it seems).
We are accessing the UART1 and for sure no other code does it at the same time. 
Also we are accessing an array and some variables in IRAM.
[... 109 lines cut ...]
	l32i temp4,                           a0, temp4_offset
	l32i temp5,                           a0, temp5_offset
	l32i temp6,                           a0, temp6_offset
	l32i temp7,                           a0, temp7_offset
	l32i temp8,                           a0, temp8_offset
	l32i temp9,                           a0, temp9_offset
	l32i temp10,                          a0, temp10_offset
	l32i temp11,                          a0, temp11_offset
	l32i temp12,                          a0, temp12_offset
	l32i temp13,                          a0, temp13_offset
	l32i temp14,                          a0, temp14_offset
	l32i temp15,                          a0, temp15_offset
	l32i temp16,                          a0, temp16_offset

	rsync

    rsr     a0, EXCSAVE_5 				// restore a0: read special register EXCSAVE (location for high level int. handler) to a0
    rfi     5							// finaly return from high priority interrupt level 5
```

The handler is called every few ms. 
After some minutes, we get the panic above. We were not able to narrow this down to a specific instruction in our handler. 

Thanks,
Simon

espressif-wzh, 2021-04-30:
Hi, sicanins, can you provide a project that contains the necessary code to reproduce the problem, which will make it easier for us to reproduce the problem you encountered.

sicanins, 2021-05-01:
Hi,
we are doing this, it will take some time to isolate it into a new project.

For now we found out, that it seems to work if the duration of the level 5 IRQ is shorter. 
Reading only 4 fifo bytes instead of 100 drops the duration from 60us to 4us.
No more wdt timeout within 2 days! 8 bytes is already too much.

Trying to understand the delay in dport_panic_highint_hdl.S. Could this be related?

Thanks,
Simon

closed 2023-03-08 by Alvin1Zhang