espressif/esp-idf #6958: esp32-c3 use idf has been CMake Error in idf-master、idf-v4.3

liantianji, 2021-04-29:
I try to  set-target   "idf.py set-target esp32" in idf-master、idf-v4.3. hass been CMake Error.

## Environment

- Development Kit:      [Ai-Thinker-C3F]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1]
- Module or chip used:  [ESP32-c3]
- IDF version:   [v4.4-dev/v4.3]
- Build System:         [|idf.py]
- Compiler version:    xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     [Windows]
- (Windows only) environment type: [ESP Command
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description
C:\Users\daniel\esp\hello_world>idf.py set-target esp32c3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'c:\users\daniel\esp\hello_world\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32c3, new sdkconfig created. Existing sdkconfig renamed to sdkconfig.old.
Running cmake in directory c:\users\daniel\esp\hello_world\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DIDF_TARGET=esp32c3 -DCCACHE_ENABLE=1 c:\users\daniel\esp\hello_world"...
-- Found Git: C:/Users/daniel/.espressif/tools/idf-git/2.30.1/cmd/git.exe (found version "2.30.1.windows.1")
fatal: not a git repository (or any of the parent directories): .git
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 8.4.0
-- The CXX compiler identification is GNU 8.4.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Check for working C compiler: C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
-- Check for working C compiler: C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe
-- Check for working CXX compiler: C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-g++.exe -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32c3
-- Project sdkconfig file C:/Users/daniel/esp/hello_world/sdkconfig
CMake Warning at D:/esp-idf/tools/cmake/crosstool_version_check.cmake:28 (message):
  Xtensa toolchain
  C:/Users/daniel/.espressif/tools/riscv32-esp-elf/1.24.0.123_64eb9ff-8.4.0/riscv32-esp-elf/bin/riscv32-esp-elf-gcc.exe
  crosstool-ng version 1.24.0.123_64eb9ff doesn't match supported version
  esp-2020r3.  Check Getting Started documentation or proceed at own risk.
Call Stack (most recent call first):
  D:/esp-idf/components/esp_common/project_include.cmake:6 (crosstool_version_check)
  D:/esp-idf/tools/cmake/build.cmake:318 (include)
  D:/esp-idf/tools/cmake/build.cmake:533 (__build_process_project_includes)
  D:/esp-idf/tools/cmake/project.cmake:361 (idf_build_process)
  CMakeLists.txt:6 (project)

CMake Error at D:/esp-idf/tools/cmake/component.cmake:305 (message):
  Include directory 'D:/esp-idf/components/mbedtls/mbedtls/include' is not a
  directory.
Call Stack (most recent call first):
  D:/esp-idf/tools/cmake/component.cmake:477 (__component_add_include_dirs)
  D:/esp-idf/components/mbedtls/CMakeLists.txt:4 (idf_component_register)

//Detailed problem description goes here.

suda-morris, 2021-04-29:
Does `git submodule update --init --recursive` help?

liantianji, 2021-04-29:
> Could please go to the mbedtls submodule and check if it exist?
> Does `git submodule update --init --recursive` help?

how to? 要到idf主目录执行？

liantianji, 2021-04-29:
> Could please go to the mbedtls submodule and check if it exist?
> Does `git submodule update --init --recursive` help?

I have git clone again.Thank you.
谢谢！

closed 2021-04-29 by liantianji