espressif/esp-idf #6959: migration from esp32 to esp32-s2 (multisocket transfer)

andrej102, 2021-04-29:
----------------------------- Delete below -----------------------------

**Reminder: If your issue is a general question, starts similar to "How do I..", or is related to 3rd party development kits/libs, please discuss this on our community forum at https://esp32.com instead.**

INSTRUCTIONS
============

Before submitting a new issue, please follow the checklist and try to find the answer.

- [ ] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [ ] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [ ] I have searched the issue tracker for a similar issue and not found a similar issue.

If the issue cannot be solved after the steps before, please follow these instructions so we can get the needed information to help you in a quick and effective fashion.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

- Development Kit:      [ESP32-Wrover-Kit|ESP32-DevKitC|ESP32-PICO-Kit|ESP32-LyraT|ESP32-LyraTD-MSC|none]
- Kit version (for WroverKit/PicoKit/DevKitC): [v1|v2|v3|v4]
- Module or chip used:  [ESP32-WROOM-32|ESP32-WROOM-32D|ESP32-WROOM-32U|ESP32-WROVER|ESP32-WROVER-I|ESP32-WROVER-B|ESP32-WROVER-IB|ESP32-SOLO-1|ESP32-PICO-D4|ESP32]
- IDF version (run ``git describe --tags`` to find it):
    // v3.2-dev-1148-g96cd3b75c
- Build System:         [Make|CMake|idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // 1.22.0-80-g6c4433a
- Operating System:     [Windows|Linux|macOS]
- (Windows only) environment type: [MSYS2 mingw32|ESP Command Prompt|Plain Command Prompt|PowerShell].
- Using an IDE?: [No|Yes (please give details)]
- Power Supply:         [USB|external 5V|external 3.3V|Battery]

## Problem Description

//Detailed problem description goes here.

### Expected Behavior

### Actual Behavior

### Steps to reproduce

1. step1
2. ...

// If possible, attach a picture of your setup/wiring here.

### Code to reproduce this issue

```cpp
// the code should be wrapped in the ```cpp tag so that it will be displayed better.
#include "esp_log.h"

void app_main()
{

}

```
// If your code is longer than 30 lines, [GIST](https://gist.github.com) is preferred.

## Debug Logs

```
Debug log goes here, should contain the backtrace, as well as the reset source if it is a crash.
Please copy the plain text here for us to search the error log. Or attach the complete logs but leave the main part here if the log is *too* long.
```

## Other items if possible

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

andrej102, 2021-04-29:
I moved a working project from ESP 32 to ESP32-S2. 
Custom board
Windows 10
Stable environment ESP-IDF-v4.2.

The project uses data transfer through 4 sockets at the same time (it determines which socket is free and data is placed there). Everything works fine on ESP 32, but it doesn't work on ECP32-S2. After several blocks of data (1408 bytes each) have been sent, sendsocket returns -1 (socket write error). 
When using a single socket, data is transmitted normally.

Please help me understand how to fix this? 

Code of the data sending function:

[code]
uint8_t PutToSocket(uint8_t * data, uint16_t len_data, uint16_t *len_data_sent)
{
    int len = 0;
    fd_set fd_write_set;
    struct timeval tv;
    int retval;
    uint8_t return_result = SOCKET_WRITE_IN_PROGRESS;
    uint8_t retry_counter = 0;
    uint8_t i = 0;

    retry_counter = 0;
    return_result = SOCKET_WRITE_IN_PROGRESS;

    if (!len_data)
    {
        return return_result;
    }

    while ( (retry_counter < 3) && (return_result != TRANSFER_OK) )
    {
        *len_data_sent = 0;

        retry_counter++;

        FD_ZERO (&fd_write_set);

        for(i = 0; i < num_sockets; i++)
        {
            FD_SET (sock[i], &fd_write_set);
        }

        tv.tv_sec = 10;
        tv.tv_usec = 0;

        retval = select (FD_SETSIZE, NULL, &fd_write_set, NULL, &tv); // Block until input arrives on one or more active sockets.

        if(retval == -1)
        {
            printf("Socket error at write data\n");
            return_result = ERROR_SOCKET_WRITE_SELECT;
        }
        else if(retval == 0)
            {
                printf("Socket timeout %d sec occurred at write\n", (uint8_t)tv.tv_sec);
                return_result = ERROR_SOCKET_WRITE_SELECT_TIMEOUT_OCCURRED;
            }
            else if (retval > 0)
                {
                    for(i = 0; i < num_sockets; i++)
                    {
                        if(FD_ISSET(sock[i], &fd_write_set))
                        {
                            len = send(sock[i], data, len_data, 0);
                            break;
                        }
                    }

                    if ((len) < 0)
                    {
                        printf("Error write to socket %d\n", len);
                        return_result = ERROR_SOCKET_WRITE_SEND;
                    }
                    else if (len == 0)
                        {
                            printf("Write ZERO to socket\n");
                            return_result = ERROR_SOCKET_WRITE_SEND_ZERO;
                        }
                        else if (len > 0)
                        {
                            if (len == len_data)
                            {
                                *len_data_sent = len;
                                return_result = TRANSFER_OK;
                            }
                            else
                            {
                                return_result = SOCKET_WRITE_IN_PROGRESS;
                            }
                        }
                }
    }

    return return_result;

}
[/code]

andrej102, 2021-05-14:
I continue to experiment with simultaneous transmission over multiple sockets.
At the moment, the results are as follows: 

1. Errors only if sending goes through several sockets (the more sockets the faster the error occurs, with two used sockets, the error occurs, but rarely, with 4 almost immediately after sending 8-10 packets of 1410 bytes) and does not depend on the number of open sockets (only on the number of used ones, if we open 4 but send only one, there will be no errors) 

2. Three types of error: 

   2.1.  WiFi just falls off (wifi:bcn_timout,ap_probe_send_start)

   2.2. Selec() returns on timeout. 

   2.3. When Select() returns the response that the socket is ready (returns > 0) to write, then Send() returns error 11 (which 
         means that an unblocked socket cannot accept all the data to send without blocking - and this seemed normal, but after 
         this error occurs, it usually never gets released, with rare exceptions, and this is not the normal behavior of the socket). 

There is an assumption that ESP32-S2 does not work correctly with several sockets at the same time, or there is not enough memory, or performance, but when working with several sockets, the wifi falls off, then inside the TCP/IP stack, it is mixed and damaged.

The main questions for the developers of espressif:  
1. can esp32-s2 work with multiple sockets at the same time, or not?
2. If it can, has it been tested?
3. If everything is fine, then what is the difference between the transfer to ESP 32 and ECP32-s2, what am I doing wrong?

Without solving this issue, the development of the project stopped. We manufacture and sell thousands of devices on ESP32 and it is extremely important for us to understand the possibility of using ESP32-S2 in our projects.

Thanks.

gabsuren, 2021-06-21:
Hi @andrej102

Thank you for your observations, in ESP32-S2 there are no limitations for using multiple sockets(it should work normally as in ESP32).
I have created a basic demo example for 5 sockets using ESP32-S2 and wasn't able to see any errors after running it multiple times(10 times). 
I have created the demo example `tcp_client` according to your sample code.
Could you please try again with the code below, and see if you still getting the error?

TCP Client
https://gist.github.com/gabsuren/dc297157a4f99e96008e8ab062589a10

Board: ESP32-S2
Environment: macOS Big Sur
IDF version: ESP-IDF-v4.2.

Thanks,
Suren

Alvin1Zhang, 2021-09-14:
Thanks for reporting, would close due to short of feedback, feel free to reopen with more updates.

closed 2021-09-14 by Alvin1Zhang