espressif/esp-idf #6961: esp_hid_device fails to build for ESP32-C3 with `'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_acl_conn'`

sztsian, 2021-04-29:
## Environment

- Development Kit:      [none]
- Kit version (for WroverKit/PicoKit/DevKitC): []
- Module or chip used:  **[ESP32-C3]**
- IDF version (run ``git describe --tags`` to find it):
   v4.4-dev-1183-g9d34a1cd4
- Build System:         [idf.py]
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    **riscv32-esp-elf-gcc (crosstool-NG 1.24.0.123_64eb9ff) 8.4.0**
- Operating System:     [Linux
- (Windows only) environment type: [].
- Using an IDE?: [No]
- Power Supply:         [USB]

## Problem Description

When building esp_hid_device for ESP32-C3 with idf.py, it fails with `'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_acl_conn'`

While compiling `wifi/getting_started/station` works just fine. So I feel this might be a bug somwhere.

### Expected Behavior

The example builds just fine.

### Actual Behavior

The example failed to build with error like

```
[ 98%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/esp_hid_gap.c.obj
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c: In function 'init_low_level':
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:675:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'mode'
         bt_cfg.mode = mode;
               ^
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:676:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_acl_conn'
         bt_cfg.bt_max_acl_conn = 3;
               ^
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:677:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_sync_conn'
         bt_cfg.bt_max_sync_conn = 3;
               ^
```

### Steps to reproduce

1. Setting up the environment following [this instruction](https://docs.espressif.com/projects/esp-idf/zh_CN/latest/esp32c3/get-started/index.html)
2. cp ../esp-idf/examples/bluetooth/esp_hid_device/ . -r
3. cd esp_hid_device
4. idf.py set-target esp32c3
5. idf.py menuconfig  (Change `Component config → ESP32C3-Specific → Minimum Supported ESP32-C3 Revision` to `Rev 2` , as the development board I have is Rev2)
6. idf.py build

### Code to reproduce this issue

Code is the [esp_hid_device example](https://github.com/espressif/esp-idf/tree/66fb5a29bbdc2482d67c52e6f66b303378c9b789/examples/bluetooth/esp_hid_device), without any change.

## Debug Logs

Tail of the compile log
```
Scanning dependencies of target __idf_esp_local_ctrl
Scanning dependencies of target __idf_wifi_provisioning
[ 98%] Built target __idf_esp_hid
Scanning dependencies of target __idf_main
[ 98%] Building C object esp-idf/esp_local_ctrl/CMakeFiles/__idf_esp_local_ctrl.dir/src/esp_local_ctrl.c.obj
[ 98%] Building C object esp-idf/esp_local_ctrl/CMakeFiles/__idf_esp_local_ctrl.dir/src/esp_local_ctrl_handler.c.obj
[ 98%] Building C object esp-idf/esp_local_ctrl/CMakeFiles/__idf_esp_local_ctrl.dir/proto-c/esp_local_ctrl.pb-c.c.obj
[ 98%] Building C object esp-idf/esp_local_ctrl/CMakeFiles/__idf_esp_local_ctrl.dir/src/esp_local_ctrl_transport_ble.c.obj
[ 98%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/esp_hid_device_main.c.obj
[ 98%] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/esp_hid_gap.c.obj
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c: In function 'init_low_level':
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:675:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'mode'
         bt_cfg.mode = mode;
               ^
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:676:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_acl_conn'
         bt_cfg.bt_max_acl_conn = 3;
               ^
/home/zsun/hardware/references/Espressif/getting_started/esp_hid_device/main/esp_hid_gap.c:677:15: error: 'esp_bt_controller_config_t' {aka 'struct <anonymous>'} has no member named 'bt_max_sync_conn'
         bt_cfg.bt_max_sync_conn = 3;
               ^
[ 99%] Linking C static library libesp_local_ctrl.a
make[2]: *** [esp-idf/main/CMakeFiles/__idf_main.dir/build.make:95: esp-idf/main/CMakeFiles/__idf_main.dir/esp_hid_gap.c.obj] Error 1
make[2]: *** Waiting for unfinished jobs....
[ 99%] Built target __idf_esp_local_ctrl
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_scan.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/manager.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/handlers.c.obj
[ 99%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_softap.c.obj
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_console.c.obj
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/wifi_config.c.obj
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_config.pb-c.c.obj
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_scan.pb-c.c.obj
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/proto-c/wifi_constants.pb-c.c.obj
make[1]: *** [CMakeFiles/Makefile2:4864: esp-idf/main/CMakeFiles/__idf_main.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Building C object esp-idf/wifi_provisioning/CMakeFiles/__idf_wifi_provisioning.dir/src/scheme_ble.c.obj
[100%] Linking C static library libwifi_provisioning.a
[100%] Built target __idf_wifi_provisioning
make: *** [Makefile:149: all] Error 2
make failed with exit code 2
```
## Other items if possible

The whole folder is simply too big. But I'll upload the sdkconfig in a comment.

- [ ] sdkconfig file (attach the sdkconfig file from your project folder)
- [ ] elf file in the ``build`` folder (**note this may contain all the code details and symbols of your project.**)
- [ ] coredump (This provides stacks of tasks.)

sztsian, 2021-04-29:
I've added a .txt suffix to satisfy Github upload limitations.
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6399804/sdkconfig.txt)

howardoer, 2021-05-02:
It seems that esp32c3 does not support Classis BT, so this demo does not work

Weijian-Espressif, 2021-05-06:
ESP32C3 does not support Classis BT. If you only need BLE HID, please refer to the attachment
[0001-hid-examples-support-esp32c3-s3.patch.zip](https://github.com/espressif/esp-idf/files/6434898/0001-hid-examples-support-esp32c3-s3.patch.zip)

Weijian-Espressif, 2021-05-06:
@sztsian @howardoer

sztsian, 2021-05-08:
> ESP32C3 does not support Classis BT. If you only need BLE HID, please refer to the attachment
> [0001-hid-examples-support-esp32c3-s3.patch.zip](https://github.com/espressif/esp-idf/files/6434898/0001-hid-examples-support-esp32c3-s3.patch.zip)

Ah okay, thanks for the reply. I'll close this.

closed 2021-05-08 by sztsian