espressif/esp-idf #6964: ESP-IDF 4.3 beta 3: "Cannot parse CMake requirements line" when component has uppercase character(s)

Elijahg, 2021-04-29:
## Environment

- Development Kit:      none
- Module or chip used:  ESP32-WROOM-32D
- IDF version v4.3-beta3
- Build System:         idf.py
- Compiler version: (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS Big Sur 11.3
- Using an IDE?: Usually VSCode, but issue occurs in CLI too.
- Power Supply:         external 3.3V

## Problem Description

Attempting to compile a component with a capital anywhere in its folder name results in a compilation failure. I was at one point compiling fine with the IDF from the 4.2 release branch, though I have since reinstalled my OS and have not retested it with that branch.

### Expected Behavior

The compilation should complete, no matter the case of the component. I am running on a case-insensitive filesystem so case sensitivity is not the cause.

### Actual Behavior

With a component named "i2cComms" compilation fails with the error:
Cannot parse CMake requirements line: __component_set_property(___idf_i2cComms REQUIRES "cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

Changing the name to i2ccomms fixes the error.

### Steps to reproduce

1. Create a new project.
2. Add a component with a name that has a capital somewhere.
3. Attempt to compile.

## Debug Logs

```
❯ idf.py reconfigure
Executing action: reconfigure
Running cmake in directory /Users/elijah/Scratch/moo/template-app/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /Users/elijah/Scratch/moo/template-app"...
-- Project is not inside a git repository, or git repository has no commits; will not use 'git describe' to determine PROJECT_VER.
-- Building ESP-IDF components for target esp32
Cannot parse CMake requirements line: __component_set_property(___idf_i2cComms REQUIRES "cxx;newlib;freertos;esp_hw_support;heap;log;lwip;soc;hal;esp_rom;esp_common;esp_system;esp32")

CMake Error at /Users/elijah/Scratch/esp-idf-v4.3-beta3/tools/cmake/component.cmake:243 (message):
Call Stack (most recent call first):
  /Users/elijah/Scratch/esp-idf-v4.3-beta3/tools/cmake/build.cmake:414 (__component_get_requirements)
  /Users/elijah/Scratch/esp-idf-v4.3-beta3/tools/cmake/project.cmake:396 (idf_build_process)
  CMakeLists.txt:6 (project)

-- Configuring incomplete, errors occurred!
See also "/Users/elijah/Scratch/moo/template-app/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1
[CMakeOutput.log](https://github.com/espressif/esp-idf/files/6402165/CMakeOutput.log)

```

## Other items if possible

[sdkconfig.zip](https://github.com/espressif/esp-idf/files/6402170/sdkconfig.zip)

igrr, 2021-04-30:
Hi @Elijahg, please follow the recommendation in this issue report: https://github.com/espressif/vscode-esp-idf-extension/issues/388#issuecomment-829075812.

Elijahg, 2021-04-30:
Thanks @igrr! Updating idf-component-manager to 0.2.1a1 fixes it - but the update has to be done manually via the CLI (in the Espressif Python virtual environment). Running `ESP-IDF: Install ESP-IDF Python Packages` doesn't update it, and weirdly `pip3 list --outdated` shows 0.2.1a0 as the latest. Maybe that's why `ESP-IDF: Install ESP-IDF Python Packages` doesn't update.

igrr, 2021-04-30:
Thanks for the note @Elijahg!

cc @brianignacio5 ^

closed 2021-06-01 by Alvin1Zhang