espressif/esp-idf #6965: https_server 握手过程解析证书错误

Zmmfly, 2021-04-30:
## Environment

- Module or chip used: 
   // ESP32-WROVER-E
- IDF version (run ``git describe --tags`` to find it): 
   // v4.3-beta3-dirty
- Build System: 
   // idf.py
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it):
    // xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:
   // Windows
- (Windows only) environment type: 
   // ESP Command Prompt
- Using an IDE?: 
   // Vscode
- Power Supply:
   // 3.3v

## Problem Description

配置好https_server后，浏览器访问https://192.168.4.1/
串口抛6次
<pre>
I (16690) esp_https_server: performing session handshake
E (16691) esp-tls-mbedtls: mbedtls_x509_crt_parse returned -0x2180
E (16692) esp-tls-mbedtls: Failed to set server pki context
E (16699) esp-tls-mbedtls: Failed to set server configurations
E (16705) esp-tls-mbedtls: create_ssl_handle failed
E (16711) esp_https_server: esp_tls_create_server_session failed
W (16717) httpd: httpd_accept_conn: session creation failed
W (16725) httpd: httpd_server: error accepting new connection
</pre>

最开始使用的是自签IP证书，更换成CA证书还是出现该错误，再更换成例程中的证书同样出现，
使用mbedtls_x509_crt_parse尝试解析证书始终返回-0x2180，print证书对比原文件无变化
嵌入方式为CMakeList.txt中使用EMBED_FILES

ESP-YJM, 2021-05-06:
@Zmmfly 这个错误是证书格式错误，能否把你的证书以及示例代码贴在这，我看一下。

Alvin1Zhang, 2021-05-10:
感谢反馈，将会关闭此问题，谢谢。

closed 2021-05-10 by Alvin1Zhang