espressif/esp-idf #6967: Compile error for mqtt/tcp example

ralphjy, 2021-04-30:
I am using esp-idf to build the mqtt/tcp example using v4.4-dev-1183-g9d34a1cd4 of the repo.

I am getting the following error:
../main/app_main.c:96:48: error: 'MQTT_ERROR_TYPE_TCP_TRANSPORT' undeclared (first use in this function); did you mean 'MQTT_ERROR_TYPE_ESP_TLS'?
         if (event->error_handle->error_type == MQTT_ERROR_TYPE_TCP_TRANSPORT) {
                                                ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                                                MQTT_ERROR_TYPE_ESP_TLS

I tried updating the submodules using "git submodule update --init --recursive" , but the error persists.

Alvin1Zhang, 2021-05-06:
@ralphjy Thanks for reporting. Would you please help provide more details as suggested in the [issue template](https://github.com/espressif/esp-idf/issues/new?assignees=&labels=&template=bug_report.md&title=)? Information like elf, sdk configuration, backtrace, log outputs, commit ID, hardware and etc. would help us debug further. Thanks.

ESP-YJM, 2021-05-06:
@ralphjy Could you follow the step and build?
* export IDF_PATH="$PWD"   //MUST enter the root directory
* ./install.sh
* . ./export.sh
* enter your example and idf.py build

ESP-YJM, 2021-05-06:
@ralphjy Could you please enter the directory `component/mqtt/esp-mqtt/include/mqtt_client.h` to check `MQTT_ERROR_TYPE_TCP_TRANSPORT` whether undeclared.

david-cermak, 2021-05-21:
Hi @RahulEspressif 

I can reproduce the issue when using IDF tagged `v4.2` to build an example from `master` (or `v4.4-dev`). This is not supported, I am afraid. I've explained it a bit here https://github.com/espressif/esp-idf/issues/6779#issuecomment-816429432, we're trying make the examples as generic as possible, but not always able to achieve the *forward* compatibility (works only from `v4.2.1`)

(duplicate of https://github.com/espressif/esp-idf/issues/6779)

Alvin1Zhang, 2021-08-31:
Thanks for reporting, will close now, feel free to reopen.

closed 2021-08-31 by Alvin1Zhang