espressif/esp-idf #6969: Monitor no access (error) over native USB after esp_wifi_init()

majodi, 2021-05-01:
- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

1. Fill in all the fields under **Environment** marked with [ ] by picking the correct option for you in each case and deleting the others.
2. Describe your problem.
3. Include [debug logs from the "monitor" tool](https://docs.espressif.com/projects/esp-idf/en/latest/get-started/idf-monitor.html#automatically-decoding-addresses), or [coredumps](https://docs.espressif.com/projects/esp-idf/en/latest/api-guides/core_dump.html).
4. Providing as much information as possible under **Other items if possible** will help us locate and fix the problem.
5. Use [Markdown](https://guides.github.com/features/mastering-markdown/) (see formatting buttons above) and the Preview tab to check what the issue will look like.
6. Delete these instructions from the above to the below marker lines before submitting this issue.

**IMPORTANT: If you do not follow these instructions and provide the necessary details, your issue may not be resolved.**

----------------------------- Delete above -----------------------------

## Environment

Custom board with ESP32-S2 WROVER module but also verified on Saola
- Module or chip used:  ESP32-S2 WROVER
- IDF version: v4.4-dev-1183-g9d34a1cd4
- Build System: idf.py
- Compiler version: 8.4.0
- Operating System: macOS
- Using an IDE?: VSCode
- Power Supply: USB

## Problem Description

When flashing DFU over native USB, monitor gets:

read failed: [Errno 6] Device not configured
Waiting for the device to reconnect

But code is working.

Narrowed down to esp_wifi_init() call. If not calling esp_wifi_init() the monitor has normal access.

(See stripped down code)

### Expected Behavior

Besides working code also access with monitor over native USB

### Actual Behavior

Code works but no access over native USB (error)

### Steps to reproduce

Uncomment call to esp_wifi_init(&cfg)

### Code to reproduce this issue

```cpp
void app_main(void)
{
    static httpd_handle_t server = NULL;

    ESP_ERROR_CHECK(nvs_flash_init());
    ESP_ERROR_CHECK(esp_netif_init());
    ESP_ERROR_CHECK(esp_event_loop_create_default());
    esp_netif_create_default_wifi_ap();
    wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
    //ESP_ERROR_CHECK(esp_wifi_init(&cfg)); // ===========> uncomment will show described behavior

    while (1) {
        ESP_LOGI(TAG, "Testing...");
        vTaskDelay(1000 / portTICK_PERIOD_MS);
    }
}
```
## Debug Logs

No access to monitor (the problem), board does not have UART bridge

majodi, 2021-05-07:
Additional info:

Again, only using the native USB post (or trying so). When flashing DFU without wifi code I get a new serial port. But if I then turn on wifi (uncomment wifi code) and do idf.py build and flash over this new port, the firmware is working but the port is now gone.

So, I can create a dfu with wifi code and flash it (dfu-flash), I get a new serial port but using monitor over this port shows [Errno 6] Device not configured. After a while, closing the monitor the new port is gone. I can reset the device to get the port back but still not working in monitor.

If I try an idf.py flash on this port I also get the same error and also the port is gone afterwards. As if wifi and native USB can not coexist.

majodi, 2022-01-08:
For my sanity (maybe I just miss something here), would it be possible to let me know if this can be confirmed as a problem? I prepared an even simpler way to demonstrate the issue. Just follow these few steps:

copy the hello world: cp -r $IDF_PATH/examples/get-started/hello_world .
set-target esp32s2
menuconfig --> set console output to CDC
dfu
dfu-flash
monitor
=== output shows up in monitor ===

now add these few lines:

#include <esp_wifi.h>

===> and just at the top of main:
wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
ESP_ERROR_CHECK(esp_wifi_init(&cfg));

repeat the dfu, dfu-flash and monitor

=== not possible to connect with monitor tool ===

Dazza0, 2022-03-04:
This think this is same issue as #8241,  should be closed on master by f82ab4c757d36a0ac9ebb384998abd158e79db7b and on release/v4.4 by bf523af21d89577ec5c95335ce7a3bc1ba61886c. If the issue isn't resolved, please feel free to reopen this issue.

closed 2022-03-04 by Dazza0