espressif/esp-idf #6971: IP_EVENT_ETH_LOST_IP missing

0xFEEDC0DE64, 2021-05-02:
I am trying to implement better error handling for all network related events and it looks like there are events for when the IP was successfully assigned and also events when we loose ip (dhcp lease timeout and dhcp server doesnt respond anymore, or other failure condition).

I found these matching pairs:
IP_EVENT_STA_GOT_IP - IP_EVENT_STA_LOST_IP
IP_EVENT_PPP_GOT_IP - IP_EVENT_PPP_LOST_IP
IP_EVENT_ETH_GOT_IP - ??? when we loose ip?

I am missing a IP_EVENT_ETH_LOST_IP

Shouldnt the same error handling be also possible for the ethernet interface?

ESP-YJM, 2021-05-06:
@0xFEEDC0DE64 Thanks for your suggestion. Could you please try this patch.
```
diff --git a/components/esp_netif/include/esp_netif_defaults.h b/components/esp_netif/include/esp_netif_defaults.h
index 7ef6526..903a678 100644
--- a/components/esp_netif/include/esp_netif_defaults.h
+++ b/components/esp_netif/include/esp_netif_defaults.h
@@ -55,7 +55,7 @@ extern "C" {
         ESP_COMPILER_DESIGNATED_INIT_AGGREGATE_TYPE_EMPTY(mac) \
         ESP_COMPILER_DESIGNATED_INIT_AGGREGATE_TYPE_EMPTY(ip_info) \
         .get_ip_event = IP_EVENT_ETH_GOT_IP, \
-        .lost_ip_event = 0, \
+        .lost_ip_event = IP_EVENT_ETH_LOST_IP, \
         .if_key = "ETH_DEF", \
         .if_desc = "eth", \
         .route_prio = 50 \
diff --git a/components/esp_netif/include/esp_netif_types.h b/components/esp_netif/include/esp_netif_types.h
index 044ca0d..7c68c71 100644
--- a/components/esp_netif/include/esp_netif_types.h
+++ b/components/esp_netif/include/esp_netif_types.h
@@ -88,6 +88,7 @@ typedef enum {
     IP_EVENT_AP_STAIPASSIGNED,         /*!< soft-AP assign an IP to a connected station */
     IP_EVENT_GOT_IP6,                  /*!< station or ap or ethernet interface v6IP addr is preferred */
     IP_EVENT_ETH_GOT_IP,               /*!< ethernet got IP from connected AP */
+    IP_EVENT_ETH_LOST_IP,              /*!< ethernet lost IP and the IP is reset to 0 */
     IP_EVENT_PPP_GOT_IP,               /*!< PPP interface got IP */
     IP_EVENT_PPP_LOST_IP,              /*!< PPP interface lost IP */
 } ip_event_t;

```

closed 2021-05-17 by espressif-bot

AxelLin, 2021-05-18:
Hi @ESP-YJM 
I update to v4.4-dev-1390-g1ca2afd98223, but I never receive the IP_EVENT_ETH_LOST_IP event.
Could you share how to test (verify) it?

ESP-YJM, 2021-05-18:
`IP_EVENT_ETH_LOST_IP` will happen after you unplug the network cable `ESP_NETIF_IP_LOST_TIMER_INTERVAL` seconds later (default 120 seconds). This is the same with `IP_EVENT_STA_LOST_IP`.

AxelLin, 2021-05-18:
Thanks, I plug the ethernet cable back too quickly in my previous test.