espressif/esp-idf #6975: openthread/ot_cli example not working

cahty82008, 2021-05-03:
I am using the [ot_cli ](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli) example, commit [64c498a](https://github.com/espressif/esp-idf/commit/64c498a58418684c78a8ee49a7a8102923fdf5c7)  

my hardware: ESP32-WROVER-B + Nrf52840DK, it works properly with the  [ot-esp32](https://github.com/openthread/ot-esp32) example 

Then i use the Nrf52840 firmware in [ot-esp32](https://github.com/openthread/ot-esp32), which use p0.06 and p0.08 as the uart pin, and i swicth the ESP32 firmware from [ot-esp32](https://github.com/openthread/ot-esp32) to the [ot_cli ](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli) example, but the ot_cli example can not get the reply from my Nrf52840 board.

output:
```
E(4432) OPENTHREAD:[CRIT]-PLAT----: HandleRcpTimeout() at ../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:2187: RadioSpinelNoResponse

abort() was called at PC 0x400d9a6f on core 1
0x400d9a6f: syscall_not_implemented_aborts at /home/jsh/esp/esp-idf/components/newlib/syscalls.c:34

Backtrace:0x400d3227:0x3ffbe1f00x40086275:0x3ffbe210 0x4008afea:0x3ffbe230 0x400d9a6f:0x3ffbe2a0 0x40114bc3:0x3ffbe2c0 0x40105592:0x3ffbe2e0 0x401055e4:0x3ffbe310 0x40105b50:0x3ffbe330 0x40105d01:0x3ffbe370 0x4010ef5a:0x3ffbe3a0 0x4010cc30:0x3ffbe3c0 0x40088c49:0x3ffbe460 
0x400d3227: panic_abort at /home/jsh/esp/esp-idf/components/esp_system/panic.c:354

0x40086275: esp_system_abort at /home/jsh/esp/esp-idf/components/esp_system/esp_system.c:126

0x4008afea: abort at /home/jsh/esp/esp-idf/components/newlib/abort.c:46

0x400d9a6f: syscall_not_implemented_aborts at /home/jsh/esp/esp-idf/components/newlib/syscalls.c:34

0x40114bc3: exit at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x40105592: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::HandleRcpTimeout() at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:2187

0x401055e4: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::WaitResponse() at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:1659

0x40105b50: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::Init(bool, bool, bool) at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:239

0x40105d01: esp_openthread_radio_init at /home/jsh/esp/esp-idf/components/openthread/port/esp_openthread_radio_uart.cpp:35

0x4010ef5a: esp_openthread_platform_init at /home/jsh/esp/esp-idf/components/openthread/port/esp_openthread.cpp:61

0x4010cc30: ot_task_worker at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../main/ot_esp_cli.c:84

0x40088c49: vPortTaskWrapper at /home/jsh/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 778a715de2063d7c

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6756
load:0x40078000,len:14224
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.4-dev-1183-g9d34a1cd4 2nd stage bootloader
I (29) boot: compile time 09:59:08
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 ot_storage       Unknown data     01 3a 00110000 00002000
I (100) boot: End of partition table
I (104) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=14d28h ( 85288) map
I (152) esp_image: segment 1: paddr=00024d50 vaddr=3ffb0000 size=023d0h (  9168) load
I (156) esp_image: segment 2: paddr=00027128 vaddr=40080000 size=08ef0h ( 36592) load
I (174) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=55208h (348680) map
I (306) esp_image: segment 4: paddr=00085230 vaddr=40088ef0 size=031e0h ( 12768) load
I (311) esp_image: segment 5: paddr=00088418 vaddr=50000000 size=00010h (    16) load
I (318) boot: Loaded app from partition at offset 0x10000
I (319) boot: Disabling RNG early entropy source...
I (334) cpu_start: Pro cpu up.
I (335) cpu_start: Starting app cpu, entry point is 0x40081100
0x40081100: call_start_cpu1 at /home/jsh/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (321) cpu_start: App cpu up.
I (351) cpu_start: Pro cpu start user code
I (351) cpu_start: cpu freq: 160000000
I (351) cpu_start: Application information:
I (356) cpu_start: Project name:     ot_esp_cli
I (361) cpu_start: App version:      v4.4-dev-1183-g9d34a1cd4
I (367) cpu_start: Compile time:     Apr 30 2021 09:58:59
I (373) cpu_start: ELF file SHA256:  778a715de2063d7c...
I (379) cpu_start: ESP-IDF:          v4.4-dev-1183-g9d34a1cd4
I (386) heap_init: Initializing. RAM available for dynamic allocation:
I (393) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (399) heap_init: At 3FFB7958 len 000286A8 (161 KiB): DRAM
I (405) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (412) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (418) heap_init: At 4008C0D0 len 00013F30 (79 KiB): IRAM
I (425) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
W (433) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.

```

chshu, 2021-05-06:
It's probably caused by the different uart pin config, the radio uart pins in ot-esp32 are 2&5, while the pins in ot_cli example here is 4&5.

cahty82008, 2021-05-07:
@chshu 
hi, shu chen, thanks for the reply.

The radio uart pins in ot-esp32 are actually the same as in  ot_cli example, both are 4&5, 

here is my uart connection: [pic](https://github.com/openthread/ot-esp32/blob/master/doc/images/connect-devices.jpg)
 G4(ESP32-WROVER-B/ESP32-WROOM-32)  -> P0.08(Nrf52840DK )
 G5(ESP32-WROVER-B/ESP32-WROOM-32)  -> P0.06(Nrf52840DK )

gjc13, 2021-05-07:
The connection is supposed to be the opposite:
G4(ESP32-WROVER-B/ESP32-WROOM-32) -> P0.06(Nrf52840DK )
G5(ESP32-WROVER-B/ESP32-WROOM-32) -> P0.08(Nrf52840DK )

For one reason or another we are using the opposite RX/TX pin as in ot-esp32.

cahty82008, 2021-05-07:
I changed the connection, ESP32 throw another error and keeps restarting,
log:
```
E(370) OPENTHREAD:[CRIT]-PLAT----: RCP is missing required capabilities: tx-security tx-timing 
E(370) OPENTHREAD:[CRIT]-PLAT----: CheckRadioCapabilities() at ../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:381: RadioSpinelIncomp

abort() was called at PC 0x400d9a6f on core 1
0x400d9a6f: syscall_not_implemented_aborts at /home/jsh/esp/esp-idf/components/newlib/syscalls.c:34

Backtrace:0x400d3227:0x3ffbe2000x40086275:0x3ffbe220 0x4008afea:0x3ffbe240 0x400d9a6f:0x3ffbe2b0 0x40114bc3:0x3ffbe2d0 0x40105b2d:0x3ffbe2f0 0x40105c2c:0x3ffbe330 0x40105d01:0x3ffbe370 0x4010ef5a:0x3ffbe3a0 0x4010cc30:0x3ffbe3c0 0x40088c49:0x3ffbe460 
0x400d3227: panic_abort at /home/jsh/esp/esp-idf/components/esp_system/panic.c:354

0x40086275: esp_system_abort at /home/jsh/esp/esp-idf/components/esp_system/esp_system.c:126

0x4008afea: abort at /home/jsh/esp/esp-idf/components/newlib/abort.c:46

0x400d9a6f: syscall_not_implemented_aborts at /home/jsh/esp/esp-idf/components/newlib/syscalls.c:34

0x40114bc3: exit at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/src/newlib/newlib/libc/stdlib/exit.c:64

0x40105b2d: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::CheckRadioCapabilities() at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:381 (discriminator 20)

0x40105c2c: ot::Spinel::RadioSpinel<esp::openthread::UartSpinelInterface, esp_openthread_mainloop_context_t>::Init(bool, bool, bool) at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../../../../components/openthread/openthread/src/lib/spinel/radio_spinel_impl.hpp:263

0x40105d01: esp_openthread_radio_init at /home/jsh/esp/esp-idf/components/openthread/port/esp_openthread_radio_uart.cpp:35

0x4010ef5a: esp_openthread_platform_init at /home/jsh/esp/esp-idf/components/openthread/port/esp_openthread.cpp:61

0x4010cc30: ot_task_worker at /home/jsh/esp/esp-idf/examples/openthread/ot_cli/build/../main/ot_esp_cli.c:84

0x40088c49: vPortTaskWrapper at /home/jsh/esp/esp-idf/components/freertos/port/xtensa/port.c:168

ELF file SHA256: 778a715de2063d7c

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:6756
load:0x40078000,len:14224
ho 0 tail 12 room 4
load:0x40080400,len:3716
0x40080400: _init at ??:?

entry 0x40080680
I (29) boot: ESP-IDF v4.4-dev-1183-g9d34a1cd4 2nd stage bootloader
I (29) boot: compile time 09:59:08
I (29) boot: chip revision: 1
I (33) boot_comm: chip revision: 1, min. bootloader chip revision: 0
I (40) boot.esp32: SPI Speed      : 40MHz
I (45) boot.esp32: SPI Mode       : DIO
I (49) boot.esp32: SPI Flash Size : 2MB
I (54) boot: Enabling RNG early entropy source...
I (59) boot: Partition Table:
I (63) boot: ## Label            Usage          Type ST Offset   Length
I (70) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (78) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (85) boot:  2 factory          factory app      00 00 00010000 00100000
I (93) boot:  3 ot_storage       Unknown data     01 3a 00110000 00002000
I (100) boot: End of partition table
I (104) boot_comm: chip revision: 1, min. application chip revision: 0
I (112) esp_image: segment 0: paddr=00010020 vaddr=3f400020 size=14d28h ( 85288) map
I (152) esp_image: segment 1: paddr=00024d50 vaddr=3ffb0000 size=023d0h (  9168) load
I (156) esp_image: segment 2: paddr=00027128 vaddr=40080000 size=08ef0h ( 36592) load
I (174) esp_image: segment 3: paddr=00030020 vaddr=400d0020 size=55208h (348680) map
I (306) esp_image: segment 4: paddr=00085230 vaddr=40088ef0 size=031e0h ( 12768) load
I (311) esp_image: segment 5: paddr=00088418 vaddr=50000000 size=00010h (    16) load
I (318) boot: Loaded app from partition at offset 0x10000
I (319) boot: Disabling RNG early entropy source...
I (334) cpu_start: Pro cpu up.
I (335) cpu_start: Starting app cpu, entry point is 0x40081100
0x40081100: call_start_cpu1 at /home/jsh/esp/esp-idf/components/esp_system/port/cpu_start.c:150

I (321) cpu_start: App cpu up.
I (351) cpu_start: Pro cpu start user code
I (351) cpu_start: cpu freq: 160000000
I (351) cpu_start: Application information:
I (356) cpu_start: Project name:     ot_esp_cli
I (361) cpu_start: App version:      v4.4-dev-1183-g9d34a1cd4
I (367) cpu_start: Compile time:     Apr 30 2021 09:58:59
I (373) cpu_start: ELF file SHA256:  778a715de2063d7c...
I (379) cpu_start: ESP-IDF:          v4.4-dev-1183-g9d34a1cd4
I (386) heap_init: Initializing. RAM available for dynamic allocation:
I (393) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (399) heap_init: At 3FFB7958 len 000286A8 (161 KiB): DRAM
I (405) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (412) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (418) heap_init: At 4008C0D0 len 00013F30 (79 KiB): IRAM
I (425) spi_flash: detected chip: generic
I (429) spi_flash: flash io: dio
W (433) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I(350) OPENTHREAD:[INFO]-PLAT----: RCP reset: RESET_SOFTWARE
I(370) OPENTHREAD:[NOTE]-PLAT----: RCP API Version: 1

```

chshu, 2021-05-08:
The two missed RCP capabilities `tx-security tx-timing` are newly added, so the old RCP FW doesn't work with the new ot_cli exmaple, please updae the RCP FW with following steps:

```
cd $IDF_PATH/components/openthread/openthread/
./bootstrap
make -f examples/Makefile-nrf52840 CFLAGS="-DUART_HWFC_ENABLED=0"
arm-none-eabi-objcopy -O ihex output/nrf52840/bin/ot-rcp ot-rcp.hex
nrfjprog -f nrf52 --chiperase --program ot-rcp.hex --reset
```

cahty82008, 2021-05-10:
thanks a lot, it's working now!

Alvin1Zhang, 2021-09-18:
Thanks for reporting, glad it is working, feel free to reopen.

closed 2021-09-18 by Alvin1Zhang