espressif/esp-idf #6976: flash.rodata linked at 0x3f400040

bpietsch, 2021-05-04:
Hello, I am having an issue that sounds eerily similar to this recently closed issue:

https://github.com/espressif/esp-idf/issues/6719

## Environment

- Module or chip used:  ESP32-WROVER-IB
- IDF version (run ``git describe --tags`` to find it): 4.2 release
- Build System:         visual studio code extension
- Compiler version (run ``xtensa-esp32-elf-gcc --version`` to find it): xtensa-esp32-elf-gcc (crosstool-NG esp-2020r3) 8.4.0
- Operating System:     macOS

## Problem Description
image generation fails due to inability to insert sha256 checksum, which I traced down to flash.rodata being linked at 0x3f400040 instead of 0x3f400020

//Detailed problem description goes here.
While this certainly seems like the same issue as the related one I noted above, I see no evidence in my map file (included below) of a forced alignment of any size (and I'm not linking any static libraries either as that person was. I am building some 3rd party source code (c++ runtime portion of the google protobuf project - https://github.com/protocolbuffers/protobuf/tree/v3.15.8), but a search of that code also does not show any signs of forced alignment (there are a couple of function pointer return values that are tagged with "__attribute__((aligned(32)))", but that is 32 anyway and as I understand it indicates the alignment of the returned memory, not the code itself.

[files.zip](https://github.com/espressif/esp-idf/files/6419023/files.zip)

bpietsch, 2021-05-04:
Well, for grins I just tried disabling my use of protobuf and the problem went away, so that is clearly the source of the issue, but I don't at all understand why. Any info to be gleaned from the map file or binary?

bpietsch, 2021-05-07:
Further isolated to this line of code in arena_impl.h:

  struct alignas(64) ThreadCache {

If I change 64 to 32 the issue goes away. Why would a data alignment attribute cause the issue I'm seeing?

bpietsch, 2021-05-09:
Filed this issue against protobuf since it technically appears to be their issue:

https://github.com/protocolbuffers/protobuf/issues/8591

o-marshmallow, 2021-05-18:
@bpietsch Thank you for your patience. Some commits were added to our internal `release/v4.2` branch to correct this issue. Once the CI passes, it will be synchronized to Github and you will then be able to retrieve the fix.

Alvin1Zhang, 2021-06-30:
Thanks for reporting, fix is available https://github.com/espressif/esp-idf/commit/974b8dd4c40cc38c451089eeb9bf90c6dc22a22d, feel free to reopen.

closed 2021-06-30 by Alvin1Zhang

bpietsch, 2021-07-22:
confirming that this resolved my issue, thank you!