espressif/esp-idf #6977: Basic ethernet example fails with LAN8742

SimonPVS, 2021-05-04:
## Environment

- Custom module
- ESP-WROOM 32
- ESP-IDF v4.2
- Cmake build system
- Compiler version: esp-2020r3-8.4.0
- Operating System:  Windows
- Using an IDE?: Yes
                           Eclipse (v4.19), IDF plugin (v9.11.1)
- Power Supply:  USB

## Problem Description

I am using a custom board with an ESP-32 WROOM and as a PHY the LAN8742a.
I have a 25MHz crystal and the PHY has a 50MHz clock output to the ESP module GPIO 0.
The LAN8742a is configured to have adress 0.

I get this error message (after the error message the esp reboots)
`E (6507) lan8720: lan8720_pwrctl(344): power up timeout
E (6507) lan8720: lan8720_init(386): power control failed
E (6507) esp_eth: esp_eth_driver_install(199): init phy failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40083ee4
file: "../main/ethernet_example_main.c" line 138
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x40083ee7 on core 0

Backtrace:0x40084357:0x3ffb55b0 0x400849f5:0x3ffb55d0 0x40088452:0x3ffb55f0 0x40083ee7:0x3ffb5660 0x400d4545:0x3ffb5680 0x400d2eda:0x3ffb56f0 0x400849fd:0x3ffb5720

ELF file SHA256: da372cd968e6f382`

I know it fails in the init loop because the phy does not seem to respond: 
[image]

If i use the automatic phy addres finder, it does not find a phy.

### Expected Behavior

The PHY should respond to the register 0 read.

### Actual Behavior

The PHY does not respond.

### Steps to reproduce

1. Load the basic example
2. Change the config to LAN8720, PHY address 0
3. RMII clock mode: input RMII clock from external
4. Flash the code

[image]
[image]

### Code to reproduce this issue

Basic ethernet example. 

## Debug Logs

```
E (6507) lan8720: lan8720_pwrctl(344): power up timeout
E (6507) lan8720: lan8720_init(386): power control failed
E (6507) esp_eth: esp_eth_driver_install(199): init phy failed
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40083ee4
file: "../main/ethernet_example_main.c" line 138
func: app_main
expression: esp_eth_driver_install(&config, &eth_handle)

abort() was called at PC 0x40083ee7 on core 0

Backtrace:0x40084357:0x3ffb55b0 0x400849f5:0x3ffb55d0 0x40088452:0x3ffb55f0 0x40083ee7:0x3ffb5660 0x400d4545:0x3ffb5680 0x400d2eda:0x3ffb56f0 0x400849fd:0x3ffb5720

ELF file SHA256: da372cd968e6f382

Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4
load:0x3fff0034,len:7072
ho 0 tail 12 room 4
load:0x40078000,len:13212
load:0x40080400,len:4568
entry 0x400806f4
```

## Other items if possible
[sdkconfig.txt](https://github.com/espressif/esp-idf/files/6419334/sdkconfig.txt)

closed 2021-05-04 by SimonPVS

panzer412, 2021-06-08:
Hi @SimonPVS, I am Ondrej from Espressif. You closed the issue, does it mean you've resolved it? If so, could you please share how?

SimonPVS, 2021-06-08:
Hi, 

Yes, this works now. I had two problems:
- I needed to change the vendor model address from 0xF to 0x13.
- There was a bad soldering

panzer412, 2021-06-08:
Thank you for the information. Glad to hear you figured it out. 
Regarding Model Number, this going to be updated soon in master branch. There is already PR [#7097](https://github.com/espressif/esp-idf/pull/7097) raised by user however we need to process it properly.