espressif/esp-idf #6978: Log memory allocation errors in esp_wifi driver

tom-van, 2021-05-04:
IDF git v4.4-dev-1183-g9d34a1cd4
ESP32-S2-WROOM module as WiFi STA WPA2-PSK

Problem:

The WiFi driver allocates memory probably for any received beacon.
If the heap runs out of space, failed allocations cause problems that **looks in the log exactly as the WiFi signal was lost.**

```
I (51848) wifi:bcn_timout,ap_probe_send_start
W (51848) wifi:m f probe req l=0
```

I know that all old ESP gurus know that this message should be interpreted as possible low heap space.
But honestly would not be better to **send a log message making clear that an allocation failure happened**?
For the test I added ESP_LOGE() to all mem allocating wrappers in esp-idf/components/esp_wifi/esp32s2/esp_adapter.c
and I got:

```
W (45248) wifi:<ba-add>idx:1 (ifx:0, xxx), tid:0, ssn:1, winSize:64
E (45258) esp_adapter: wifi_malloc: no memory
E (45258) esp_adapter: wifi_malloc: no memory
```
... about 50 failed malloc() ...

```
E (51788) esp_adapter: wifi_malloc: no memory
I (51848) wifi:bcn_timout,ap_probe_send_start
W (51848) wifi:m f probe req l=0
```

Moreover the first part of log shows that there is low memory even in WiFi connect.
This problem was invisible without added memory alloc logs:

```
I (438) wifi:Init data frame dynamic rx buffer num: 32
I (438) wifi:Init management frame dynamic rx buffer num: 32
I (448) wifi:Init management short buffer num: 32
I (448) wifi:Init dynamic tx buffer num: 32
I (448) wifi:Init static rx buffer size: 1600
I (458) wifi:Init static rx buffer num: 10
I (458) wifi:Init dynamic rx buffer num: 32
I (598) wifi:mode : sta (xxx)
I (598) wifi:enable tsf
connecting to network...
I (1218) wifi:new:<6,0>, old:<1,0>, ap:<255,255>, sta:<6,0>, prof:1
I (2538) wifi:state: init -> auth (b0)
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2548) esp_adapter: wifi_malloc: no memory
I (2558) wifi:state: auth -> assoc (0)
I (2568) wifi:state: assoc -> run (10)
I (2578) wifi:connected with xxxx, aid = 3, channel 6, BW20, bssid = xxxx
I (2578) wifi:security: WPA2-PSK, phy: bgn, rssi: -53
I (2578) wifi:pm start, type: 1
 
W (2608) wifi:<ba-add>idx:0 (ifx:0, xxx), tid:6, ssn:1, winSize:64
I (2628) wifi:AP's beacon interval = 102400 us, DTIM period = 2
```

The WiFi driver code is not published so I don't know if the driver uses some workaround (uses a static buffer) in case of failed allocation. If so it does not work at least for beacon reception.
I strongly prefer to be informed about allocation errors in the log.

AxelLin, 2021-05-04:
I have similar issue with v4.4-dev-1183-g9d34a1cd42f6, testing with ESP32-WROOM-32E.
I notice one of my test device disconnect due to WIFI_REASON_BEACON_TIMEOUT for 1300+ times.
My device can still re-join the AP, but the free memory becomes very low (about 10 ~ 16K).

xueyunfei998, 2021-09-14:
hi @tom-van 
The device may receive a large number of probe reqs in a certain period of time. This phenomenon will occur when responding to the probe response

xueyunfei998, 2022-03-18:
hi @tom-van 

Your example sta mode, so there is no situation of receiving a large number of probe reqs.

According to your log, when the system just got up, it ran out of memory.

I (2538) wifi:state: init -> auth (b0)
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2538) esp_adapter: wifi_malloc: no memory
E (2548) esp_adapter: wifi_malloc: no memory

I have a few questions：

-  Did you apply for a lot of memory in your project?
- Does your board use psram?

If you want to save memory, you can reduce the size of wifi tx rx buffer in menuconfig.

AxelLin, 2022-04-08:
@tom-van 
Any update regarding the comment in https://github.com/espressif/esp-idf/issues/6978#issuecomment-1072076852 ?
Do you still hit the issue with v4.4 release or recent update of v4.4 branch?

tom-van, 2022-04-08:
@xueyunfei998 

> I have a few questions：
> 
>     * Did you apply for a lot of memory in your project?

I did. And of course decreasing the amount of requested memory in the app fixed the original issue.

>     * Does your board use psram?

No.

Please be aware, that topic is about **Logging memory allocation errors**.  In case of failed mem alloc the current IDF wifi adapter code does not output anything relevant to the log and even worse it behaves just as if it lose wifi signal. It would be much easier to fix the application if you did know you were running out of memory.

@AxelLin
At least components/esp_wifi/esp32/esp_adapter.c and components/esp_wifi/esp32s2/esp_adapter.c still does not log memory allocation errors in current git master.

AxelLin, 2022-04-08:
> @AxelLin At least components/esp_wifi/esp32/esp_adapter.c and components/esp_wifi/esp32s2/esp_adapter.c still does not log memory allocation errors in current git master.

Maybe you can try heap_caps_register_failed_alloc_callback().

tom-van, 2022-04-08:
All wifi mem alloc goes through wrappers in esp_adapter.c, so adding ESP_LOGE() to wifi_malloc(), realloc, calloc and other 4 allocations in this source works for me just fine. I did it just for esp32s2 where I experienced the problem. I'm not eager to make a complete PR for all devices including those I don't have.

AxelLin, 2022-04-08:
@tom-van 
Just curious how much free memory available after your application started?

closed 2022-11-16 by Alvin1Zhang